FreeCAD/FreeCAD #15134: New dark theme: Tab buttons have inconsistent right/left padding

dapetcu21, 2024-07-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]

### Full version info

```shell
OS: macOS 14.4.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git)
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MisterMakerNL, 2024-07-07:
Related https://github.com/FreeCAD/FreeCAD/pull/14592
It's annoying to find the correct settings so it takes a bit longer.

MisterMakerNL, 2025-07-24:
Mac has set to Fusion style so if QT worked as it should this shouldn't be a thing anymore. So need confirmation.

JohnOCFII, 2025-07-24:
> Looks like it was an issue on macOS, perhaps [@JohnOCFII](https://github.com/JohnOCFII) can weigh-in ?

Other than switching to FreeCAD Dark (which I did) vs the OpenDark I usually use, not sure what else might need to be tweaked.  My `View | Data` tabs look fine.

Also - heading camping in a couple of hours, so won't be near a computer to test again until Monday.

[image]

```
OS: macOS 15.5
Architecture: arm64
Version: 1.1.0dev.42653 (Git)
Build date: 2025/07/21 16:35:39
Build type: Release
Branch: main
Hash: c99cd73cd60874f231bda6233477e9051afcea57
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * OpenTheme 2025.5.20
  * AddonManager 2025.7.17
  * Telemetry 1.0.1
```

FEA-eng, 2025-10-29:
Looks fixed (can't check on Mac though).

[image]

```
OS: Windows 11 build 26200
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/10/21 12:07:13
Build type: Release
Branch: main
Hash: 2096d0bd8603b429fdc0c746c34b3d73c3289fe1
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/113.268/1.25
```

FEA-eng, 2026-04-29:
@maxwxyz Can you check if it's also fixed on macOS?