FreeCAD/FreeCAD #15135: New dark theme: Certain buttons are missing left-right padding

dapetcu21, 2024-07-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]
[image]

This is not an issue on Classic:

[image]
[image]

### Full version info

```shell
OS: macOS 14.4.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git)
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-08-27:
@dapetcu21 is this still relevant on latest dev build ?

MisterMakerNL, 2024-09-24:
there is some other issues that just adding padding causes, some buttons that need to be smaller then get too big. So this isn't as easy as it looks.

luzpaz, 2024-10-16:
@benj5378 any ideas here ?

maxwxyz, 2025-03-09:
@MisterMakerNL solved by your PR?

closed 2025-07-11 by kadet1090