FreeCAD/FreeCAD #15136: Dark Theme: In settings menu, checkboxes are misaligned

dapetcu21, 2024-07-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The checkboxes don't align with other text labels in the same column. 

New theme:

[image]

Classic:

[image]

### Full version info

```shell
OS: macOS 14.4.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git)
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

MisterMakerNL, 2024-07-20:
I re-did the checkboxes here https://github.com/FreeCAD/FreeCAD/pull/15471
After merge need to test this.

luzpaz, 2024-07-24:
@dapetcu21 please test most up to date dev version that includes these fixes and report back pls. TIA

dapetcu21, 2024-07-24:
Still misaligned in 3fa339e9
(also related to this, I know it might be subjective, but I think there should be about 2px more padding between checkboxes and their label)

[image]

luzpaz, 2024-07-27:
@MisterMakerNL any idea for a solution going forward ?

MisterMakerNL, 2024-07-27:
Did some googling found out you can add   "spacing:8px;" to checkboxes.
This seems to work perfectly, as it leaves out any single used checkbox and any text.

MisterMakerNL, 2024-07-28:
I btw made it 8px but the radio button already had 4px, so maybe 8 is overkill?

FEA-eng, 2024-08-03:
It seems that today's Windows weekly has more space between each checkbox and text but the checkboxes are aligned in the same way as they were before.

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38385 (Git)
Build type: Release
Branch: main
Hash: 58241e027e7187a1ecaeeb75668ef9a497437c68
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

MisterMakerNL, 2024-08-03:
Could you be more specific what alignment you mean?

FEA-eng, 2024-08-03:
> Could you be more specific what alignment you mean?

I'm referring to the original issue description. From what I understand, it's about the checkboxes not being in line with the first letter of each preference text without a checkbox:

[image]

MisterMakerNL, 2024-08-03:
Ah this is pretty easy fix, seems to be Qlabel having 2px padding!

closed 2024-08-05 by chennes