FreeCAD/FreeCAD #15138: Additive Loft Pad Issue

Nigelkdx200, 2024-07-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I can successfully create the part in Freecad 0.21 as expected however trying to do the same thing in 0.22 doesn't work.  To recreate the issue create 2 sketches in Part Design on the XY plane. For the first sketch draw an outer circle of 60mm diameter and an inner circle of 50mm diameter. Then on the 2nd sketch draw an outer circle of 40mm and an inner circle of 30mm. Then change the attachment position for the 2nd sketch and change the Z to 50mm. Now that the 2nd sketch is above the 1st sketch (see attached image) you can select both sketches 1 and 2 and apply an additive loft. This seems to work fine however when you try to pad directly from either cone face this doesn't work. Even adding a new sketch to the face won't pad properly either. 

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git)
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/New Zealand (en_NZ)
Installed mods: 
  * Curves 0.6.23
  * Help 1.0.3
```

### Subproject(s) affected?

None

### Anything else?

[Cone.zip](https://github.com/user-attachments/files/16075448/Cone.zip)
[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-03:
Can confirm

maxwxyz, 2024-07-03:
@bgbsww sorry to bother but could this be because of TNP changes?

FL0WL0W, 2024-07-03:
I'd like to add that you could loft between 2 non parallel planes in FreeCAD 0.21, but this no longer works on the latest dev branch.

bgbsww, 2024-07-03:
It seems quite likely.

Curacaoflip, 2024-07-05:
Confirm, also the possibility to sketch on top of it (for example a closed top) by selecting the upper face, adding a sketch, drawing a circle which covers the top. Closing it and padding it won't work. In the current 0.21 version it works flawlessly, in the 0.22 build 38001 and earlier builds not at all.

chennes, 2024-07-19:
I can confirm the problem, and note that when making the pad, if you reverse it you get a small clue: the original cone shape's faces break, and the solid is no longer a solid.

chennes, 2024-07-22:
Here's a script that demonstrates the problem.
[PadFail.zip](https://github.com/user-attachments/files/16326839/PadFail.zip)

Nigelkdx200, 2024-08-04:
@chennes I just downloaded the latest dev version (released yesterday) and it appears to still have this issue so currently unresolved?

maxwxyz, 2024-08-04:
@bgbsww should this be fixed by #15577 ?

bgbsww, 2024-08-06:
Note that this loft example is currently returning five faces and not four - the base face is duplicated.  This is leading to a lot of problems.  It's on my radar to get to.

Nigelkdx200, 2024-08-06:
@bgbsww It's totally understandable that you're busy and will get to it whenever. It's just that a couple of weeks ago this issue was closed and then when I happened to test it I found that it still was an issue. So I enquired about it and it's been reopened as it's still an issue that will need resolving at some stage.

PR #15577 (2024-07-24): Toponaming: Fix #15138
Patch and test to Fix #15138.  Part of a pattern of places with TopoShape() that should potentially have a tag and be TopoShape(0) that will be investigated seperately.  Remaining question of if the parent volumes should be correct.
PR #15809 (2024-08-08): Toponaming:  fix loft mistake and complete test
Fix #15138
closed 2024-08-08 by chennes