FreeCAD/FreeCAD #15145: FreeCADGui.Selection.getSelectionEx()[0].SubObjects broken for sketches

Roy-043, 2024-07-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

`FreeCADGui.Selection.getSelectionEx()[0].SubObjects` is broken for sketches, it returns an empty tuple.

1. Create a sketch with some lines.
2. Close the sketch.
3. Select some of the edges from the sketch.
4. Paste `FreeCADGui.Selection.getSelectionEx()[0].SubObjects` in the Python console.

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git)
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-07-04:
Git bisect has https://github.com/FreeCAD/FreeCAD/commit/7a4bc95d47885e0d704dcad03d9db2db9c80864f as responsible which also relates to #15102 which Werner has given a guide to fixing https://github.com/FreeCAD/FreeCAD/issues/15102#issuecomment-2205587838 which hopefully is relevant to this also.

Syres916, 2024-07-04:
@bgbsww I'm really struggling with this one. Let me explain my findings, before https://github.com/FreeCAD/FreeCAD/commit/7a4bc95d47885e0d704dcad03d9db2db9c80864f#diff-b9dbc09c2c4b8417d6df9b1640fe0b24441cf026658dd5220d8917aa1ebcf356 `getSubObjects` function did not exist so at https://github.com/FreeCAD/FreeCAD/blob/main/src/Gui/SelectionObjectPyImp.cpp#L132 each `subObj` would be pushed back as `PyObj` was not a nullptr.

After the addition of the `getSubObjects` function in SketchObject the `pyObj` is now a nullptr so no subObjs are pushed back hence the report of an empty tuple. So what are we supposed to do, use a different name for the function such as `getSketchSubObjects` or is there something really obvious that I  am missing??

bgbsww, 2024-07-04:
Thanks for the sleuthing!  And a good reproducable.

~~ I think the really obvious thing is that SketchObject::getSubObject must not be right.  :( 

That's very much likely caused because of the difference of having an Exports property to look in in LS3 and not in main.

So this is a bug of incomplete code transfer into main for SketchObject, I suspect. ~~

Looking at this in main and LS3 in a debugger,  SketchObject::getSubObject gets called with a short element name  ( "Edge3" ) in LS3 and with a prefixed mapped name segment in main ( ";g3;SKT.Edge3")

So somewhere along this path we aren't doing the right thing:

(LS3) 
```
Sketcher::SketchObject::getSubObject SketchObject.cpp:10976
App::GeoFeatureGroupExtension::extensionGetSubObject GeoFeatureGroupExtension.cpp:544
App::OriginGroupExtension::extensionGetSubObject OriginGroupExtension.cpp:89
App::ExtensionContainer::queryExtension<…> ExtensionContainer.h:170
App::DocumentObject::getSubObject DocumentObject.cpp:886
Part::Feature::getSubObject PartFeature.cpp:368
PartDesign::Body::getSubObject Body.cpp:662
App::SubObjectT::getSubObject DocumentObserver.cpp:637
Gui::SelectionSingleton::format Selection.cpp:969
```
I've been noticing some places that we seem to have missed stuff in App, where all the fundamental element naming and map stuff is rock solid, but there are edge cases where code wasn't discovered.

All of that says that this is a definitive and deep (although probably really small) TNP bug, and so I've put it on my list unless someone gets to it first.

bgbsww, 2024-07-04:
PartDesign::Body::getSubObject is missing some code that looks highly likely.

Syres916, 2024-07-04:
@bgbsww thanks for the detailed feedback, the LS3 path is going to be very useful for me to look through, but, I've just done a clean compile along with PR15194 and neither a sketch created from Sketcher workbench (no Body) nor a Sketch created in PartDesign gives any edges from the command `FreeCADGui.Selection.getSelectionEx()[0].SubObjects` in the Python Console.

So while I'm not against having the PR merged, for me it doesn't actually fix @Roy-043's reported issue.

Is there a possibility you've some other code that you've been using to fix another bug in the build you are testing with?

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38027 (Git) + PR15194
Build type: Release
Branch: main
Hash: a3cc73ce15fb1e447af1cd7c4a57fae39d922dc2
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * Silk 0.1.5
  * FreeCAD-themes 2024.7.24
  * SketchArch
  * freecad.gears 1.2.0
  * Curves 0.6.23
  * CfdOF 1.25.11
  * fasteners 0.5.17
  * sheetmetal 0.4.1

```

bgbsww, 2024-07-04:
No, that was a clean build from main, the only other thing there is a change to a .ui file on Macro recording.   Testing again:

1) New file, Parameter Part
2) Create Sketch
3) Add Rectangle
4) Close Sketch
5) Select Line in Sketch  " Unnamed#Body.Edge1 [] (Sketch)" is in the Selection View
6)  
>>> FreeCADGui.Selection.getSelectionEx()[0].SubObjects
(<Edge object at 0x5ea45bc13b90>,)
>>> 

OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37928 (Git)
Build type: Debug
Branch: main
Hash: e7699b531eba4bee262f7214699f07b96f66ba7b
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)

Any chance you selected the sketch, and not edges in the sketch?

Syres916, 2024-07-04:
> Version: 0.22.0dev.37928 (Git) Build type: Debug Branch: main Hash: [e7699b5](https://github.com/FreeCAD/FreeCAD/commit/e7699b531eba4bee262f7214699f07b96f66ba7b) Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3 Locale: English/United States (en_US)
> 
> Any chance you selected the sketch, and not edges in the sketch?

Nope, I'm only selecting edges, remember I spent all last night tracking down the commit!! But your hash and commit are nearly 100 behind `main` Also note that this must work as a 'standalone' sketch i.e. not a PartDesign created one but from Sketcher Wb.

If needed I'll write a quick Python test and set it against the CI.

bgbsww, 2024-07-04:
Right you are.  Sorry, missed that rebase, didn't release that separate workarea had gotten that old.  So, Body::getSubObject is getting called with `"Sketch.;g68;SKT.Edge4"` which is pretty suspect; the Object name prepended to a mapped name is weird, and the parsing code will get thrown off by that first '.'.

Upstream of that:
```
Sketcher::SketchObject::getSubObject SketchObject.cpp:9499
App::GeoFeatureGroupExtension::extensionGetSubObject GeoFeatureGroupExtension.cpp:406
App::OriginGroupExtension::extensionGetSubObject OriginGroupExtension.cpp:89
App::DocumentObject::getSubObject DocumentObject.cpp:848
Part::Feature::getSubObject PartFeature.cpp:372
PartDesign::Body::getSubObject Body.cpp:535
operator() SelectionView.cpp:170
Gui::DockWnd::SelectionView::onSelectionChanged SelectionView.cpp:187
Gui::SelectionObserver::_onSelectionChanged Selection.cpp:152
Gui::SelectionSingleton::notify Selection.cpp:424
Gui::SelectionSingleton::addSelection Selection.cpp:964
Gui::SoFCUnifiedSelection::setSelection SoFCUnifiedSelection.cpp:672
Gui::SoFCUnifiedSelection::handleEvent SoFCUnifiedSelection.cpp:756
...
AddSelection is called with "Unnamed"  "Body"  and "Sketch.Edge4"   which seems reasonable.

getSelectionName inside SelectionView::onSelectionChanged()  seems mighty suspicious, and the code there is much different than LS3 which does not have that method.  This will be changes from #14699.  Restoring the changes will put the mapped names back into the selection view, which is unfortunate, but clearly this approach to removing them has a nasty side effect that wasn't clear because of the missing code.

Syres916, 2024-07-04:
OK, thanks, this feels like an absolute maze with trip hazards everywhere!! Anyway if you want here's the PD specific test and tomorrow I'll add a 'standalone' Sketcher one : https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:PDGui_Test

bgbsww, 2024-07-04:
Thank you!  I'll have that patch tweaked a little later to do the right thing.  The tests are very much appreciated.

bgbsww, 2024-07-05:
Well, !@#!.  This is the tip of an iceberg.  There are multiple places (8) in SoFCUnifiedSelection.cpp that are doing element name things.  Gonna take some work.

Also, this calls into question finding every place that Data::ComplexGeoData appears in LS3 code and hasn't been migrated to scope out potential needed changes.
```
brad@bradC940:~/git$ grep -r  "Data::ComplexGeoData" FreeCAD_linkstage/src | wc
    224    1028   27910
brad@bradC940:~/git$ grep -r  "Data::ComplexGeoData" FreeCADtoponaming/src | wc
     79     334    9351
brad@bradC940:~/git$ grep -r  "Data::" FreeCADtoponaming/src | wc
    568    2575   67556
brad@bradC940:~/git$ grep -r  "Data::" FreeCAD_linkstage/src | wc
    739    3488   89121
brad@bradC940:~/git$ grep -r  "Data::ComplexGeoData::has" FreeCAD_linkstage/src | wc
     15      51    1868
brad@bradC940:~/git$ grep -r  "Data::has" FreeCADtoponaming/src | wc
     13      59    1376
brad@bradC940:~/git$ grep -r  "Data::ComplexGeoData::find" FreeCAD_linkstage/src | wc
     44     240    5566
brad@bradC940:~/git$ grep -r  "Data::find" FreeCADtoponaming/src | wc
     24     135    2725
brad@bradC940:~/git$ 
```
I'm gonna stop here and go cry.  Then regroup and figure out a better assessment of how much is hiding here.

bgbsww, 2024-07-05:
Okay, I've assessed that list and added the necessary 136 places to check to my worklist.

Three of them happen to be in checkSubName in SketchObject, and that missing code does appear to solve this selection problem cleanly - and the names look the same in side by side debug runs with LS3.

PR #15194 (2024-07-17): Toponaming: Missing getSubObject code in PartDesign::Body
Fix #15145.
closed 2024-07-17 by chennes