FreeCAD/FreeCAD #15147: BIM: IfcOpenShell Update command on sandboxed packages fails with error

furgo16, 2024-07-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This is a follow-up to https://github.com/FreeCAD/FreeCAD/issues/14499. Unfortunately, the linked PR did not seem to fix the original issue (or a new one popped up):

Steps to reproduce:

1. Install latest daily snap package (`sudo snap install freecad --channel=edge`)
2. Start FreeCAD
3. Switch to BIM workbench
4. Run the the <kbd>Utils</kbd> > <kbd>IfcOpenShell update</kbd> command
5. Result: failure with the following error message in the Report panel
   ```python-traceback
   12:47:10  Unable to run pip. Please ensure pip is installed on your system.
   12:47:10  Running the Python command 'IFC_UpdateIOS' failed:
   Traceback (most recent call last):
     File "/snap/freecad/1065/usr/Mod/BIM/nativeifc/ifc_openshell.py", line 51, in Activated
       comp = self.compare_versions(avail, version)
     File "/snap/freecad/1065/usr/Mod/BIM/nativeifc/ifc_openshell.py", line 161, in compare_versions
       arr2 = [int(i) for i in arr2]
     File "/snap/freecad/1065/usr/Mod/BIM/nativeifc/ifc_openshell.py", line 161, in <listcomp>
       arr2 = [int(i) for i in arr2]
   
   invalid literal for int() with base 10: '0-c7830e9e5'
   ```

### Full version info

```shell
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38025 (Git) Snap 1065
Build type: Release
Branch: main
Hash: 04cc890fd0685bc4143de1a8000e92dce0580d28
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.5.16
  * Curves 0.6.39
  * ezydxf
  * ExplodedAssembly
  * BillOfMaterials 0.0.8.5
  * Ondsel-Lens 2024.5.6.01
  * Assembly4 0.50.13
```

### Subproject(s) affected?

BIM/Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2024-07-03:
Wow, hadn't seen that one... Not sure how we could convert 0-c7830e9e5 into something useful... Maybe just keep the 0?

yorikvanhavre, 2024-07-03:
In any case I'll add a default catch for all those pesky rev numbers

maxwxyz, 2024-07-07:
Same error on Windows:
[image]

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git)
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * CfdOF 1.27.3
  * Curves 0.6.40
  * dodo 1.0.1
  * fasteners 0.5.24
  * freecad.gears 1.2.0
  * OpenTheme 2024.7.3
  * OpticsWorkbench 1.0.18
  * Rocket 3.3.0
  * sheetmetal 0.4.21

```

furgo16, 2024-07-08:
> A similar but different error on Windows:

This seems to have a different traceback and grant its own issue.

yorikvanhavre, 2024-07-08:
This should be related indeed... In both cases pip is not run, but things did not stop there. I'll have a look!

yorikvanhavre, 2024-07-08:
This should fix the problems reported here... Please test!

PR #15292 (2024-07-11): BIM: more tests in IfcOpenShell tool
fixes #15147
commit 8a0604c4 (2024-07-11): BIM: more tests in IfcOpenShell tool - fixes #15147
closed 2024-07-11 by yorikvanhavre