FreeCAD/FreeCAD #15148: Core: Values of inverse properties get reset

pieterhijma, 2024-07-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

After adding and setting the value of a property of type `App::PropertyInverseLength`, `App::PropertyInverseArea`, or `App::PropertyInverseVolume`, the value is reset to zero when the editor of these properties is selected again.

To reproduce:
- Open a new file
- Enable the Part workbench
- Create a Cube
- Select the cube to make the properties visible in the Property view
- Right-click in the property view and select "Add property"
- Select `App::PropertyInverseLength` as type (or any of the other three)
- Name the variable and click ok
- Give the property a value, for example "5" (which becomes "5.00 1/mm")
- Click in the property value again (the editor): the value is reset to "0.00 1 / mm"

### Full version info

```shell
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38025 (Git)
Build type: DEBUG
Branch: main
Hash: 04cc890fd0685bc4143de1a8000e92dce0580d28
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk , OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * OpenTheme 2024.5.16
  * ondsel-lens 2024.7.2.01
```

### Subproject(s) affected?

Core

### Anything else?

Reported here: https://github.com/Ondsel-Development/FreeCAD/issues/58#issuecomment-2204805080

Although reported as being a problem of variable sets, it is independent of it as shown in the steps to reproduce.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2024-07-14:
Is this a regression ?

pieterhijma, 2024-07-14:
No, at least 0.21.2 has the same problem.

FEA-eng, 2025-07-17:
Still relevant

pieterhijma, 2026-02-13:
Maybe together? I don't have much time, unfortunately. I checked a little bit and in `QuantitySpinBox.cpp`, when you set a value, for example `5.00 1/mm`, `d->quantity` is set to this value. However, also `d->cached` should be set to that value but this doesn't happen because the value isn't validating. On a click, `d->quantity` is set to `d->cached` which is then still `0.00 1/mm`.

Ultimately, the problem seems to be that `5.00 1/mm` is not a valid expression (you can enter it in the expression editor and it will tell you so).  I believe that `5.00 1/mm` should be parsed as a valid expression and that would automatically resolve this issue. In `tests/src/App/ExpressionParser.cpp` it would be possible to add a test for this.

FlachyJoe, 2026-02-14:
Can't we simply compute spaces as multiplications ?