FreeCAD/FreeCAD #15151: Sketcher: Offset doesn't support B-splines

FEA-eng, 2024-07-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

As mentioned [here](https://github.com/FreeCAD/FreeCAD/issues/11289#issuecomment-1794616456), B-splines are one of the geometry types currently not supported by the Offset tool due to OCC issues.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git)
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kadet1090, 2024-10-06:
I think that it is actually not possible to implement in current sketcher. B-Splines offset is impossible to represent as another B-Spline. Same goes for Ellipses - ellipse offset is no longer an ellipse - so it is not possible to represent in the sketcher. What we need is computed geometry in Sketcher that is not real geometry, only defined in terms of some other stuff and impossible to modify by the user. Then we could utilize https://dev.opencascade.org/doc/refman/html/class_geom___offset_curve.html - which should be good enough. This would also help with things like patterns and/or text as we would not have to deal with additional complexity of having the extra geometry.

Shkolik, 2025-02-21:
> What we need is computed geometry in Sketcher that is not real geometry, only defined in terms of some other stuff and impossible to modify by the user.

And we actually have it -> external geometry. If we could make offset curve based on sketch geometry, but treat it like external it will be recalculated on each change and user will be able to reference it, but not directly modify.

pieterhijma, 2025-03-13:
I added this issue to be tracked for the OCCT Liaison position, but currently it is unclear what I can do. If anyone has an idea what I can do in my role, please let me know.

stefano2734, 2026-09-27:
In occt 8.1 will come new possibilities with extrema package. 
Perhaps an improvement on road to here wished B-spline-offset curve.
https://github.com/Open-Cascade-SAS/OCCT/pull/976

dpasukhi, 2026-09-27:
That draft already splitted ;) I will have a few PR related to that. (Integrate multiple `Extrema*_*` packages)
But I also need to inform, that transaction to new Extrema package will be "sequential". It will be not used by older algorithms for some period of time.
The same happening with `Math*_*` packages, which are replacement for math_* solvers.

Offset for BSpline is existed, in case of some bugs or unexpected behaviours - welcome to OCCT issues ;)

PaxTux, 2026-09-28:
I think this issue and [https://github.com/FreeCAD/FreeCAD/issues/24200](https://github.com/FreeCAD/FreeCAD/issues/24200) are duplicates?