FreeCAD/FreeCAD #15154: Sketcher - Adding a point to the origin force closes.

GAZ082, 2024-07-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Create a sketch, add a point to the origin, watch how the program force closes.

https://github.com/FreeCAD/FreeCAD/assets/100194/82accf78-5e7c-4ed4-9d09-473ea863ee74

### Full version info

```shell
OS: Arch Linux (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git) AppImage
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Curves.backup1713713759.0032098 0.6.29 (Disabled)
  * QuickMeasure 2022.10.28
  * ExtremeProDark 2.9.0
  * Curves.backup1693059780.2922277 0.6.13 (Disabled)
  * CfdOF.backup1710694906.6805208 1.24.9 (Disabled)
  * Assembly4.backup1696800837.713216 0.50.2 (Disabled)
  * Curves.backup1714093860.7275996 0.6.34 (Disabled)
  * OpenTheme 2024.5.16
  * CfdOF.backup1699129791.898633 1.24.7 (Disabled)
  * Assembly4.backup1710902627.637761 0.50.9 (Disabled)
  * Defeaturing.backup1710696629.0288966 1.2.2 (Disabled)
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-07-04:
Git bisect gives https://github.com/FreeCAD/FreeCAD/commit/7a4bc95d47885e0d704dcad03d9db2db9c80864f as responsible, Werner has given advice on fix in https://github.com/FreeCAD/FreeCAD/issues/15102#issuecomment-2205587838

Syres916, 2024-07-04:
This was my (probably poor) effort https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:Fix_ElementName_is_Null it stops the crash and passes CI tests but I've not done any serious real world regression testing.

Syres916, 2024-07-07:
As I cannot get the error handling to deal with the vertex at origin without breaking multiple Unit Tests someone else will have to try. LinkStage3 doesn't need any of this specific handling which tells me there's some code that hasn't been brought across.

GAZ082, 2024-07-07:
This bug has meme potential. We cannot ship 1.0 with it :D

mfraser, 2024-07-12:
I'm seeing this when adding a point anywhere

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7e77fe442520]
#1  /usr/lib/freecad-daily-python3/lib/libFreeCADApp.so(+0x358fb2) [0x7e7800f58fb2]
#2  0x7e7800f54177 in Data::ElementMap::addName(Data::MappedName&, Data::IndexedName const&, QVector<App::StringIDRef> const&, bool, Data::IndexedName*) from /usr/lib/freecad-daily-python3/lib/libFreeCADApp.so+0x117
#3  0x7e7800f55f24 in Data::ElementMap::setElementName(Data::IndexedName const&, Data::MappedName const&, long, QVector<App::StringIDRef> const*, bool) from /usr/lib/freecad-daily-python3/lib/libFreeCADApp.so+0x2c4
#4  0x7e77d84953e7 in Sketcher::SketchObject::buildShape() from /usr/lib/freecad-daily-python3/lib/Sketcher.so+0x267
#5  0x7e77d8496a5c in Sketcher::SketchObject::execute() from /usr/lib/freecad-daily-python3/lib/Sketcher.so+0xdc
#6  0x7e7800dc8ffa in App::DocumentObject::recompute() from /usr/lib/freecad-daily-python3/lib/libFreeCADApp.so+0x3a
#7  0x7e77e0e54cbd in Part::Feature::recompute() from /usr/lib/freecad-daily-python3/lib/Part.so+0xd
#8  0x7e7800d9936d in App::Document::_recomputeFeature(App::DocumentObject*) from /usr/lib/freecad-daily-python3/lib/libFreeCADApp.so+0x34d
#9  0x7e7800d9655d in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /usr/lib/freecad-daily-python3/lib/libFreeCADApp.so+0x5bd
#10  0x7e7800e17a81 in App::DocumentPy::recompute(_object*) from /usr/lib/freecad-daily-python3/lib/libFreeCADApp.so+0x2f1
#11  0x7e7800e0e26c in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /usr/lib/freecad-daily-python3/lib/libFreeCADApp.so+0x2c
#12  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x128018) [0x7e7800328018]
#13  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyObject_MakeTpCall+0x8c) [0x7e78002dfaec]
#14  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(_PyEval_EvalFrameDefault+0x9dbc) [0x7e780027ba2c]
#15  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x1c23cf) [0x7e78003c23cf]
#16  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyEval_EvalCode+0xbe) [0x7e78003bd3fe]
#17  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(+0x20c4ed) [0x7e780040c4ed]
#18  /lib/x86_64-linux-gnu/libpython3.10.so.1.0(PyRun_StringFlags+0x79) [0x7e780040d519]
#19  0x7e78008ea30c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /usr/lib/freecad-daily-python3/lib/libFreeCADBase.so+0x6c
#20  0x7e78016cf6d7 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x97
#21  0x7e78016cf877 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0xe7
#22  0x7e78016d3827 in Gui::Command::updateActive() from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x47
#23  0x7e77a1f79aa0 in SketcherGui::tryAutoRecompute(Sketcher::SketchObject*, bool&) from /usr/lib/freecad-daily-python3/lib/SketcherGui.so+0xd0
#24  0x7e77a1f79b26 in SketcherGui::tryAutoRecomputeIfNotSolve(Sketcher::SketchObject*) from /usr/lib/freecad-daily-python3/lib/SketcherGui.so+0x26
#25  /usr/lib/freecad-daily-python3/lib/SketcherGui.so(+0x2109fa) [0x7e77a20109fa]
#26  /usr/lib/freecad-daily-python3/lib/SketcherGui.so(+0x1f6e50) [0x7e77a1ff6e50]
#27  /usr/lib/freecad-daily-python3/lib/SketcherGui.so(+0x1f64c9) [0x7e77a1ff64c9]
#28  0x7e77a20f80ab in SketcherGui::ViewProviderSketch::mouseButtonPressed(int, bool, SbVec2s const&, Gui::View3DInventorViewer const*) from /usr/lib/freecad-daily-python3/lib/SketcherGui.so+0x22b
#29  0x7e78019e6a2e in Gui::ViewProvider::eventCallback(void*, SoEventCallback*) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x15e
#30  0x7e77fde4d357 in SoEventCallback::handleEvent(SoHandleEventAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0xb7
#31  0x7e77fdc76065 in SoAction::traverse(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x65
#32  0x7e77fde10868 in SoChildList::traverse(SoAction*, int, int) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x168
#33  0x7e77fde57090 in SoGroup::doAction(SoAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x50
#34  0x7e77fde6bd7c in SoSeparator::doAction(SoAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x2c
#35  0x7e77fdc76065 in SoAction::traverse(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x65
#36  0x7e77fde10868 in SoChildList::traverse(SoAction*, int, int) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x168
#37  0x7e77fde57090 in SoGroup::doAction(SoAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x50
#38  0x7e77fde6bd7c in SoSeparator::doAction(SoAction*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x2c
#39  0x7e77fdc76065 in SoAction::traverse(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x65
#40  0x7e77fdc7fabb in SoHandleEventAction::beginTraversal(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0xab
#41  0x7e77fdc766b8 in SoAction::apply(SoNode*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0xc8
#42  0x7e77fde25809 in SoEventManager::actuallyProcessEvent(SoEvent const*) from /lib/x86_64-linux-gnu/libCoin.so.80c+0x79
#43  0x7e780197de62 in SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x22
[... cut ...]
#68  0x7e77feeb875b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x12b
#69  0x7e77feec0cf4 in QCoreApplication::exec() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x94
#70  0x7e780162236a in Gui::Application::runApplication() from /usr/lib/freecad-daily-python3/lib/libFreeCADGui.so+0x8ea
#71  freecad-daily(+0x6d54) [0x62b1e988bd54]
#72  /lib/x86_64-linux-gnu/libc.so.6(+0x29d90) [0x7e77fe429d90]
#73  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7e77fe429e40]
#74  freecad-daily(+0x7ea5) [0x62b1e988cea5]
```

adrianinsaval, 2024-07-13:
For me it does not crash when adding the point but it does when exiting the sketch after creation

GAZ082, 2024-07-14:
I confirm, putting a point anywhere in the sketch, force closes.

Larry0815, 2024-07-14:
Same when using FreeCAD_weekly-builds-38127-conda-Windows-x86_64-py311.
Very annoying bug ...

closed 2024-07-17 by WandererFan

neoh4x0r, 2024-07-18:
> Git bisect gives 
https://github.com/FreeCAD/FreeCAD/commit/7a4bc95d47885e0d704dcad03d9db2db9c80864f 
as responsible, Werner has given advice on fix in 
https://github.com/FreeCAD/FreeCAD/issues/15102#issuecomment-2205587838

My first experience of this issue was in 
dfe0c5d4ee4841f11e62da1104e322e95834aba2 and the 
problem did not happen in 
f310d2bd67e901c3cfdf57580770076ca0aff26b

I did a bisect with the following:

```
$ git bisect start
$ git bisect bad dfe0c5d4ee4841f11e62da1104e322e95834aba2
$ git bisect good f310d2bd67e901c3cfdf57580770076ca0aff26b
```

It says commit 1e00c0b9d1830a6156ea95f6fbcb988ed13e24cc ''LineAttributes' is at fault.

commit c8046b1c (2024-07-17): Sketcher: Fix crash on recompute with points

Fixes #15154.