FreeCAD/FreeCAD #15158: sketch: external geometry from error

Vincenzoferrara, 2024-07-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

after updating to the latest version of freecad (beta) all sketches with external geometry give me an error.

I tried to create another file with a simple cube (with part design) and then use the external geometry, and here too it gives me the same error.

if you remove the external geometries everything works.

the problem only happens in the latest beta version.

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git) AppImage
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Italian/Italy (it_IT)
Installed mods: 
  * IconThemes
  * btl 0.9.9
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.5.16
  * fasteners 0.5.24
  * TabBar
  * Assembly4 0.50.13
```

### Subproject(s) affected?

Sketcher

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-07-03:
This is a duplicate of https://github.com/FreeCAD/FreeCAD/issues/15117 and Werner has already created a fix as https://github.com/FreeCAD/FreeCAD/issues/15117#issuecomment-2203411302

Vincenzoferrara, 2024-07-03:
I had searched if the problem already existed, but I didn't notice this