FreeCAD/FreeCAD #15159: Gui: Dialog for adding properties allows property types that give exceptions

pieterhijma, 2024-07-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The Add Property dialog in the property view allows users to select property types that lead to exceptions:
- App::Property
- App::PropertyComplexGeoData
- ExpressionContainer
- ExpressionEngine
- Geometry
- IntegerSet
- LinkBase
- LinkListBase
- Lists
- XLinkContainer
- Sketcher::PropertyConstraintList

I'm inclined to remove those properties from the VarSet dialog, but perhaps it is better to remove them from the standard dialog to add properties as well?

### Full version info

```shell
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38025 (Git)
Build type: DEBUG
Branch: varset-property-types
Hash: 04cc890fd0685bc4143de1a8000e92dce0580d28
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk , OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * OpenTheme 2024.5.16
  * ondsel-lens 2024.7.2.01
```

### Subproject(s) affected?

Core

### Anything else?

Related to https://github.com/FreeCAD/FreeCAD/issues/15156

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2024-07-03:
> I'm inclined to remove those properties from the VarSet dialog, but perhaps it is better to remove them from the standard dialog to add properties as well?

It would probably be best for the user to keep both dialogs in sync.

pieterhijma, 2024-07-03:
Yes, that makes this issue related to https://github.com/FreeCAD/FreeCAD/issues/14633.

pieterhijma, 2024-07-03:
Yes, I agree.

krushia, 2024-07-06:
Hopefully not too late... some fixes for your list:

`App::PropertyExpressionEngine` Can be created just fine, and I'm currently using that one in a VarSet to workaround a Spreadsheet bug,
`App::PropertyIntegerSet` Doesn't throw an error on creation, though I haven't tried doing anything with it yet.

pieterhijma, 2024-07-09:
@wwmayer's PR filters out properties that are declared to be "abstract" and should not be instantiated directly. I think this makes sense. It also means that `App::PropertyExpressionEngine` and `App::PropertyIntegerSet` remain in the list.

PR #15191 (2024-07-10): Core: In dialog to add property only list types that can be instantiated
Fixes #15159: Dialog for adding properties allows property types that give exceptions
commit 5b9b611f (2024-07-10): Core: In dialog to add property only list types that can be instantiated

Fixes #15159: Dialog for adding properties allows property types that give exceptions
commit a8bbeac6 (2024-07-15): Core: In dialog to add property only list types that can be instantiated

Fixes #15159: Dialog for adding properties allows property types that give exceptions
closed 2024-07-10 by wwmayer