FreeCAD/FreeCAD #15162: Part Design Migrate tool seems to be outdated and unnecessary

FEA-eng, 2024-07-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

It seems that it doesn't make sense to keep the [Part Design Migrate](https://wiki.freecad.org/PartDesign_Migrate) tool which was supposed to be used to convert models from 0.16 or below to 0.17. We are getting close to 0.22/1.0 now so this tool seems to be highly outdated and won't help with the migration from 0.21.2 to 1.0, for instance. Thus, I would suggest removing it (at least from GUI) whenever possible (unless I'm wrong and this tool can still be useful but I doubt that's the case).

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git)
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Jookia, 2024-07-04:
I would +1 dropping this as it's partially broken regarding undo.

closed 2024-07-05 by wwmayer

Jookia, 2024-07-05:
> Closed #15162 as completed via #15196.

Could we keep this open for actual removal from the codebase?

FEA-eng, 2024-07-05:
> Could we keep this open for actual removal from the codebase?

I can create a separate issue suggesting such a future clean up. We have some other deprecated tools that were only removed from GUI and not from the code (so that they could be fixed at some point). Mostly in the FEM workbench. But the Migrate tool is unlikely to ever be useful.

Jookia, 2024-07-05:
That would be good!

PR #15196 (2024-07-05): PD: Remove the Migrate tool from GUI
Fixes #15162

This tool has no toolbar button so the PR only removes it from the PD menu.