FreeCAD/FreeCAD #15163: QtStyle is not reset after switching back from OpenTheme to Classic

FEA-eng, 2024-07-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Initially, I reported this in the OpenTheme repo: https://github.com/obelisk79/OpenTheme/issues/84 but it seems to be an issue with FreeCAD itself. Apparently, QtStyle is not reset after switching back to the Classic theme.

Classic (before applying OpenLight):

[image]

OpenLight:

[image]

Classic (after switching back from OpenLight and restarting FreeCAD):

[image]

Manually removing the QtStyle key removes those remaining changes and brings back the default appearance.

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git)
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-07-03:
@FEA-eng just a suggestion, possibly add the Python code to remove the key to https://github.com/FreeCAD/FreeCAD/blob/main/src/Gui/PreferencePacks/Classic/post.FCMacro
Not sure if it'll need another restart though.

FEA-eng, 2024-07-03:
@Syres916 Thank you, it's working.

PR #15168 (2024-07-08): Remove QtStyle when applying Classic theme
Fixes #15163
closed 2024-07-08 by adrianinsaval