FreeCAD/FreeCAD #15164: Part Design WB > Aditive Pipe > Remove Edge Button does not work.

leoheck, 2024-07-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Part Design WB > Aditive Pipe > Remove Edge Button does not work.

However `Right click > Remove` works

[image]

### Full version info

```shell
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git) AppImage
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * POV-Ray-Rendering
  * Assembly4 0.50.12
  * InventorLoader 1.4.0
  * ose-piping
  * OpenTheme 2024.6.22
  * SheetMetal 0.4.20
  * fasteners 0.5.24
  * asm4_disable_version_check_popup
  * Render 2024.6.5
  * OpticsWorkbench 1.0.18
  * lattice2 1.0.0
  * kicadStepUpMod
```
```

### Subproject(s) affected?

PartDesign

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-04:
For me it works the same as `Add edge` works. Click the button, select the edge in the 3D view.
I agree, the UI workflow is bad here. It should also be possible to do this like in the Fillet task dialog. Toggle select and preview mode.

leoheck, 2024-07-04:
Ah, you have to click the button first?

> It should also be possible to do this like in the Fillet task dialog. Toggle select and preview mode.
That is what I would like to suggest

Actually, this kind of operation should be always the same everywhere, I guess.

maxwxyz, 2024-07-04:
yes, it's different in any WB and dialog, some request a preselection of the line then click the button, other the other way around.
@FreeCAD/design-working-group is there an effort to unify task dialogs and feature selection?

kadet1090, 2024-07-04:
Like this? https://github.com/FreeCAD/FreeCAD/issues/14432

maxwxyz, 2024-07-04:
yes, I knew that it was mentioned somewhere. We could add placement of "OK, Apply, Cancel" buttons as on some dialogs they are on the top and on some on the bottom, hidden on small screens where you need to scroll the dialog (https://github.com/FreeCAD/FreeCAD/issues/13120).
@leoheck will close this one then.

closed 2024-07-04 by maxwxyz