FreeCAD/FreeCAD #15170: Material: Changing the ShapeAppearance of the whole object does not remove face appearance override

Roy-043, 2024-07-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

In V0.21 if you change the color of a face and then change the ShapeColor of the whole object, the entire object receives the new ShapeColor. The face override is removed.

In V0.22 this is no longer possible. The face override is kept and only the 1st face of the object receives the new ShapeAppearance.

Steps:
1. Create a Box.
2. Change the ShapeAppearance of "Face3" (Tree view context menu > Set appearance per face > Custom Appearance > Diffuse color)
3. Change the DiffuseColor of the ShapeAppearance of the whole object, making sure to select a unique color.
4. Result: "Face3" has retained its override color, "Face1" has the new color, all other faces have not been changed.

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git)
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

davesrocketshop, 2024-07-15:
I'm not able to reproduce in either a repository build or the latest weekly build. @Roy-043 could you please check if you're still seeing it?

```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38231 (Git)
Build type: Release
Branch: main
Hash: 29889bfd9a7ba0f0496f2e2699d454ea5fb5a05a
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/Canada (en_CA)
Installed mods: 
  * CfdOF 1.25.11
  * MaterialTest
  * OpenTheme 2024.5.3
  * Render 2024.1.23
  * Rocket 4.0.0
  * sheetmetal 0.4.13

```

Roy-043, 2024-07-17:
In step 3 I change the Diffuse color in the Property View panel. I even see evidence of the issue in the Python console:
`FreeCADGui.getDocument('Unnamed').getObject('Box').ShapeAppearance = (App.Material(DiffuseColor=(1.00,0.67,0.00),AmbientColor=(0.33,0.33,0.33),SpecularColor=(0.53,0.53,0.53),EmissiveColor=(0.00,0.00,0.00),Shininess=(0.90),Transparency=(0.00),), App.Material(DiffuseColor=(0.33,1.00,1.00),AmbientColor=(0.33,0.33,0.33),SpecularColor=(0.53,0.53,0.53),EmissiveColor=(0.00,0.00,0.00),Shininess=(0.90),Transparency=(0.00),), App.Material(DiffuseColor=(1.00,0.33,1.00),AmbientColor=(0.33,0.33,0.33),SpecularColor=(0.53,0.53,0.53),EmissiveColor=(0.00,0.00,0.00),Shininess=(0.90),Transparency=(0.00),), App.Material(DiffuseColor=(0.33,1.00,1.00),AmbientColor=(0.33,0.33,0.33),SpecularColor=(0.53,0.53,0.53),EmissiveColor=(0.00,0.00,0.00),Shininess=(0.90),Transparency=(0.00),), App.Material(DiffuseColor=(0.33,1.00,1.00),AmbientColor=(0.33,0.33,0.33),SpecularColor=(0.53,0.53,0.53),EmissiveColor=(0.00,0.00,0.00),Shininess=(0.90),Transparency=(0.00),), App.Material(DiffuseColor=(0.33,1.00,1.00),AmbientColor=(0.33,0.33,0.33),SpecularColor=(0.53,0.53,0.53),EmissiveColor=(0.00,0.00,0.00),Shininess=(0.90),Transparency=(0.00),), )`

```
Version:
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38231 (Git)
Build type: Release
Branch: main
Hash: 29889bfd9a7ba0f0496f2e2699d454ea5fb5a05a
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

davesrocketshop, 2024-07-17:
Ok, got it. I can reproduce. Thanks

PR #15632 (2024-07-29): Materials: Reset per face appearance when using property editor
In v0.21 changing the color using the property editor reset the per face colors. This recreates that behavior when setting the appearance.

fixes #15170
commit 695fc55c (2024-07-29): Materials: Reset per face when using property editor

In v0.21 changin the color using the propert editor reset the per face
colors. This recreates that behavior when settin the appearance.

fixes #15170
closed 2024-07-29 by chennes