FreeCAD/FreeCAD #15173: Core: Missing interpreter lock

pieterhijma, 2024-07-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

For context, see https://github.com/FreeCAD/FreeCAD/issues/14957.

This issue is found with the same environment.

To reproduce:
- Open a document
- Create a cube
- Close FreeCAD and discard the document.

```
Fatal Python error: _PyInterpreterState_GET: the function must be called with the GIL held, after Python initialization and before Python finalization, but the GIL is released (the current Python thread state is NULL)
Python runtime state: initialized

Current thread 0x00007f070df4aa40 (most recent call first):
  <no Python frame>

Extension modules: FreeCAD, FreeCADGui, pivy._coin, shiboken6.Shiboken, PySide6.QtCore, StartGui, Materials, Part, Measure, MeasureGui, PySide6.QtGui, PySide6.QtWidgets, PySide6.QtNetwork, _cffi_backend, Sketcher, _PartDesign, MatGui, PartGui, SketcherGui, PartDesignGui (total: 20)
Aborted (core dumped)
```

Full stack trace:

```
#0  __pthread_kill_implementation (threadid=<optimized out>, signo=signo@entry=6, no_tid=no_tid@entry=0) at pthread_kill.c:44
#1  0x00007ffff08a8eb3 in __pthread_kill_internal (threadid=<optimized out>, signo=6) at pthread_kill.c:78
#2  0x00007ffff0850a30 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#3  0x00007ffff08384c3 in __GI_abort () at abort.c:79
#4  0x00007ffff32a9b6f in fatal_error_exit (status=status@entry=-1) at Python/pylifecycle.c:2737
#5  0x00007ffff32cc28a in fatal_error (fd=2, header=header@entry=1, prefix=prefix@entry=0x7ffff35fe670 <__func__.43.lto_priv.0> "_PyInterpreterState_GET", msg=msg@entry=0x7ffff3611fb8 "the function must be called with the GIL held, after Python initialization and before Python finalization, but the GIL is released (the current Python thread state is NULL)", status=status@entry=-1) at Python/pylifecycle.c:2918
#6  0x00007ffff32cc302 in _Py_FatalErrorFunc (func=func@entry=0x7ffff35fe670 <__func__.43.lto_priv.0> "_PyInterpreterState_GET", msg=msg@entry=0x7ffff3611fb8 "the function must be called with the GIL held, after Python initialization and before Python finalization, but the GIL is released (the current Python thread state is NULL)") at Python/pylifecycle.c:2934
#7  0x00007ffff32cd240 in _Py_EnsureFuncTstateNotNULL (func=0x7ffff35fe670 <__func__.43.lto_priv.0> "_PyInterpreterState_GET", tstate=<optimized out>) at ./Include/internal/pycore_pystate.h:108
#8  _PyInterpreterState_GET () at ./Include/internal/pycore_pystate.h:131
#9  0x00007ffff32cf73e in _Py_DECREF_DecRefTotal () at Objects/object.c:224
#10 0x00007ffff405d48d in Py_DECREF (filename=0x7ffff424ad4e "/usr/include/python3.12d/object.h", lineno=798, op=0x55555b0d65c8) at /usr/include/python3.12d/object.h:688
#11 0x00007ffff405d3dc in Py_XDECREF (op=0x55555b0d65c8) at /usr/include/python3.12d/object.h:798
#12 0x00007ffff405d3a5 in Py::_XDECREF (op=0x55555b0d65c8) at /home/pieter/current/ondsel/freecad-main/src/CXX/IndirectPythonInterface.cxx:559
#13 0x00007ffff41c437d in Py::SmartPtr::release (this=0x555559a994d8) at /home/pieter/current/ondsel/freecad-main/src/Base/SmartPtrPy.cpp:45
#14 0x00007ffff41c4583 in Py::SmartPtr::~SmartPtr (this=0x555559a994d8) at /home/pieter/current/ondsel/freecad-main/src/Base/SmartPtrPy.cpp:92
#15 0x00007ffff4ec00a5 in App::Extension::~Extension (this=0x555559a994c8) at /home/pieter/current/ondsel/freecad-main/src/App/Extension.cpp:68
#16 0x00007ffff4ec1c65 in App::DocumentObjectExtension::~DocumentObjectExtension (this=0x555559a994c8) at /home/pieter/current/ondsel/freecad-main/src/App/DocumentObjectExtension.cpp:40
#17 0x00007fffb11f9944 in Part::AttachExtension::~AttachExtension (this=0x555559a994c8) at /home/pieter/current/ondsel/freecad-main/src/Mod/Part/App/AttachExtension.cpp:158
#18 0x00007fffb11e8000 in Part::Primitive::~Primitive (this=0x555559a98c00) at /home/pieter/current/ondsel/freecad-main/src/Mod/Part/App/PrimitiveFeature.cpp:79
#19 0x00007fffb114ec6b in Part::Box::~Box (this=0x555559a98c00) at /home/pieter/current/ondsel/freecad-main/src/Mod/Part/App/FeaturePartBox.h:34
#20 0x00007fffb114ec99 in Part::Box::~Box (this=0x555559a98c00) at /home/pieter/current/ondsel/freecad-main/src/Mod/Part/App/FeaturePartBox.h:34
#21 0x00007ffff4d5ee9a in App::DocumentP::clearDocument (this=0x5555599697c0) at /home/pieter/current/ondsel/freecad-main/src/App/private/DocumentP.h:126
#22 0x00007ffff4d41a39 in App::Document::~Document (this=0x5555593f15a0) at /home/pieter/current/ondsel/freecad-main/src/App/Document.cpp:898
#23 0x00007ffff4d41fc9 in App::Document::~Document (this=0x5555593f15a0) at /home/pieter/current/ondsel/freecad-main/src/App/Document.cpp:883
#24 0x00007ffff51f5dbc in std::default_delete<App::Document>::operator() (this=0x7fffffffa7d0, __ptr=0x5555593f15a0) at /usr/bin/../lib64/gcc/x86_64-pc-linux-gnu/14.1.1/../../../../include/c++/14.1.1/bits/unique_ptr.h:93
#25 0x00007ffff51dc5f3 in std::unique_ptr<App::Document, std::default_delete<App::Document> >::~unique_ptr (this=0x7fffffffa7d0) at /usr/bin/../lib64/gcc/x86_64-pc-linux-gnu/14.1.1/../../../../include/c++/14.1.1/bits/unique_ptr.h:398
#26 0x00007ffff51be773 in App::Application::closeDocument (this=0x55555576b9f0, name=0x555559c1b770 "Unnamed") at /home/pieter/current/ondsel/freecad-main/src/App/Application.cpp:564
#27 0x00007ffff51be901 in App::Application::closeAllDocuments (this=0x55555576b9f0) at /home/pieter/current/ondsel/freecad-main/src/App/Application.cpp:571
#28 0x00007ffff6655594 in Gui::Application::tryClose (this=0x7fffffffc250, e=0x7fffffffb2f0) at /home/pieter/current/ondsel/freecad-main/src/Gui/Application.cpp:1342
#29 0x00007ffff70bdaca in Gui::MainWindow::closeEvent (this=0x7fffffffc220, e=0x7fffffffb2f0) at /home/pieter/current/ondsel/freecad-main/src/Gui/MainWindow.cpp:1495
#30 0x00007ffff1f4b9ac in QWidget::event (this=0x7fffffffc220, event=0x7fffffffb2f0) at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qwidget.cpp:9200
#31 0x00007ffff70bace8 in Gui::MainWindow::event (this=0x7fffffffc220, e=0x7fffffffb2f0) at /home/pieter/current/ondsel/freecad-main/src/Gui/MainWindow.cpp:1104
#32 0x00007ffff1efc51c in QApplicationPrivate::notify_helper (this=<optimized out>, receiver=0x7fffffffc220, e=0x7fffffffb2f0) at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qapplication.cpp:3287
#33 0x00007ffff683ea2e in Gui::GUIApplication::notify (this=0x7fffffffc458, receiver=0x7fffffffc220, event=0x7fffffffb2f0) at /home/pieter/current/ondsel/freecad-main/src/Gui/GuiApplication.cpp:83
#34 0x00007ffff0f44d28 in QCoreApplication::notifyInternal2 (receiver=0x7fffffffc220, event=0x7fffffffb2f0) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qcoreapplication.cpp:1142
#35 0x00007ffff0f44d7d in QCoreApplication::sendSpontaneousEvent (receiver=<optimized out>, event=<optimized out>) at /usr/src/debug/qt6-base/qtbase/src/corelib/kernel/qcoreapplication.cpp:1597
#36 0x00007ffff1f468ee in QWidgetPrivate::handleClose (this=0x555555bf63e0, mode=QWidgetPrivate::CloseWithSpontaneousEvent) at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qwidget.cpp:8619
#37 0x00007ffff1f63067 in QWidgetWindow::closeEvent (this=<optimized out>, event=0x7fffffffb870) at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qwidgetwindow.cpp:832
#38 0x00007ffff15df4e0 in QWindow::event (this=0x555559a91a80, ev=0x7fffffffb870) at /usr/src/debug/qt6-base/qtbase/src/gui/kernel/qwindow.cpp:2587
#39 0x00007ffff1efc51c in QApplicationPrivate::notify_helper (this=<optimized out>, receiver=0x555559a91a80, e=0x7fffffffb870) at /usr/src/debug/qt6-base/qtbase/src/widgets/kernel/qapplication.cpp:3287
#40 0x00007ffff683ea2e in Gui::GUIApplication::notify (this=0x7fffffffc458, receiver=0x555559a91a80, event=0x7fffffffb870) at /home/pieter/current/ondsel/freecad-main/src/Gui/GuiApplication.cpp:83
[... 28 lines cut ...]
```

### Full version info

```shell
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38025 (Git)
Build type: DEBUG
Branch: main
Hash: 04cc890fd0685bc4143de1a8000e92dce0580d28
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk , OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * OpenTheme 2024.5.16
  * ondsel-lens 2024.7.2.01
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-04:
Isn't this fixed by  #14983 ?

Jookia, 2024-07-04:
The GIL should be held before doing ANY Python API calls, it might be wiser to just stop micromanaging it and move it to the Python interface.

PR #15174 (2024-07-14): Core: Add a missing interpeter lock
Closes #15173
closed 2024-07-14 by chennes