FreeCAD/FreeCAD #15176: An issue with the Addon manager when clicking Python button

leoheck, 2024-07-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

An issue with the Addon manager when clicking Python button

```
00:00:58  Traceback (most recent call last):
00:00:58    File "/tmp/.mount_FreeCABwnl0s/usr/Mod/AddonManager/AddonManager.py", line 736, in show_python_updates_dialog
00:00:58      self.manage_python_packages_dialog.show()
00:00:58      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
00:00:58  AttributeError: 'NoneType' object has no attribute 'show'
00:01:06  Traceback (most recent call last):
00:01:06    File "/tmp/.mount_FreeCABwnl0s/usr/Mod/AddonManager/AddonManager.py", line 736, in show_python_updates_dialog
00:01:06      self.manage_python_packages_dialog.show()
00:01:06      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
00:01:06  AttributeError: 'NoneType' object has no attribute 'show'
```

[image]

### Full version info

```shell
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git) AppImage
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * POV-Ray-Rendering
  * Assembly4 0.50.12
  * InventorLoader 1.4.0
  * ose-piping
  * OpenTheme 2024.7.3
  * SheetMetal 0.4.20
  * fasteners 0.5.24
  * asm4_disable_version_check_popup
  * Render 2024.6.5
  * OpticsWorkbench 1.0.18
  * lattice2 1.0.0
  * kicadStepUpMod
```
```

### Subproject(s) affected?

Addon Manager

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-04:
Can confirm. Please don't add code blocks for the about info, they are inserted automatically. If possible, try do use a more specific name for your issues.
@chennes FYI

leoheck, 2024-07-04:
> Can confirm. Please don't add code blocks for the about info, they are inserted automatically. If possible, try do use a more specific name for your issues. @chennes FYI

I did not do that. Ah, I know what happened. I closed the multiline code block twice and GitHub broke.

chennes, 2024-07-13:
Fixed by 0800eea8d00046aba42ac1cf51d0180b3960d6a9

closed 2024-07-13 by chennes