FreeCAD/FreeCAD #15178: Start: using "New file" icons get different results depending on the previous new file created

MiqCG, 2024-07-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I casually encountered this issue #13991 and while checking if it was already detected and making several tests, I have found an inconsistent behaviour when creating different types of files from Start page. See attached video and/or

1. open FreeCAD with Start page enabled.
2. when clicking "Parametric Part" for the first time, an empty 3D View appears. Expected result.
3. if closed and then clicking in "2D Draft", a grid appears on the same plane. Expected result.
4. if closed and then click in "Parametric Part" again, the new file appears but with a Draft grid in perspective view. Inconsistent result.
5. close again and create a new "2D Draft" and the grid (sometime) fails to appear.
6. if then a "Parametric Part" is created again closed and then re-created consecutively, the grid doesn't appear.

With "Assembly" creation, a Draft grid also appears if created after "2D Draft" of "BIM/Architecture". "Empty file" gets contaminated too.

Tested only with Classic theme (just in case of #15026 ) with recently deleted .local/share .cache and .config folders. Same problem in revisions 37898 and 37928.

I have searched in Github with [is:issue start new body 2D] and [is:issue start new parametric 2D] without relevant results.

### Full version info

```shell
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git) AppImage
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Installed mods: 
  * Movie 2024.2.15
  * FreeCAD-motion-workbench-master
  * dodo 1.0.1
  * PieMenu 1.2.7
  * fasteners 0.5.24
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

https://github.com/FreeCAD/FreeCAD/assets/18265355/53263964-64fd-40a9-9d63-90d8960224fc

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

furgo16, 2024-07-04:
I cannot reproduce on:

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38025 (Git) Snap 1065
Build type: Release
Branch: main
Hash: 04cc890fd0685bc4143de1a8000e92dce0580d28
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.6.22
  * Curves 0.6.39
  * ezydxf
  * Ondsel-Lens 2024.5.6.01
  * Assembly4 0.50.13
```

1. Do you have `Always show the grid` enabled in Edit > Preferences... > Draft > Grid and snapping?
1. One thing that I can reproduce is the strangely aligned grid default on the Draft workbench. It appears aligned to the isometric view, and at an angle: [image]

FEA-eng, 2024-07-04:
> 2\. One thing that I can reproduce is the strangely aligned grid default on the Draft workbench. It appears aligned to the isometric view, and at an angle:

It's a known issue with the default setting _Automatic_. Switching it to _Top_, _Front_ or _Side_ eliminates the problem.

furgo16, 2024-07-04:
> It's a known issue with the default setting Automatic. Switching it to Top, Front or Side eliminates the problem.

Ah, I can see where to set the default now, thanks!
[image]

However, and not to derail too much from the original report, I think the issue is less about the grid being aligned to the isometric view, but that the grid appears to be rotated slightly when the isometric view is chosen. If I click on the Working plane selector, then the grid is rotated to the expected position, even before clicking on any of the options in the Tasks' Working plane setup dialog.

What's related to this issue is that the above only happens when using the Start page's buttons. If I create a new file with the File > New menu, then the grid is still aligned to the isometric view, but it appears NOT at an angle, as expected:

[image]

MiqCG, 2024-07-04:
> 1. Do you have `Always show the grid` setting enabled in Edit > Preferences... > Draft > Grid and snapping?

Yes, I had "Always show the grid" active and I didn't know it. Looks like it's the default configuration on a "new Appimage install".

A side of the grid orientation, what looks strange to me is that the grid doesn't appear when creating a "Parametric Part" or "Assembly" for the first time but it appears when a "2D Draft" or "BIM/Architecture" has been previously "called". I suppose that the Draft preferences get active only after Draft WB has been selected before.
Oddly enough this behaviour disappears if a "Parametric Part" or "Assembly" is created twice (even if the first creation is not closed). Are the Draft Preferences disabled after the second "Parametric Part" creation? Why not after the first one?

Of course, if you need to hide the grid created in "Parametric Part", you will need to go to Draft WB to switch it off... BUT I have discovered that if you simply go from Part Design WB to the Draft WB and then come back to Part Design WB (without hiding the grid by button or shortcut), the grid will disappear anyway. So definitely FreeCAD "deep inside knows" that the Grid doesn't belong to PD ;-)

closed 2024-07-04 by maxwxyz

MiqCG, 2024-07-04:
Excuse me, @maxwxyz , how has been the presented problems solved?
Or are you considering that those are not really issues?

maxwxyz, 2024-07-04:
I also cannot reproduce, as @furgo16 also mentioned it seems to be dependent on the setting.
Please rephrase the issue and feel free to reopen. If there are multiple issues, please create new ones for each.

MiqCG, 2024-07-04:
Maybe is dependant on the appimage but as I said, my settings where the default (all FreeCAD config folders deleted) and as you can see in the video, the first time that "Parametrical Part" is created, there is no Grid. So, the inconsistency is evident to me.

On the other hand I don't know how to rephrase the title more appropriately and, in fact, the main issue wasn't about the grid orientation but about the appearance of it (as a Draft WB element) in Part Design WB.

@furgo16 , do you think that the behaviour (the grid appearing in Part Design) is consistent with "Always show the grid" Draft Preferences?

furgo16, 2024-07-04:
Let me try the AppImage later on today, to see if I can reproduce it there.