FreeCAD/FreeCAD #15180: [TD] Default Line Width inconsistence with preferences

j-wiedemann, 2024-07-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

On a fresh install, it seem that there is inconsistent between the default lines width of a techdraw view and the Line Width Group shown in preferences dialog.

Default line width are:
Extra = 1.40 mm, Hidden = 0.35 mm, Iso = 0.50, Line = 0.7 mm
but in preference this is group FC 0.25mm, which is :
Extra = 0.50 mm, Hidden = 0.13 mm, Iso = 0.18, Line = 0.25 mm

It's a problem because if you want FC 0.25mm, then you don't change the value of the combobox in pref, then the new FC group is not triggered.
It's also a problem because it's inconsistent.

I think it's easily fixable:
either change default of techdraw value to FC 0.25mm group.
set the combo box in preference with the right group line : FC 0.70mm

### Full version info

```shell
OS: Pop!_OS 22.04 LTS (pop:GNOME/pop)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git) AppImage
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
```

### Subproject(s) affected?

Techdraw

### Anything else?

Forum discussion : https://forum.freecad.org/viewtopic.php?t=88889

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-04:
@WandererFan FYI

WandererFan, 2024-07-06:
Should be fixed by [TD]fix default line width on empty config 7d8df6cd55

closed 2024-07-06 by maxwxyz