FreeCAD/FreeCAD #15181: Material: ShapeAppearance does not handle emissive color per face correctly

Roy-043, 2024-07-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

ShapeAppearance does not handle emissive color per face correctly. The emissive color of the first material in the list is used for the whole object.

Seems similar to #15027.

Test code:

```python
import FreeCAD as App

doc = App.ActiveDocument
obj1 = doc.addObject("Part::Box", "obj1")
obj2 = doc.addObject("Part::Box", "obj2")
obj2.Placement.Base = App.Vector(30, 0, 0)

# default material:
mat_def = App.Material()
# emissive yellow material
mat_yel = App.Material()
mat_yel.EmissiveColor = (1.0, 1.0, 0.0, 0.0)

obj1.ViewObject.ShapeAppearance = (mat_yel, ) + (mat_def, ) * 5
obj2.ViewObject.ShapeAppearance = (mat_def, ) + (mat_yel, ) + (mat_def, ) * 4

doc.recompute()
```

### Full version info

```shell
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git)
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Dutch/Netherlands (nl_NL)
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-04:
@davesrocketshop FYI

davesrocketshop, 2024-07-04:
Using "Inspect Appearance..." it shows the colors are being assigned correctly, but not being displayed correctly

luzpaz, 2024-07-28:
Hi @davesrocketshop, should ths be pre- or post- 1.0 ?

davesrocketshop, 2024-07-29:
I;m not targeting this one specifically, but it's probably related to others I'm working on.

luzpaz, 2024-10-23:
> but it's probably related to others I'm working on.

Has there been any impact on this ticket at all since last communication ?

maxwxyz, 2025-03-09:
is this still an issue @Roy-043 ?

maxwxyz, 2025-06-16:
same is true for specular color

davesrocketshop, 2025-06-16:
I'm taking an extended break, but I'll be back

Roy-043, 2025-06-17:
> but I'll be back

That is good news. Thank you Dave.