FreeCAD/FreeCAD #15183: New measure tool doesn't work correct

Panda631, 2024-07-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I simply measure the distance beween 2 edges.
1. Select the caliper icon
2. Selecting first edge
3. Selcting the second edge

Thats the result:

[image]

The measured value was correct displayed in the status line 

[image]

The issue was discused here:
https://forum.freecad.org/viewtopic.php?p=769262#p769262

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git)
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * EM 2.1.1
  * fasteners 0.5.21
  * Help 1.0.3
  * kicadStepUpMod 10.17.1
  * Pyramids-and-Polyhedrons
  * Render 2024.1.23
  * sheetmetal 0.2.61
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Panda631, 2024-07-04:
Here it is...

[Measure.zip](https://github.com/user-attachments/files/16098763/Measure.zip)

FEA-eng, 2024-07-04:
Now I can confirm this. But it happens for all modes and anything selected in this model. Messages like this appear:

`MeasureDistanceHandler did not retrieve shape for Body, ;g39;SKT;:H163,E;:H-169,E;:H169,E;:H1d8,E;:H179,E;:H180,E.Edge108`

@hlorus

hlorus, 2024-07-04:
Haven't yet checked in detail but this is likely resolved by either #14918 or #14962.

wwmayer, 2024-07-04:
https://github.com/FreeCAD/FreeCAD/pull/14962 fixes most of the issues but not everything. Measuring angles don't work and report NAN when it should be 90 deg

QuinFromCanada, 2024-07-10:
I'm noticing this issue in testing as well, if you recompute the body it fixes the measuring tool.

I see this in while working in the part design WB whenever I edit a feature that is not at the bottom of the  feature tree.

If it will help I could post a small video showing the issue.

luzpaz, 2024-07-27:
@QuinFromCanada can you paste the About info revision that you can reproduce this on ? (build that integrates the proposed fixes)

QuinFromCanada, 2024-07-30:
> @QuinFromCanada can you paste the About info revision that you can reproduce this on ? (build that integrates the proposed fixes)

> @QuinFromCanada can you paste the About info revision that you can reproduce this on ? (build that integrates the proposed fixes)

I'll post the about when I get home but it was happening starting with "weekly-build-37709-conda-windows-x86_64-py311.7z" bare bones (no mods or extras installed).  I've been running the weekly builds as they come out so it won't be the same version that I commented on originally.

I've been playing around with this since my original post and notice that this happens whenever you make a change to a feature in the Part Design WB that causes the body to not fully update (no blue icon).

I was getting this in my model because the changes I was making would cause a fillet to be lost (or going into the feature would show '?edge#'). But hitting recompute or just going into the fillet and hitting ok would fix the feature.  Once all the feature in the body are recompiled the measurement tool would work again.

Hope that helps, and keep up the great work, the weekly builds are awesome, and I love the new measurement feature, great job.

QuinFromCanada, 2024-08-01:
> @QuinFromCanada can you paste the About info revision that you can reproduce this on ? (build that ```
integrates the proposed fixes)

OS: Windows 10 build 19045

Word size of FreeCAD: 64-bit

Version: 0.22.0dev.37898 (Git)

Build type: Release

Branch: main

Hash: 308013fb975ec3bf0dc5cb616ef76a4095e71cc5

Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2

Locale: English/United States (en_US)
```

maxwxyz, 2024-08-04:
is this still an issue in the latest dev build?

FEA-eng, 2024-08-04:
> is this still an issue in the latest dev build?

I don't have any issues like this with the attached model (even without recreating it) but Werner's comment suggests that there can still be issues with angle measurements. I haven't experienced them and maybe they were fixed by TNP PRs but it might be good to check carefully on different systems just in case.

Panda631, 2024-08-04:
The measurement seems to be OK, but the label is at the wrong position.

[image]

Panda631, 2024-08-17:
A found, that transform operations are not considered for the label positioning.

Label OK:

[image]

Label still displayed at old position:

[image]

maxwxyz, 2024-08-17:
@Panda631 you mean https://github.com/FreeCAD/FreeCAD/issues/13808

Panda631, 2024-08-17:
> @Panda631 you mean #13808

@maxwxyz  maybe not.
The label is placed allways at the original position of the body. If you transform it to any other position, the label does not change its "old" position.

luzpaz, 2024-08-17:
Lets open a new ticket for that then ?

luzpaz, 2024-08-17:
Is this a dupe of #14269 ?

maxwxyz, 2024-08-21:
IMO not. The original issue cannot be reproduced. Comments mentioned that the label is not updated when the body is transformed, this issue was solved recently.
I think this issue could be closed @Panda631 please try on the latest weekly build.

Panda631, 2024-08-23:
> IMO not. The original issue cannot be reproduced. Comments mentioned that the label is not updated when the body is transformed, this issue was solved recently. I think this issue could be closed @Panda631 please try on the latest weekly build.

I can confirm.
Works now with built from today as expected .
Thanks to all.

closed 2024-08-23 by maxwxyz