FreeCAD/FreeCAD #15189: Missing toolbar commands

boboxx, 2024-07-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I have been noticing this alot more recently. if I switch between workbench they don't seem to load correctly and I'm missing some of the buttons. I f I close Freecad and reopen it fixes the issue. I'm not sure what I'm doing but I'm sure i'm not the only one.

For exemple, in this scrrenshot, I had just finished a object built from the sketcher workbenched and moved to the CAM workbench. I have also seen this hapening in the Sketcher WB (those are 2 workbench I use the most)

This is not specific to this release

### Full version info

```shell
OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37928 (Git)
Build type: Release
Branch: main
Hash: e7699b531eba4bee262f7214699f07b96f66ba7b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * OpenTheme 2024.5.16
```

### Subproject(s) affected?

None

### Anything else?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-04:
I cannot reproduce this, what are your steps to get to this behavior?
Is this also happening in the stable 0.21.2 for you?

boboxx, 2024-07-04:
I havent found a way to reproduce but it's happening quite often. It must be when I have issue with my drawing and I switch from different WB.

I do not remember seeing this in version 0.21 but I only used the Sketch WB in that version. I just recently started using the other WB.

I know my answer is not helpful, I was just hopping other users would also be experiencing this issue and we could maby indentify why this was happening.

QuinFromCanada, 2024-07-10:
I also have had this happen in the sketchers workbench, some of the constrains will be missing, the one I've noticed is the parallel and equals.  Closing the sketch and going back in fixes it.

Let me know if you would like me to take a screenshot or video when it happens. 

Cheers.

maxwxyz, 2024-09-01:
Is it the same as https://github.com/FreeCAD/FreeCAD/issues/13243 and would be fixed by https://github.com/FreeCAD/FreeCAD/pull/15933
@kadet1090 FYI, does it need something extra for CAM WB or other WB (https://github.com/FreeCAD/FreeCAD/issues/15717)

maxwxyz, 2024-11-08:
Please comment if this is still an issue.

closed 2024-11-08 by maxwxyz

QuinFromCanada, 2024-11-12:
> Please comment if this is still an issue.

I have not seen this error in the last few versions.  I opened and closed a sketch 100 times using the weekly build 39175 this morning and didn't see it once.

Great Job!