FreeCAD/FreeCAD #15193: Start: Recent files don't ask for "open as" options

maxwxyz, 2024-07-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When opening a recent file from the menu, the user can choose how to open the file, e.g. an .svg file as image or paths or geometry. It asks every time when trying to open the file.
When I open the file from the Start page recent files list, it opens automatically as geometry instead of an image. It should ask the user on filetypes where multiple import options are possible e.g. also .stl files.

Demo:

https://github.com/FreeCAD/FreeCAD/assets/6246609/4699ba05-f4ad-45ca-b2aa-4308e43947cd

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git)
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Assembly3 0.12.2
  * CfdOF 1.26.2
  * Curves 0.6.40
  * dodo 1.0.1
  * fasteners 0.5.24
  * freecad.gears 1.2.0
  * OpenTheme 2024.6.22
  * OpticsWorkbench 1.0.18
  * Rocket 3.3.0
  * sheetmetal 0.4.21
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2025-08-02:
@maxwxyz, I cannot reproduce on

```ruby
OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.42788 (Git)
Build date: 2025/07/31 11:43:31
Build type: Unknown
Branch: master
Hash: d4303f19ca32b6f6d62363a03d985915d4f9a2a2
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, boost 1_83, Eigen3 3.4.0, PySide 5.15.13
shiboken 5.15.13, SMESH 7.7.1.0, xerces-c 3.2.4, OCC 7.6.3
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Gesture/Trackball/Window center
Stylesheet/Theme/QtStyle: unset/FreeCAD Classic/Fusion
Logical DPI/Physical DPI/Pixel Ratio: 96.2463/96.2463/1
```

Does it depend on the SVG file?

maxwxyz, 2025-08-02:
this SVG was created with FreeCAD, I will try to create a new one

maxwxyz, 2025-08-02:
Ok I no longer can reproduce with current main.

closed 2025-08-02 by maxwxyz