FreeCAD/FreeCAD #15198: Linux FreeCAD Copy and paste issue with wayland

Expore-it-all, 2024-07-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This is my first time posting an issue on Git Hub for this project, if I have missed anything please let me know.
Hi all, I have been using Freecad on Linux and after creating a body I wanted to copy and paste it like I have been able to do so on windows. but I have found that I am getting an error.

Here is the error I am seeing for it whenever I try to copy and paste through the context menu or shortcut keys.

```
20:13:43  (qt.qpa.clipboard) QXcbClipboard::setMimeData: Cannot set X11 selection owner

```

I can copy and paste text normally but trying to do so on a body causes this. I have a hunch it is because of Wayland and XWaylandBridge Issues, though Im not 100% sure on this.
I have attached a video of this in action. I tried both context and shortcuts to copy but it still happens.

### Full version info

```shell
OS: KDE neon 6.0 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Dracula 0.0.5
```

### Subproject(s) affected?

Core

### Anything else?

https://github.com/FreeCAD/FreeCAD/assets/16739847/9d64428c-8da6-4e64-ac89-5d209b445dcf

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Expore-it-all, 2024-07-04:
After some more looking I have found that there is also another Issue related to XWaylandBridge and Copy and Pasting
#10251 
Only difference is this is the AppImage Format not flatpack.

krushia, 2024-07-06:
Begin rant.

This is a Wayland limitation. Wayland only "supports" (this is [generous](https://github.com/Slackadays/Clipboard/issues/171#issuecomment-1759936147) wording) the primary X copy buffer by design, which it uses for wl_clipboard. Here we have an error that is about the selection buffer, which Wayland has no support for at all.

Usually when a project "fixes" this, they also break middle-click paste for people on X11 (CherryTree is one example). This makes me sad. Properly supporting Wayland's weirdness can be quite involved; see for example https://github.com/rustdesk/rustdesk/pull/5900.

End rant.

Simply rebuilding the Flatpak will probably solve it.

Expore-it-all, 2024-07-07:
I wish I knew more about coding in the Linux Environment so I could help with this, but my knowledge is quite limited in that aspect. Thanks for letting me know about the issues about it.

luzpaz, 2024-07-22:
Any care to let the FC flatpak packager in on this discussion ?

djkato, 2024-08-25:
Please at least in the meantime add an option to "duplicate" under the right click menu while this is broken. Duplicating sketches/bodies etc shouldn't be broken because of the systems clipboard manager. Imo this should be possible inside FreeCAD regardless.(SS from this weeks weekly build)
[image]

tarman3, 2024-12-07:
Still relevant in AppImage

<details>
  <summary>FreeCAD 1.0.0.39109 (Git) Conda AppImage, Arch Linux (KDE/plasma/xcb)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Conda AppImage
Build type: Release
Branch: (HEAD detached at 1.0.0)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.31; sheetmetal 0.5.8; Silk 0.1.5; Curves 0.6.51
```
</details>

<details>
  <summary>1.1.0dev.39416 (Git) Conda AppImage, Arch Linux (KDE/plasma/wayland)</summary>

```
OS: Arch Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.34; freecad.gears 1.3.0; fasteners 0.5.32; sheetmetal 0.5.10; Silk 0.1.5; Curves 0.6.51
```
</details>

Makotophoto, 2024-12-16:
I would like to give some hints, since i encountered the copy/paste issue by the usage of a  workaround for another problem.

My first problem was the 3d display not rendering with console message `Couldn't get current GLX context`.
The workaround was  to add the environment variable `QT_QPA_PLATFORM=xcb`
This one is generating the copy/paste problem.

I solved the 2 issues by replacing this environment variable by : `__GLX_VENDOR_LIBRARY_NAME=mesa __EGL_VENDOR_LIBRARY_FILENAMES=/usr/share/glvnd/egl_vendor.d/50_mesa.json`  (recipe found for another app getting display problems)

> Using Manjaro Unstable / KDE / Wayland / Nvidia / Freecad-git (AUR)

Groberen, 2025-04-16:
Still an issue with Fedora 41 KDE. See forum post for additional details:
https://forum.freecad.org/viewtopic.php?p=821402#p821402

Full error message on attempt to copy:
(qt.qpa.clipboard) QXcbClipboard::setMimeData: Cannot set X11 selection owner

ElementW, 2025-04-18:
Reproducible in KDE **if Klipper is enabled** with native 1.0.0 (when `QT_QPA_PLATFORM=xcb`), and in 1.1.0dev flatpak.
The `QXcbClipboard::setMimeData: Cannot set X11 selection owner` warning is due to [Qt expecting the running app to be the clipboard owner](https://github.com/qt/qtbase/blob/53ff8897c5c8bc6175cf94ed24e2d2c5fa17365b/src/plugins/platforms/xcb/qxcbclipboard.cpp#L338), which will never be true if any clipboard manager is running.
Other software running in XWayland have the clipboard working just fine, [unless they don't because they don't set their clipboard expectations right](https://gitlab.com/inkscape/inkscape/-/issues/5528) (i.e. that anything can request all your advertised clipboard MIME types at any time and this should *never* cause a crash or slowdown of the UI thread).
<details>
<summary>Native 1.0.0 info</summary>
OS: Void Linux (KDE/xcb)
Architecture: x86_64
Version: 1.0.0.2_voidlinux
Build type: Release
Python 3.13.1, Qt 6.7.2, Coin 4.0.3, Vtk 9.3.1, OCC 7.7.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * fasteners 0.5.31
  * OpenTheme 2024.9.1
  * FreeCAD-Ribbon 1.4.3 (Disabled)
</details>
<details>
<summary>1.1.0dev flatpak</summary>
OS: Void Linux (KDE/xcb)
Architecture: x86_64
Version: 1.1.0dev.41138 (Git) Conda AppImage
Build date: 2025/04/04 17:31:11
Build type: Release
Branch: main
Hash: cab9053be825b8f09d686d5894f333915e4ee378
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, IfcOpenShell 0.8.1, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/92.6073/1
Installed mods: 
  * fasteners 0.5.31
  * OpenTheme 2024.9.1
  * FreeCAD-Ribbon 1.4.3 (Disabled)
</details>
Most likely this is caused by FreeCAD being unable to seralize and deserialize its selection properly in this case—if it even can.

---

@krushia Your rant is misplaced and stating outright lies
1. The base Wayland protocol does *not* support X primary selection.  
   The only relevant thing it supports here is [`wl_data_device::set_selection`](https://wayland.freedesktop.org/docs/html/apa.html#protocol-spec-wl_data_device-request-set_selection`), which is described as an abstract way to implement "copy-and-paste"; in fact the entire spec does not mention "clipboard" once, nor the semantics of it. And status quo in compositors and clients alike is this is the secondary (Ctrl-C/Ctrl-V) clipboard.
2. `wl_clipboard` does not exist (the [wl-clipboard](https://github.com/bugaevc/wl-clipboard) tool is not the same)
3. X11 primary selection-style copying has been available on Wayland for a while anyway as:
   - `wp_primary_selection_unstable_v1` [is 6+ years old by now](https://gitlab.freedesktop.org/wayland/wayland-protocols/-/commit/2c3b11d76fd498fc3856cdd36ac76c1193686bf1) (2018-10-18) and
     * Kwin [implemented it 2020-06-01](https://invent.kde.org/plasma/kwin/-/commit/e712c5db2bd00f4be59331e39ef42312d5cf3b1c)
     * wlroots [implemented it 2019-01-24](https://gitlab.freedesktop.org/wlroots/wlroots/-/commit/8163f7e1e21d9b039319cc3e38993811bfbeea4f)
     * Mutter switched from their protocol [on 2020-05-13](https://gitlab.gnome.org/GNOME/mutter/-/merge_requests/1255) ([commit](https://gitlab.gnome.org/GNOME/mutter/-/commit/2becb3dd29f8b62f4b68938ecbfa1ec41a0d9fca))
     * Qt [implemented it 2019-05-10](https://github.com/qt/qtwayland/commit/5ec182df699041699f514d164a161c299fde5d19)
     * Gtk switched from their protocol [on 2020-09-27](https://gitlab.gnome.org/GNOME/gtk/-/commit/9ea0469ab0a97fa5bdb515098559e681739127ba)
   - The precursor `gtk_primary_selection` [is 9 years old](https://gitlab.gnome.org/GNOME/mutter/-/commit/c6aad6e7353abe77c821330830472fbe342fd068)
     * Mutter (origin for the protocol) implemented it 2016-02-23
     * Gtk [introduced it on the very same day](https://gitlab.gnome.org/GNOME/gtk/-/commit/787e1d71133089668f0ef8ec19b4d50eb1d89e4e)
     * wlroots [implemented it 2018-11-23](https://gitlab.freedesktop.org/wlroots/wlroots/-/commit/811a4d997b2e00c7cdc3851ccc55a6f52d79126c)
     * Kwin and others, as well as toolkits like Qt never implemented it

You clearly have no idea what you are talking about, so don't go saying it's limited just *because it's not X11*. Yes it's still rough, but a whole lot compatibility problems come from developers expecting an X12 of sorts when it is not.

cyberwolfie1, 2025-04-24:
I have the same behavior:

```
OS: KDE Flatpak runtime (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.39109 (Git) Flatpak
Build type: Release
Branch: (HEAD detached at 2fcc531)
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.11.11, Qt 6.7.3, Coin 4.0.3, Vtk 9.2.6, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: 
  * fasteners 0.5.38
```

It _sometimes_ work, but seemingly randomly and I've not found a good way to reproduce when it does work vs. not. Trying to copy the same object twice can lead to first success then failure. It fails maybe 95% of attempts.

luzpaz, 2025-05-10:
@cyberwolfie1 would you be willing to test 1.1dev[^1] for a bit and see if you can reproduce the issue ?  
Note: backup your models 1.0 models before opening them in 1.1dev

[^1]: https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds#assets

samin50, 2025-05-14:
Hi, if it's any help - I am having the same issue using Arch running KDE Plasma 6. It comes with a clipboard manager Klipper and whilst this is running I get the `QXcbClipboard::setMimeData: Cannot set X11 selection owner` error if I try to copy in FreeCAD.

This happens in both 1.1dev and the official 1.0 release, and the flatpak version for me. If I disable Klipper, I no longer see this issue on 1.1dev, at least (I have not tried the others). Obviously this isn't very desirable because a clipboard manager is quite useful to have, so if there's anything I can help with let me know

luzpaz, 2025-05-15:
@samin50 can you post full about info for 1.1dev you've tested this on ?

samin50, 2025-05-17:
Of course, I am using:
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0dev.41628 (Git) AppImage
Build date: 2025/05/09 10:50:32
Build type: Release
Branch: main
Hash: 9577d108e333c80ee640211271a76105f52baf14
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/125.865/1.5

Some information about my system:

Linux archlinux 6.14.5-arch1-1 #1 SMP PREEMPT_DYNAMIC Sat, 03 May 2025 13:34:12 +0000 x86_64 GNU/Linux

NAME="Arch Linux"
PRETTY_NAME="Arch Linux"
ID=arch
BUILD_ID=rolling
ANSI_COLOR="38;2;23;147;209"
HOME_URL="https://archlinux.org/"
DOCUMENTATION_URL="https://wiki.archlinux.org/"
SUPPORT_URL="https://bbs.archlinux.org/"
BUG_REPORT_URL="https://gitlab.archlinux.org/groups/archlinux/-/issues"
PRIVACY_POLICY_URL="https://terms.archlinux.org/docs/privacy-policy/"
LOGO=archlinux-logo

Let me know if anything else is required!

cyberwolfie1, 2025-05-18:
> [@cyberwolfie1](https://github.com/cyberwolfie1) would you be willing to test 1.1dev[1](#user-content-fn-1-8e2c2d9ba124adf2ab668aac64fb3247) for a bit and see if you can reproduce the issue ? Note: backup your models 1.0 models before opening them in 1.1dev
> ## Footnotes
> 
>     1. https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds#assets [↩](#user-content-fnref-1-8e2c2d9ba124adf2ab668aac64fb3247)

I've tried it with the latest nightly build, and I still get the same error: `(qt.qpa.clipboard) QXcbClipboard::setMimeData: Cannot set X11 selection owner`. I have not tried to disable Klipper as the user above me said would fix the issue, but I would also not like to do without my clipboard. 

```
OS: TUXEDO OS (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0dev.41726 (Git) AppImage
Build date: 2025/05/16 17:15:43
Build type: Release
Branch: main
Hash: 1ca108f8d503b2b68455844bfdfabb62fa1461ed
Python 3.11.12, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.2.5, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/70.8094/2
```

coolasbreeze, 2025-07-31:
Was there any solution to this? I am having the same issue. 

I am running Debian testing but switching back to x11 seems to be okay

`OS: Debian GNU/Linux 13 (trixie) (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.0.0.
Build type: Release
Python 3.13.5, Qt 5.15.15, Coin 4.0.3, Vtk 9.3.0, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * fasteners 0.5.38
  * ThreadProfile 1.96.0
  * 3D_Printing_Tools
  * CurvedShapes 1.0.13
  * Pyramids-and-Polyhedrons
  * Curves 0.6.64
`

forest-hermit, 2025-11-17:
> > the today appimage dev build worked.
> 
> [@forest-hermit](https://github.com/forest-hermit) can you post the full [About](https://wiki.freecad.org/About) info of the version it worked on please ? Thank you.

```
OS: Ubuntu 25.10 (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.1.0dev.20251111 (Git shallow) AppImage
Build date: 2025/11/11 16:45:10
Build type: Release
Branch: main
Hash: 4fbb389ce1295e04a5ad7f3a63f68b63b4f4036c
Python 3.11.14, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Spanish/Spain (es_ES)
Navigation Style/Orbit Style/Rotation Mode: Blender/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 96/141.951/1
Installed mods: 
  * Silk 0.2.0
  * OpenTheme 2025.5.20
  * lattice2 1.0.0
  * FusedFilamentDesign 0.25.380
  * OpenDark 2024.3.13
  * Curves 0.6.70
  * CurvedShapes-backup-before-zip-migration 1.0.13 (Disabled)
  * StructureTools 0.0.1
  * CurvedShapes 1.0.14
```

Gummy35, 2026-02-15:
Just had the same problem minutes ago. I was able to copy/paste after clearing klipper history.

luzpaz, 2026-09-06:
Is this still a relevant Wayland related issue ?

VolkerRaab, 2026-09-07:
At least with the "standard" (i.e. Debian-Trixie) distributed components it is still relevant for me:
FreeCAD: 1.0.0
libwayland-bin: 1.23.1-3
xwayland: 2:24.1.6-1
kwin-wayland: 4:6.3.6-1
I still have to first copy a simple object (sphere, box) and can only then copy a more complex object like a part. Otherwise the copy-hierarchy is shown but upon pasting nothing happens.
(I am lost whether it is wayland-related, kwin-related or FreeCAD-related)

luzpaz, 2026-09-07:
>  FreeCAD: 1.0.0

What? This isn't latest stable 
You're not using hte official FreeCAD executable but the Debian project packaged FreeCAD. Can you reproduce with the AppImage or FLatpak ?

VolkerRaab, 2026-09-07:
I'm awfully sorry, to have wasted your time.
However, Yes, it is still happening with FreeCAD 1.1.3 (AppImage) as of 25th of July.

ElementW, 2026-10-02:
@maxwxyz You can drop the https://github.com/FreeCAD/FreeCAD/labels/Packaging%3A%20Flatpak (also happens on AppImage/native) and https://github.com/FreeCAD/FreeCAD/labels/Status%3A%20Duplicate (#10251 was closed) labels and can add https://github.com/FreeCAD/FreeCAD/labels/3rd%20party%3A%20Qt.

DaAngus, 2026-10-31:
Still happening with 1.1.4-3 

`(qt.qpa.clipboard) QXcbClipboard::setMimeData: Cannot set X11 selection owner`

UlyssesApokin, 2026-11-06:
It seems the problem has been resolved and no longer recurs.

See also: https://github.com/FreeCAD/FreeCAD/issues/25258#issuecomment-6693792024

adrianinsaval, 2026-11-14:
I cannot reproduce this in plasma 6.8 and freecad 1.1.4 or weekly build, this might be a bug in kde 6.3