FreeCAD/FreeCAD #15201: Start: Python console opened by default on fresh build

marcuspollio, 2024-07-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When opening a freshly compiled build on GNU/Linux with no previous .cache, .config and .local/share folders, the Start Page has the Python console opened by default :

[image]

The console takes screen estate, hiding the New Files dialog and is rather intimidating for first-install users.
It should be hidden by default, as it's always accessible at View > Panels > Python console.

### Full version info

```shell
OS: GNU/Linux Ubuntu 22.04 LTS (DE: Gnome/pop)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38039 (Git)
Build type: Release
Branch: main
Hash: 7a9fd501c3ea0e259f843dc21f488ef6af43c201
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk , OCC 7.6.3
```

### Subproject(s) affected?

Other (specify in description)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-05:
Same on Windows.

wwmayer, 2024-07-05:
These messages don't come from FreeCAD. Maybe an add-on or macro executed at startup prints the messages.

Syres916, 2024-07-05:
`flag` and `flag2` are from dodo workbench as far as I remember.

wwmayer, 2024-07-05:
https://github.com/oddtopus/dodo/blob/91b6ee4cd03270bc107512212d1573b913ebc5c3/InitGui.py#L33

PR #15209 (2024-07-05): Gui: Don't show Python console by default
Fixes #15201
commit 7ab2b92f (2024-07-05): Gui: Don't show Python console by default

Fixes #15201
closed 2024-07-05 by wwmayer