FreeCAD/FreeCAD #15202: Assembly is crashing a lot

leoheck, 2024-07-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Segfault when using Parallel constraint of the new Assembly (segfaults are happening all the time)

Steps to reproduce:
0. Activate the Assembly
1. Select the bottom face (gray) of the device
2. Apply the Parallel Joint
3. Flip the Parallel Joint orientation.

[image]

Also sometimes I can reproduce, sometimes I cannot, but this Assembly is crashing every few features.

For instance here are other segfaults that happen while playing with this.
[image]

[image]

[image]

Also, grabbing the objects and moving them around makes the joints be ignored sometimes.
For instance, by the applied joints, this should not be possible.
One object overlaps another (this is why we see 3 only), and one is not touching another one.

Here is my file to play with:
[image]

### Full version info

```shell
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git) AppImage
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * POV-Ray-Rendering
  * Assembly4 0.50.12
  * InventorLoader 1.4.0
  * ose-piping
  * OpenTheme 2024.7.3
  * SheetMetal 0.4.20
  * fasteners 0.5.24
  * asm4_disable_version_check_popup
  * Render 2024.6.5
  * OpticsWorkbench 1.0.18
  * lattice2 1.0.0
  * kicadStepUpMod
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-05:
I cannot reproduce, no crashes on my side. Why did you use 2 parallel joints per part?

leoheck, 2024-07-05:
How many parallel joints I can use? 

I want to make things parallel to more than one object.
I also don't want to type unnecessary dimensions. I would like to assemble based on existing features. For instance, with Ass4, this works like a charm never ever had a problem. I cannot use A2+ either, the Assembly crashes all the time. 

Is there any special workflow I should use? I may be doing something completely wrong then, but I could not find a wiki explaining the workflow. However Freecad should not crash that often.

I am trying hard to use this WB since it was released. But It breaks every time. Every single thing I do crashes Freecad. I could not assembly more than a couple of objects.

I saw a post on Reddit that this particular version of Freecad is very buggy. It may also be the reason of the crashes and not entirely due to the Assembly wb.

leoheck, 2024-07-06:
This may be related https://github.com/FreeCAD/FreeCAD/issues/15225

maxwxyz, 2024-11-09:
@leoheck is this still relevant or fixed in the latest weekly as #15225 was fixed?

MiqCG, 2024-12-03:
Just for reference,  #15225 wasn't fix but closed in favour of #14185, that was fixed between june and july.

I couldn't reproduce the issue with the last appimage (rev. 39416) nor with 1.0.0.
No messages in the report view, neither in the terminal.

Followed steps:
1. open test file in FreeCAD
2. activate the assembly container with double click
3. recompute the assembly with Control-r
4. drag the various cylinders, one at a time

I noted that one of the batteries is coincident with another one (the first and the last), but I guess this is by design.

```
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce/xcb)
Architecture: x86_64
Version: 1.1.0dev.39416 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: f6268ffd2877dadb4440ade541e12c388a517d6e
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
Stylesheet/Theme/QtStyle: FreeCAD Light.qss/FreeCAD Light/Fusion
```

luzpaz, 2025-05-20:
@leoheck can you still reproduce ?