FreeCAD/FreeCAD #15203: Export filename preset issue when object label contains a dot '.'

jffmichi, 2024-07-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Trying to export an object whose label contains a dot causes the filename preset in the save dialog to be truncated at said dot. This only occurs on the first export after opening FreeCAD.

Steps to reproduce:

Open new FreeCAD instance
Open attached file [export_filename_issue.zip](https://github.com/user-attachments/files/16105870/export_filename_issue.zip)
Click on object `ring_30.0mm`
File > Export...
Filename preset reads `export_filename_issue-ring_30.3mf`

Expected behaviour:

Filename preset should read `export_filename_issue-ring_30.0mm.3mf`

When you now export said object or any other object, FreeCAD behaves as expected and the filename preset includes `.0mm` for every export from now on.

### Full version info

```shell
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37927 (Git)
Build type: Release
Branch: makepkg
Hash: 4d6b738062fd29d7c6098a7197551bc291b13b1a
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US) [ OS: English/United States (en_US) ]
Installed mods: 
  * kicadStepUpMod 11.1.7
  * dodo 1.0.1
  * sheetmetal 0.4.16
  * fcgear 1.0.0
  * fasteners 0.5.21
  * Assembly4 0.50.12
  * Curves 0.6.37
  * lattice2 1.0.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2024-07-06:
This feature comes from _Qt_. The last letters after dot are considered to be _completeSuffix_ by _QFileInfo_. The _FileDialog_ (a derivation of _QFileDialog_) is used to obtain the file name or just `QFileDialog::getSaveFileName` if _dontUseNativeFileDialog_ has been set to _false_.

I've experimented and set the name to `file_1.0.0`, it becomes `file_1.0.3mf`.
`file_1.0.` becomes `file_1.0..3mf`
`file_1.0.3mf` becomes `file_1.0..3mf`

luzpaz, 2024-07-14:
@wwmayer if you have a moment, could you check this out ? :pray:

DeniseBryson, 2024-07-23:
Hi,
I found that on the first export no extension is added. On the second export the is a "last used extension" which gets added to the filename before passing it to the filedialog.
In the second case the actual extension gets cut, in the first case just a part of the filename gets cut if there is a dot in the name.
I tried to solve this by adding a sample extension ".xxx" if there is no last extension. I hope this can be a solution, but maybe I am missing something?

commit 7c584433 (2024-08-12): Gui: Fix #15203 filename extension export
closed 2024-08-12 by adrianinsaval