FreeCAD/FreeCAD #15205: Sketcher: Basic referencing of exterlal geometry brakes the sketch

shaise, 2024-07-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Simple referencing of external geometry breaks the sketch.
Way to reproduce:
1. Create Body
2. Create rectangle sketch
3. Pad
4. Select a face and create new sketch
5. Select one of the pad edges and press "Create external geometry"
6. Add any feature that references this geometry (i.e. create rectangle with one corner on the referenced edge)
7. close the sketch
An error appears and the sketch is broken. Only way to fix it is to remove the external geometry.
```
11:28:30  696.772 <App> Document.cpp(3152): exception in Unnamed#Sketch001 thrown: argument not found
11:28:30  Sketch001: argument not found
```

[image]
[image]
  

### Full version info

```shell
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git)
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * BIM_Workbench 2021.12.0
  * fasteners 0.5.24
  * Manipulator 1.5.7
  * OpenTheme 2024.7.3
  * PieMenu 1.7.4
  * Render 2024.6.5
  * SelectorToolbar
  * sheetmetal 0.4.21
  * TabBar (Disabled)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-05:
Maybe related? https://github.com/FreeCAD/FreeCAD/issues/14669

maxwxyz, 2024-07-05:
Does this work in the stable 0.21.2 version?

PaddleStroke, 2024-07-05:
@bgbsww does this issue ring a bell to you?

bgbsww, 2024-07-05:
It is certainly suspiciously close, and fails in the ExternalGeometryFacade::initExtensions call inside the SketchObject::buildShape() call trying to format a warning message.
```
        Base::Console().Warning(
            "%s\nSketcher External Geometry without ExternalGeome
            boost::uuids::to_string(Geo->getTag()).c_str());
```

The two console Warning lines in that source file are commented out in LS3, and the getTag() and to_string both return workable values, but Warning call fails, presumably because of two %s but only one parameter.

Commenting out those lines appears to solve this problem, and the lines themselves appear to be vestigial and useless.  The enabling of the code in buildShape that calls this happened in #14994 so this was triggered by TNP changes.

shaise, 2024-07-05:
> Does this work in the stable 0.21.2 version?

Of course it works on stable 0.21.2. (and I tested it again just to be sure) This is the most common usage of sketcher. In fact I'm rather surprised no one else encountered it.

shaise, 2024-07-05:
> Werner had already fixed the broken messages a few hours ago, but they don't appear to need to be there at all.
I can confirm that with the latest changes of Werner everything is working again. 
There is a warning message though, if it maters:
`Sketcher External Geometry without ExternalGeometryExtension: bad1f1b1-c018-4284-9d2e-3b3a80abaa05`
Closing this issue.
Thanks!! I can continue my model now...

closed 2024-07-05 by shaise

hkienle, 2024-07-06:
> There is a warning message though, if it maters:
> `Sketcher External Geometry without ExternalGeometryExtension: bad1f1b1-c018-4284-9d2e-3b3a80abaa05`

Maybe a bit too early to close?

The message is in red. If everything works fine now, a red warning message seems inappropriate!?

[image]

OS: PureOS 10 (Byzantium) (GNOME/gnome-xorg)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git) AppImage
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.7

PR #15214 (2024-07-09): Toponaming: Comment out console warnings triggered by ex…
…ternal geometry

Fix #15205.

Werner found and fixed the misformatting in #15060; the warnings seem to be superflous, so this removes the noise.  Triggered by enabling of TNP code in #14994.