FreeCAD/FreeCAD #15206: [CAM] Helix and inside Profile cutting direction wrong

jffmichi, 2024-07-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When using Helix or Profile to cut a hole and selecting cutting direction "Climb" the cutter moves clockwise (should be counter-clockwise) with a "Forward" rotating tool. Outside Profile moves in the same direction which is correct in this case.

The attached file shows from left to right a Helix, an inside Profile (with ramp dressup for better visualization) and an outside Profile. All have cutting direction "Climb" selected.

[cut_direction_issue.zip](https://github.com/user-attachments/files/16107471/cut_direction_issue.zip)

[image]

### Full version info

```shell
OS: Arch Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37927 (Git)
Build type: Release
Branch: makepkg
Hash: 4d6b738062fd29d7c6098a7197551bc291b13b1a
Python 3.12.4, Qt 6.7.2, Coin 4.0.2, Vtk 9.3.0, OCC 7.8.1
Locale: English/United States (en_US) [ OS: English/United States (en_US) ]
Installed mods: 
  * kicadStepUpMod 11.1.7
  * dodo 1.0.1
  * sheetmetal 0.4.16
  * fcgear 1.0.0
  * fasteners 0.5.21
  * Assembly4 0.50.12
  * Curves 0.6.37
  * lattice2 1.0.0
```

### Subproject(s) affected?

CAM/Path

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

jbaehr, 2024-08-31:
I think this issue here is a direct consequence of just renaming "CW"/"CCW" to "Climb"/"Conventional" (#14364). The old names have been correct, in the sense how the generated paths behave but required some thinking of the person operating the software. The new naming implies that the implementation takes care of inside/outside profiles, but it's not.

maxwxyz, 2024-11-11:
fixed by https://github.com/FreeCAD/FreeCAD/pull/17655

PR #17655 (2024-11-11): CAM: Fix CW/CCW to Climb/Conventional rename
With #14364 a fairly large search/replace PR was merged in order to address #14314.
Unfortunately, this resulted in quite some collateral damage (e.g. #15643 and #15009) and misunderstandings (e.g. #15206 -- just renaming things does not mean the feature got implemented).

With this PR I try to fix the above mentioned bugs, address the critics on #14364 from @Connor9220 and @jffmichi and actually implement #14314.

It's still incomplete and I know that it's quite late for v1.0, but let's hope for the best. I just don't want to see a v1.0 released th [... cut]
closed 2024-11-11 by maxwxyz