FreeCAD/FreeCAD #15208: [Feature Request] Assembly: Flexible subassembly

czinehuba, 2024-07-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This is a feature request would be to enable subassemblies to bring forward their degree of freedom to the main assembly.
I have this sub assembly for example and I configured the limits and joints. Which works great. Kudos.
[image]

Now, if it would be possible to allow for those degree of freedoms to pass into the next level of parent assembly, as I can't move the block on the linear rail in the main assembly, just the sub assembly. 
[image]

A good, intuitive solution would be to add an option for subassembly to be flexible. That would move a DOF from the subassembly to the parent assembly. Also it adds an icon to indicate this.
[image]

### Full version info

```shell
OS: Windows 11 build 22621
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git)
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * 3DfindIT 1.2.0
  * CfdOF 1.27.3
  * Curves 0.6.40
  * fasteners 0.5.24
  * freecad.gears 1.2.0
  * lattice2 1.0.0
  * OpenTheme 2024.7.3
  * sheetmetal 0.4.21
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

PaddleStroke, 2024-07-05:
Currently if you put an assembly directly into another assembly, ie NOT a link to an assembly (as the insert component would do), then the sub-assembly should be dynamic (there might be some bugs though).
If you put a link rather than the assembly itself then it becomes more difficult to handle because if you have several links to the same assembly then what? The components of the sub-assembly would be the same accross the links.

So it would need a way to override assembly components placement in the links. Perhaps something with copyOnChange.

czinehuba, 2024-07-05:
OK. That is working if I copied the subassembly into the assembly. But that would mean a lot of duplication?
Also, how would that work where I need, for example, to adjust the subassembly. Then I have to copy the whole thing again into the main one?

Your suggestion works with the top-down process, where everything is in one file. But not in bottom up design. Where you build start from individual files, next subassemblies, and possibly nesting a few levels. Plus, I had to get rid of the grounding the rail and replace with a fixed constraint.

The ideal workflow should be as explained before. Would that be possible? Or is that a limitation of the solver?

PaddleStroke, 2024-07-05:
I know, I just said it works with the sub-assembly itself.
It doesn't work with link because it's a limitation of freecad. Here's a video to explain why it currently can't be done : 

https://github.com/FreeCAD/FreeCAD/assets/19984177/183f62f0-b3ec-47fe-a8ce-09d4cd2dcf9e

PaddleStroke, 2024-07-05:
I think the way would be to create a new document object type called 'AssemblyLink' which would actually not be a Link.
Instead if would be a special object that instead of linking to the subassembly it would have links to the components of the assembly. And it would keep the list of components correct by updating it on recompute.

So basically AssemblyLink would derives from AssemblyObject, and it would populate itself with links to the component of the real assembly.
Sounds mad enough to me but I don't see another way.

czinehuba, 2024-07-05:
Thanks for looking into this.

Zolko-123, 2024-07-20:
> A good, intuitive solution would be to add an option for subassembly to be flexible. That would move a DOF from the subassembly to the parent assembly

that's what [varianLinks](https://forum.freecad.org/viewtopic.php?t=62767) do. You just need to use the right tool

furgo16, 2024-07-23:
Related forum thread: https://forum.freecad.org/viewtopic.php?p=772142#p772142

PaddleStroke, 2024-07-26:
> that's what [varianLinks](https://forum.freecad.org/viewtopic.php?t=62767) do. You just need to use the right tool

I don't think variant link would work for this issue.
Links' placements are already different between each link. You don't need variant link for this.

The components in the sub assembly-links are not different links, they are the same object. And one object cannot have different placements.

Zolko-123, 2024-07-26:
> > that's what [varianLinks](https://forum.freecad.org/viewtopic.php?t=62767) do. You just need to use the right tool
> 
> I don't think variant link would work for this issue. 

which only shows that you don't know what `variantLinks` are and what they can do

> The components in the sub assembly-links are not different links, they are the same object. And one object cannot have different placements.

of course they can, that's what `variantLinks` do : in the `Placement` of the part in the sub-assembly is an `AttachmentOffset` that you can set with a `Variable`, and this `Variable` can be piloted in the assembly for each sub-assembly individually, so that each sub-assembly will hold the same parts but placed differently. And what's more it's transitive, meaning that you can set `Variables` in a sub-sub-assembly – and sub-sub-sub-sub-assembly if you want – directly from the top level assembly.

How do you think that the example of the Theo Jansen sandwalker works ?

PaddleStroke, 2024-07-27:
Thanks for the reply and explanations.
I will need to check this out in more details then as it would be best to
use these.

PaddleStroke, 2024-08-06:
So I have looked into it from the example file. And I was right, the same object cannot have two different placements.

In asm4's VariantLink you are not giving different placement to the same object. You are making different link to the same object and these links have different placement.

VariantLinks are not even App::Links, they are FeaturePythons. They are created in temporary documents and linked to.

MiqCG, 2024-08-07:
> Currently if you put an assembly directly into another assembly, ie NOT a link to an assembly (as the insert component would do), then the sub-assembly should be dynamic (there might be some bugs though).

Hi, PIerre,

I was willing to try the flexible subassembly possibilities but I am not even able to "put an assembly directly into another assembly". I am trying to move one existing assembly (that has to be the subassembly) into another new assembly by drag and drop (with several "techniques") without results.
Or are you referring to create one or more assemblies into another (main) assembly and then populate each one with parts and joints from the beginning?

PaddleStroke, 2024-09-11:
@MiqCG it appears that there are functionality issues with assemblies directly in other assemblies. That will be fixed in my current PR. Dragging one into another should work. I have not tested in current main but in my current branch it works at least.

PaddleStroke, 2024-09-11:
@realthunder Could you please comment on this flexible assembly topic? There's a video describing the problem and potential solutions in the forum thread : https://forum.freecad.org/viewtopic.php?t=89650

prokoudine, 2024-09-27:
@PaddleStroke 

I think there's a mistake in the task panel option. The checkbox says 'Rigid sub-assemblies', but it should be 'Flexible sub-assemblies' instead, because when the checkbox is on, joints from a sub-assembly are preserved and used.

luzpaz, 2024-10-16:
@PaddleStroke ☝️

closed 2024-10-28 by yorikvanhavre