FreeCAD/FreeCAD #15210: Exception thrown when trying to create new body

trohsb, 2024-07-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

`Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'Base.FreeCADError'>: yaml-cpp: error at line 27, column 8: operator[] call on a scalar (key: "DisplayName")`

Steps to reproduce:
1. Start FreeCad
2. Create new empty document
3. Select Part Design workbench (Already selected for me)
4. Switch to Tasks List
5. Click Create body task

### Full version info

```shell
OS: Gentoo Linux (KDE)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.Unknown
Build type: RelWithDebInfo
Python 3.12.3, Qt 6.7.1, Coin 4.0.2, Vtk 9.2.5, OCC 7.7.0
Locale: English/United Kingdom (en_GB)
```

### Subproject(s) affected?

None

### Anything else?

FreeCad is build from the latest git revision.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-05:
Cannot reproduce on 
```
OS: Windows 11 build 22631
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git)
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Installed mods: 
  * Assembly3 0.12.2
  * CfdOF 1.26.2
  * Curves 0.6.40
  * dodo 1.0.1
  * fasteners 0.5.24
  * freecad.gears 1.2.0
  * OpenTheme 2024.6.22
  * OpticsWorkbench 1.0.18
  * Rocket 3.3.0
  * sheetmetal 0.4.21

```

trohsb, 2024-07-05:
Opening the preferences dialog produces the following error:

`21:15:32  C++ exception thrown for 'MatGui::DlgSettingsDefaultMaterial' (yaml-cpp: error at line 27, column 8: operator[] call on a scalar (key: "DisplayName"))`

maxwxyz, 2024-07-06:
Is it working for you in the stable 0.21.2 version?

trohsb, 2024-07-06:
Hi. After several hours of compilation (gentoo) I can confirm that 0.21.2 works fine.

[code]
OS: Gentoo Linux (KDE)
Word size of FreeCAD: 64-bit
Version: 0.21.2.Unknown
Build type: RelWithDebInfo
Python 3.11.9, Qt 5.15.14, Coin 4.0.2, Vtk 9.2.5, OCC 7.7.0
Locale: English/United Kingdom (en_GB)
[/code]

Syres916, 2024-07-06:
So based on the error creating a Body and opening Preferences it would appear there is an issue parsing the UI file in each case, not sure whether the parser is expecting a `displayname` field for the file. I did try to compile Gentoo probably two years ago but gave up after a few hours so I've not got a VM to do any testing.

trohsb, 2024-07-06:
I rebuild FreeCAD version 0.22.0dev with the following command:

`USE="-python_single_target_python3_12 python_single_target_python3_11 positioning quick qml" emerge -aD --with-bdeps=y media-gfx/freecad`

FreeCAD about:

OS: Gentoo Linux (KDE)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.Unknown
Build type: RelWithDebInfo
Python 3.11.9, Qt 5.15.14, Coin 4.0.2, Vtk 9.2.5, OCC 7.7.0
Locale: English/United Kingdom (en_GB)

Now FreeCAD runs as expected. 

Previously I'd build FreeCAD version 0.22.0dev with qt6 support, rather than qt5, so that I didn't need to fiddle with the python version.

maxwxyz, 2024-07-06:
So could this issue be closed then?

Syres916, 2024-07-06:
To complete this properly https://wiki.freecad.org/Compile_on_Linux#Gentoo should be updated with the command used and a little explanation for future help to others.

trohsb, 2024-07-06:
There is a discussion [here](https://forums.gentoo.org/viewtopic-t-1169467-highlight-freecad.html)

To build FreeCAD from the HEAD of the git repository I have the following in /etc/portage/package.accept_keywords/freecad

`media-gfx/freecad   **`

As well as the following in /etc/portage/package.use/freecad

`media-gfx/freecad -python_single_target_python3_12 python_single_target_python3_11 positioning quick qml
`

closed 2024-07-08 by trohsb