FreeCAD/FreeCAD #15211: Body is always displayed as grey

trilbytim, 2024-07-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Further to this bug report for a previous build where I reported objects not staying the colour they were set to and reverting back to grey after being changed to any other colour: https://github.com/FreeCAD/FreeCAD-Bundle/issues/253
With the latest weekly build it now seems impossible to display a part in any colour at all other than grey. here you can see that the selected body is set to a colour:
[image]
However it still appears as grey:
[image]

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git) AppImage
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * CfdOF 1.25.13
  * OpenTheme 2024.5.3
  * Plot 2022.4.17
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-05:
Is this a duplicate of: https://github.com/FreeCAD/FreeCAD/issues/14779 ?

Related: https://github.com/FreeCAD/FreeCAD/issues/15170

trilbytim, 2024-07-05:
> Is this a duplicate of: #14779 ?
> 
> Related: #15170

Sorry, yes, this is a duplicate of https://github.com/FreeCAD/FreeCAD/issues/14779

closed 2024-07-05 by maxwxyz