FreeCAD/FreeCAD #15218: Regression : error "'NoneType' object has no attribute 'setObjectName' prevent luanching some WB

Grubuntu, 2024-07-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Since the latest weekly build 38001,  there are errors when starting Freecad which prevents some workbenches from working.

`
17:45:16  During initialization the error "'NoneType' object has no attribute 'setObjectName'" occurred in C:\Users\Victus\AppData\Roaming\FreeCAD\Mod\fasteners\.\InitGui.py
17:45:16  Please look into the log file for further information

17:45:17  During initialization the error "'NoneType' object has no attribute 'setObjectName'" occurred in C:\Users\Victus\AppData\Roaming\FreeCAD\Mod\PieMenu\.\InitGui.py
17:45:17  Please look into the log file for further information

17:45:18  During initialization the error "'NoneType' object has no attribute 'setObjectName'" occurred in C:\Users\Victus\AppData\Roaming\FreeCAD\Mod\Reinforcement\.\InitGui.py
17:45:18  Please look into the log file for further information
`

Test case:
Intall last weekly 38001
Install add-on Fasteners, and/or Reinforcement and/or PieMenu
Restart FreeCAD

https://forum.freecad.org/viewtopic.php?t=84101&start=110#p769221
https://forum.freecad.org/viewtopic.php?t=88897

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git)
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: French/France (fr_FR)
Installed mods: 
  * A2plus.backup1694457321.268 0.4.60k
  * Assembly4.backup1694457569.8246403 0.12.6
  * Curves.backup1710863482.8671153 0.6.29
  * fasteners 0.5.24
  * freecad.gears 1.2.0
  * PieMenu 1.7.4
  * Reinforcement
  * sheetmetal 0.4.21
```

### Subproject(s) affected?

Other (specify in description)

### Anything else?

This at least impacts these add-ons: 
Fasteners, 
PieMenu 
Reinforcement 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-07-05:
The commit that caused this issue is https://github.com/FreeCAD/FreeCAD/commit/d57f7a4

I just backed out the four Python files: _DraftGui.py_, _BimViews.py_, _BimIfcExplorer.py_ & _BimStatusBar.py_ and all loads error free. I'm sure there will be a way for the workbenches to load a minimal amount of code and only on Activation ensure that the Draft and Bim toolbars are fully loaded before proceeding, that was my thought anyway for what it's worth.

@kadet1090 have you any inspiration for the workbench authors?

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git)
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * FreeCAD-themes 2024.7.24
  * Reinforcement
```

Grubuntu, 2024-07-05:
A clue perhaps: in FreeCAD_weekly-builds-38001-conda-Windows-x86_64-py311\Mod\Draft\DraftGui.py
 I replaced the line 208
 ` self.tray = FreeCADGui.UiLoader().createWidget("Gui::ToolBar")`
by  `self.tray = QtWidgets.QToolBar()`
```

        # add only a dummy widget, since widgets are created on demand
        self.baseWidget = DraftBaseWidget()
        # self.tray = FreeCADGui.UiLoader().createWidget("Gui::ToolBar")
        self.tray = QtWidgets.QToolBar()
        self.tray.setObjectName("Draft tray")
        self.tray.setWindowTitle("Draft tray")
```
and it works.

kadet1090, 2024-07-05:
That is quite strange... That would suggest some kind of race condition, because it would imply that `FreeCADGui.UiLoader().createWidget("Gui::ToolBar")` returns `None` while it should not. And why code in Draft results in breaking addons, while itself it works correctly? 

Only thing that comes to my mind is that somehow it forces Draft to initialize before ToolBarManager is initialized and so the "Gui::ToolBar" widget is not registered yet. 

@maxwxyz can you assign me?

kadet1090, 2024-07-17:
I cannot reproduce this on Qt6 so it might be a problem with qt5? I have a patch in mind but as I cannot reproduce on my local install I need someone who can compile and test it for me. Any volunteers? @Syres916?

KimK, 2024-07-17:
Compile and test in the usual manner? OK, sure, here you go, recent build.

OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38257 (Git)
Build type: Unknown
Branch: main
Hash: e4c707b98948e695dd8431a5c11f8bbff55287d8
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * PieMenu 1.7.5
  * Reinforcement
  * fasteners 0.5.24

I installed Fasteners first, then PieMenu, then Reinforcement.
With just Fasteners and PieMenu, everything seems to work OK. 

With Reinforcement installed, I get the error posted at the end (much like the three errors reported two weeks ago in the original post. )

Using the WB pulldown, I can select Fasteners. But Reinforcement does not appear on the list. 

That's all I found. Error copy-and-pastes follow.
Does this help at all?

(Error on startup with Reinforcement installed):
During initialization the error "'NoneType' object has no attribute 'setObjectName'" occurred in /home/zack/.local/share/FreeCAD/Mod/Reinforcement/./InitGui.py
Please look into the log file for further information
connect failed: No such file or directory

(Occasional error on shutdown with Reinforcement installed (maybe once out of 3 or 4 shutdowns): 
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x42520) [0x7fec78c0c520]
#1  /lib/x86_64-linux-gnu/libshiboken2.cpython-310-x86_64-linux-gnu.so.5.15(+0x18142) [0x7fec7793f142]
#2  0x7fec779498e7 in Shiboken::BindingManager::retrieveWrapper(void const*) from /lib/x86_64-linux-gnu/libshiboken2.cpython-310-x86_64-linux-gnu.so.5.15+0x17
#3  /lib/x86_64-linux-gnu/libpyside2.cpython-310-x86_64-linux-gnu.so.5.15(+0x1df09) [0x7fec7805ff09]
#4  /lib/x86_64-linux-gnu/libpyside2.cpython-310-x86_64-linux-gnu.so.5.15(+0x1b18b) [0x7fec7805d18b]
#5  /lib/x86_64-linux-gnu/libQt5Core.so.5(+0x2f9fec) [0x7fec7941ffec]
#6  0x7fec7942068b in QVariant::~QVariant() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x4b
#7  0x7fec7940c6cc in QObjectPrivate::~QObjectPrivate() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0x31c
#8  0x7fec7940c7bd in QObjectPrivate::~QObjectPrivate() from /lib/x86_64-linux-gnu/libQt5Core.so.5+0xd
#9  /lib/x86_64-linux-gnu/libQt5Gui.so.5(+0x1cc30d) [0x7fec7985130d]
#10  /lib/x86_64-linux-gnu/libc.so.6(+0x45495) [0x7fec78c0f495]
#11  /lib/x86_64-linux-gnu/libc.so.6(on_exit+0) [0x7fec78c0f610]
#12  /lib/x86_64-linux-gnu/libc.so.6(+0x29d97) [0x7fec78bf3d97]
#13  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x80) [0x7fec78bf3e40]
#14  bin/FreeCAD(+0x2cfc5) [0x56009092cfc5]

KimK, 2024-07-17:
I could try again with just Reinforcement installed, and leave Fasteners and PieMenu out of the mix, if you think that would help?

kadet1090, 2024-07-17:
For me having PieMenu installed results in segfault on start:
```
Program received signal SIGSEGV, Segmentation fault.
#0  /usr/lib/libc.so.6(+0x3cae0) [0x79ecd6850ae0]
#1  /usr/lib/libQt6Widgets.so.6(+0x10b45f) [0x79ecd7f0b45f]
#2  0x79ecd7f0c848 in QBoxLayout::minimumSize() const from /usr/lib/libQt6Widgets.so.6+0x38
#3  /usr/lib/python3.12/site-packages/PySide6/QtWidgets.cpython-312-x86_64-linux-gnu.so(+0x5e3ce6) [0x79eca25e3ce6]
#4  0x79ecd7f2b802 in QLayout::totalMinimumSize() const from /usr/lib/libQt6Widgets.so.6+0x32
#5  /usr/lib/python3.12/site-packages/PySide6/QtWidgets.cpython-312-x86_64-linux-gnu.so(+0x5e7aa6) [0x79eca25e7aa6]
#6  0x79ecd7f2792c in qSmartMinSize(QWidget const*) from /usr/lib/libQt6Widgets.so.6+0x4c
#7  0x79ecd7f27a9e in QStackedLayout::minimumSize() const from /usr/lib/libQt6Widgets.so.6+0x5e
#8  0x79ecd7f2b802 in QLayout::totalMinimumSize() const from /usr/lib/libQt6Widgets.so.6+0x32
#9  0x79ecd8136724 in QTabWidget::minimumSizeHint() const from /usr/lib/libQt6Widgets.so.6+0xb4
#10  /usr/lib/python3.12/site-packages/PySide6/QtWidgets.cpython-312-x86_64-linux-gnu.so(+0x55071e) [0x79eca255071e]
#11  0x79ecd7f2792c in qSmartMinSize(QWidget const*) from /usr/lib/libQt6Widgets.so.6+0x4c
#12  /usr/lib/libQt6Widgets.so.6(+0x3079e4) [0x79ecd81079e4]
#13  /usr/lib/libQt6Widgets.so.6(+0x308732) [0x79ecd8108732]
#14  /usr/lib/python3.12/site-packages/PySide6/QtWidgets.cpython-312-x86_64-linux-gnu.so(+0x494038) [0x79eca2494038]
#15  /usr/lib/libpython3.12.so.1.0(+0x1b0aac) [0x79ecd91b0aac]
#16  /usr/lib/libpython3.12.so.1.0(_PyObject_MakeTpCall+0x2fb) [0x79ecd9180abb]
#17  /usr/lib/libpython3.12.so.1.0(_PyEval_EvalFrameDefault+0x84f) [0x79ecd918931f]
#18  /usr/lib/libpython3.12.so.1.0(PyEval_EvalCode+0xb5) [0x79ecd924d0f5]
#19  /usr/lib/libpython3.12.so.1.0(+0x268706) [0x79ecd9268706]
#20  /usr/lib/libpython3.12.so.1.0(_PyEval_EvalFrameDefault+0x3d09) [0x79ecd918c7d9]
#21  /usr/lib/libpython3.12.so.1.0(PyEval_EvalCode+0xb5) [0x79ecd924d0f5]
#22  /usr/lib/libpython3.12.so.1.0(+0x2703ea) [0x79ecd92703ea]
#23  /usr/lib/libpython3.12.so.1.0(+0x26b2ef) [0x79ecd926b2ef]
#24  /usr/lib/libpython3.12.so.1.0(PyRun_StringFlags+0x76) [0x79ecd925e586]
#25  0x79ecd9ee4352 in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /home/kadet/workspace/FreeCAD/build/lib/libFreeCADBase.so+0x1d2
#26  0x79ecdc4553c6 in Gui::Application::runInitGuiScript() from /home/kadet/workspace/FreeCAD/build/lib/libFreeCADGui.so+0x46
#27  0x79ecdc6933a7 in Gui::StartupPostProcess::showMainWindow() from /home/kadet/workspace/FreeCAD/build/lib/libFreeCADGui.so+0x77
#28  0x79ecdc69185c in Gui::StartupPostProcess::execute() from /home/kadet/workspace/FreeCAD/build/lib/libFreeCADGui.so+0x7c
#29  0x79ecdc4555a5 in Gui::Application::runApplication() from /home/kadet/workspace/FreeCAD/build/lib/libFreeCADGui.so+0x1b5
#30  ./bin/FreeCAD(+0x2337d) [0x558a36d1637d]
#31  /usr/lib/libc.so.6(+0x25c88) [0x79ecd6839c88]
#32  /usr/lib/libc.so.6(__libc_start_main+0x8c) [0x79ecd6839d4c]
#33  ./bin/FreeCAD(+0x207b5) [0x558a36d137b5]
```

nontheless, please try this patch:
```
diff --git a/src/Gui/ToolBarManager.cpp b/src/Gui/ToolBarManager.cpp
index eb8c4d50f4..37153f5e0d 100644
--- a/src/Gui/ToolBarManager.cpp
+++ b/src/Gui/ToolBarManager.cpp
@@ -407,8 +407,6 @@ ToolBarManager::ToolBarManager()
     setupConnection();
     setupTimer();
     setupMenuBarTimer();
-
-    setupWidgetProducers();
 }
 
 ToolBarManager::~ToolBarManager() = default;
@@ -524,11 +522,6 @@ void ToolBarManager::setupMenuBarTimer()
     });
 }
 
-void Gui::ToolBarManager::setupWidgetProducers()
-{
-    new WidgetProducer<Gui::ToolBar>;
-}
-
 ToolBarArea ToolBarManager::toolBarArea(QWidget *widget) const
 {
     if (auto toolBar = qobject_cast<QToolBar*>(widget)) {
diff --git a/src/Gui/resource.cpp b/src/Gui/resource.cpp
index ee3ff5d8ab..ce6b3f230d 100644
--- a/src/Gui/resource.cpp
+++ b/src/Gui/resource.cpp
@@ -131,5 +131,6 @@ WidgetFactorySupplier::WidgetFactorySupplier()
     new WidgetProducer<Gui::DoubleSpinBox>;
     new WidgetProducer<Gui::QuantitySpinBox>;
     new WidgetProducer<Gui::ExpLineEdit>;
+    new WidgetProducer<Gui::ToolBar>;
 }
 // clang-format on
```

and check if it is working correctly

KimK, 2024-07-18:
My apologies, I'm still learning. Because I was delayed by other events, I made what turned out to be the mistake of updating and rebuilding, instead of leaving my build the same. So now I get "patch does not apply". So it looks like somebody may have already pushed a fix for this? Sorry, I'll know better next time.

KimK, 2024-07-19:
OK, updated and rebuilt yet again, here we go: 

OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38260 (Git)
Build type: Unknown
Branch: main
Hash: ff61f56d19f65dcb9b8cb4b4768d54c28346c359
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * PieMenu 1.7.5
  * Reinforcement
  * fasteners 0.5.24

Error on startup: 
During initialization the error "'NoneType' object has no attribute 'setObjectName'" occurred in /home/zack/.local/share/FreeCAD/Mod/Reinforcement/./InitGui.py
Please look into the log file for further information

As before, Fasteners and PieMenu appear to work normally, but Reinforcement does not appear on the WB pulldown, so it can't be selected. Shall I uninstall Fasteners and PieMenu and try again?

KimK, 2024-07-22:
And 21 upstream commits later, let's sync and rebuild, and try again. 
OK, here we go: 
OS: Linux Mint 21.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38281 (Git)
Build type: Unknown
Branch: main
Hash: 187296d67ae546a80fe83ff1c05af19688cb1fa4
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * PieMenu 1.7.5
  * Reinforcement
  * fasteners 0.5.24

And once again, Fasteners and PieMenu work, Reinforcement does not work. Here's the log:

Msg: FreeCAD 0.22.0, Libs: 0.22.0devR38281 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Log: Time = 2024-Jul-22 05:54:42
Log: AboutImage = freecadabout
Log: AppDataSkipVendor = true
Log: AppHomePath = /home/zack/FreeCAD-KimK/build/
Log: AppIcon = freecad
Log: AppTempPath = /tmp/
Log: BOOST_VERSION = 1_74
Log: BinPath = /home/zack/FreeCAD-KimK/build/bin/
Log: BuildRepositoryURL = git://github.com/KimK/FreeCAD.git main
Log: BuildRevision = 38281 (Git)
Log: BuildRevisionBranch = main
Log: BuildRevisionDate = 2024/07/22 05:08:51
Log: BuildRevisionHash = 187296d67ae546a80fe83ff1c05af19688cb1fa4
Log: BuildVersionMajor = 0
Log: BuildVersionMinor = 22
Log: BuildVersionPoint = 0
Log: BuildVersionSuffix = dev
Log: Console = 0
Log: CopyrightInfo = © Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

Log: Debug = 0
Log: DocPath = /home/zack/FreeCAD-KimK/build/doc/
Log: EIGEN_VERSION = 3.4.0
Log: ExeName = FreeCAD
Log: ExeVendor = FreeCAD
Log: ExeVersion = 0.22.0
Log: LoggingConsole = 1
Log: LoggingFile = 1
Log: LoggingFileName = /home/zack/.local/share/FreeCAD/FreeCAD.log
Log: MaintainerUrl = https://www.freecad.org/wiki/Main_Page
Log: OCC_VERSION = 7.6.3
Log: PATH = /home/zack/depot_tools:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/home/zack/depot_tools:/home/zack/scripts:/snap/bin
[... 189 lines cut ...]
Log: ----------------------------------------------------------------------------------------------------
Err: During initialization the error "'NoneType' object has no attribute 'setObjectName'" occurred in /home/zack/.local/share/FreeCAD/Mod/Reinforcement/./InitGui.py
Err: Please look into the log file for further information
Log: Init:      Initializing /home/zack/.local/share/FreeCAD/Mod/fasteners/./... done
Log: All modules with GUIs using InitGui.py are now initialized
Log: All modules with GUIs initialized using pkgutil are now initialized
Log: Init: Loading FreeCAD GUI
Log: Ignoring unknown NoneWorkbench workbench found in user preferences.
Log: Init: Running FreeCADGuiInit.py start script... done
Log: Init: Activating default workbench PartDesignWorkbench
Log: Loading Sketcher module... done
Log: Loading PartDesign module... done
Log: Loading GUI of Sketcher module... done
Log: Loading GUI of PartDesign module... done
Log: Init: Showing main window
Log: Main window restored
Log: Show main window
Log: Toolbars restored
Log: Couldn't connect to spacenav daemon. Please ignore if you don't have a spacemouse.
Log: Init: Entering event loop
Log: Init: Executing event loop...
Log: Init: Processing command line files
Log: Getting resources
Log: Hide main window
Log: Show main window

krushia, 2024-07-30:
For some reason uninstalling PieMenu fixes fasteners for me.

luzpaz, 2024-07-31:
There a quite a few Piemenu related bugs in this bugracker atm

Grubuntu, 2024-07-31:
What is surprising is that other versions of FreeCAD (Ondsel, 0.21, Link and 0.22 before 38001) are not affected by this problem.

Grubuntu, 2024-08-01:
It seems that it's the DraftGui translate import `from DraftGui import translate` that's causing the problem, so I've removed it from the PieMenu code and that seems to solve the problem for PieMenu and Fasteners.

But for "Reinforcement" it's not solved: even after removing PieMenu and any other add-on, "Reinforcement" still crashes on start-up...

I can close this issue if you agree. Perhaps we should open an issue on the Reinforcement github?

closed 2024-08-02 by luzpaz