FreeCAD/FreeCAD #15221: One assembly inside of another

leoheck, 2024-07-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

I put one assembly inside another, but this is a cyclic dependency, however I cannot undo this. As soon as I try to do anything to fix this, Freecad Breaks. I also tried to disable recomputions but it continues breaking.

```
➜ freecad-daily 
Latest FreeCAD /home/lheck/Applications/FreeCAD_weekly-builds-38001-conda-Linux-x86_64-py311.AppImage
QSocketNotifier: Can only be used with threads started with QThread
FreeCAD 0.22.0, Libs: 0.22.0devR38001 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2024
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####  
  #                    #      # #   #   # 
  #     ##  #### ####  #     #   #  #   # 
  ####  # # #  # #  #  #     #####  #   # 
  #     #   #### ####  #    #     # #   # 
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

2.4e-08 <App> Document.cpp(2527): The graph must be a DAG.
0.0863821 SoFCUnifiedSelection.cpp(1527): Cyclic scene graph: test#Assembly002
/tmp/.mount_FreeCA1MNUl1/AppRun: line 41: 35242 Segmentation fault      (core dumped) "${MAIN}" "$@"
```

[test.zip](https://github.com/user-attachments/files/16112661/test.zip)

### Full version info

```shell
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git) AppImage
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * POV-Ray-Rendering
  * Assembly4 0.50.12
  * InventorLoader 1.4.0
  * ose-piping
  * OpenTheme 2024.7.3
  * SheetMetal 0.4.20
  * fasteners 0.5.24
  * asm4_disable_version_check_popup
  * Render 2024.7.4
  * OpticsWorkbench 1.0.18
  * lattice2 1.0.0
  * kicadStepUpMod
```

### Subproject(s) affected?

Assembly

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-05:
Can confirm, dragging crashed FreeCAD
@PaddleStroke FYI

adrianinsaval, 2024-07-05:
note that the file has a dependency loop

leoheck, 2024-07-06:
@adrianinsaval yes that's the issue. How do you fix it if Freecad keeps crashing because of that? Also, this situation could be prevented by the Assembly itself.

adrianinsaval, 2024-07-06:
your issue is legitimate and this should be fixed in freecad. I'm just noting it for information completeness here

PaddleStroke, 2024-07-08:
[image]
Indeed there's a loop.
Can you please advise how you made this in the first place? So we can prevent it from happening.

PaddleStroke, 2024-07-08:
I found one way this can crash : 
- Create Assembly
- Create second assembly (it will be added as child of the first)
- Toggle the second assembly active
- Insert Component
- Attempt to insert the first assembly -> crash

Was that how you ended up in this situation?
Else this situation cannot be made manually by dragging in the tree as trying to drag in the tree will be averted and an error message shown.

leoheck, 2024-07-08:
> Was that how you ended up in this situation?

I think so. If I remember I tried to add the root Assembly into the nested one.

closed 2024-07-15 by chennes