FreeCAD/FreeCAD #15222: Nautilus does not know the .FCStd extension anymore.

leoheck, 2024-07-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Nautilus does not know the FCStd extension anymore.

[image]

### Full version info

```shell
OS: Ubuntu 24.04 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git) AppImage
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * POV-Ray-Rendering
  * Assembly4 0.50.12
  * InventorLoader 1.4.0
  * ose-piping
  * OpenTheme 2024.7.3
  * SheetMetal 0.4.20
  * fasteners 0.5.24
  * asm4_disable_version_check_popup
  * Render 2024.7.4
  * OpticsWorkbench 1.0.18
  * lattice2 1.0.0
  * kicadStepUpMod
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2024-07-05:
sounds like a nautilus issue not a freecad issue. The appimage doesn't register file associations

luzpaz, 2024-07-05:
Is there an upstream ticket ?

adrianinsaval, 2024-07-06:
could also be something coming from appimagelauncher, if we had more details about what's the mechanism at play here we could see if there's something we can do on our end

leoheck, 2024-07-06:
this is a regression.

adrianinsaval, 2024-07-06:
how was it before? what registered fcstd files and how? I can't do anything if you don't give me information. Like I said before this is not stuff we are doing ourselves.

leoheck, 2024-07-06:
I can give extra information, I just don't know what do you need as information. 

Before, it used to display the Freecad icon on `.FCStd` files. Also, `.FCbak` could have a defined icon. 

Perhaps this issue belongs to the Freecad-Bundle instead. The regression may be introduced when the new logo was launched. But I am not sure.

maxwxyz, 2024-07-06:
Do you have the stable version installed too or had it in the past?

leoheck, 2024-07-06:
had it installed some weeks ago.

furgo16, 2024-07-06:
In case it helps: https://askubuntu.com/questions/1421738/nautilus-changing-a-file-icon-from-command-line

furgo16, 2024-07-06:
The FreeCAD package from the Ubuntu archives ships the custom icon: https://packages.ubuntu.com/jammy/all/freecad-common/filelist. Around the end of the list, see:

```
/usr/share/freecad/share/icons/hicolor/scalable/mimetypes/application-x-extension-fcstd.svg
```

furgo16, 2024-07-06:
In summary, I think:
- The custom icon is shipped on installable packages. I'm not familiar enough with AppImage to tell if it's something it also does or should do.
- The shipped custom icon has the right FreeCAD logo
- The shipped custom icon is failing to register with the `.FCStd` extension

This applies to .snap and .deb packages, as the issue description hints at the host system being Ubuntu.

leoheck, 2024-07-06:
I also don't like this extension format. It is wierd  It also has upper case and lower case characters. It is horrible.

nazar-pc, 2024-07-08:
This is packaging-related issue that AppImage can't solve directly since it is not installed, but FreeCAD can hypothetically add a menu entry that would automate described below for those using AppImage (only updates, at least for dev builds, will be problematic due to unique name used for downloaded file rather than replacing existing file instead).

Basically to fix this you need to do the following steps.

### Create desktop shortcut for FreeCAD

Create file `~/.local/share/applications/freecad.desktop` (and necessary directories if they don't exist) with following contents:
```ini
[Desktop Entry]
Version=1.0
Type=Application
Name=FreeCAD
Icon=freecad
Exec="/home/user/.local/bin/FreeCAD.AppImage" %f
Terminal=false
```

`/home/user/.local/bin/FreeCAD.AppImage` should point to your downloaded AppImage.

Now you'll see FreeCAD among installed apps and will be able to launch it like anything else.

### Register mime type

For FreeCAD files to be recognized as something other than simply a ZIP file, separate mime type association needs to be created.

Create file `~/.local/share/mime/packages/org.freecadweb.FreeCAD.xml` (and necessary directories if they don't exist) with following contents (mostly taken from `freecad-common` file of old Ubuntu package):
```xml
<?xml version="1.0" encoding="UTF-8"?>
<mime-info xmlns='http://www.freedesktop.org/standards/shared-mime-info'>
  <mime-type type="application/x-extension-fcstd">
    <!-- <sub-class-of type="application/zip"/> -->
    <comment>FreeCAD document files</comment>
    <glob pattern="*.fcstd"/>
    <glob pattern="*.fcbak"/>
    <icon name="org.freecadweb.FreeCAD"/>
  </mime-type>
</mime-info>
```

And update mime database:
```bash
update-mime-database ~/.local/share/mime
```

Now FreeCAD files will be recognized as a separate type and assuing the theme you're using has icons, you will get a dedicated icon for those files as well (I use Numix Circle theme and it does have icon for FreeCAD) or you may have to place one at `~/.local/share/icons/hicolor/scalable/apps/freecad.svg`.

### Create association between FreeCAD and new type of files.

Simpy try to open the file with file manager (Nautilus, Nemo, etc.), choose FreeCAD from the list of apps and click "Set as default" to open this file type with FreeCAD going forward. Physically this will add `application/x-extension-fcstd=freecad.desktop` line to both `[Default Applications]` and `[Added Associations]` sections of `~/.config/mimeapps.list` file.

leoheck, 2024-07-08:
I use the `AppImageLauncher` which does this for us.

Here is the comment from the Author
https://github.com/TheAssassin/AppImageLauncher/issues/275#issuecomment-564818880

This is his first sentence:
> File associations are supported and work fine. There are some prominent examples like FreeCAD or Ultimaker Cura (which even supports STL files out of the box!) which demonstrate this.

This thing was working fine using the `AppImageLauncher`, now it does not work anymore for some reason.

adrianinsaval, 2024-07-08:
the desktop file we ship defines supported mimetypes https://github.com/FreeCAD/FreeCAD/blob/de508e437d13e7eb35baffceea496fb9107888cb/src/XDGData/org.freecad.FreeCAD.desktop#L24C10-L24C21 so I don't know what would be missing here.