FreeCAD/FreeCAD #15223: CAM: Sketch change breaks drilling operation

jkvfc, 2024-07-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

If the plate length is changed, for example, from 200 to 150, the drilling operation is not updated.
[image]
[image]
The following error message is displayed:
[image]
There is no problem in FreeCAD 0.21.1!

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38039 (Git)
Build type: Unknown
Branch: main
Hash: 7a9fd501c3ea0e259f843dc21f488ef6af43c201
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * MyScripts
```

### Subproject(s) affected?

CAM/Path

### Anything else?

[PlateParametricManufacturing.zip](https://github.com/user-attachments/files/16113134/PlateParametricManufacturing.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Syres916, 2024-07-05:
@shaise I assume https://github.com/FreeCAD/FreeCAD/pull/15177 will fix this?

Syres916, 2024-07-06:
This appears to be a more indepth issue. When the Drilling operation is created all the Base Geometry is shown as expected and the path generated is correct but without actually changing anything, just Mark the document to Recompute the list of geometry is  created incorrectly.

The output of the list should be `('Face5', 'Face6', 'Face7', 'Face8')` but it's actually showing `('?Face5', '?Face6', '?Face7', '?Face8')` hence the `getElement` is returning an invalid response, garbage in garbarge out scenario.

shaise, 2024-07-06:
when a '?FaceXX' is returned it means the TNP resolver failed to identify the faces.
Please try the same scenario on v0.21. If it works there, its definitely a TNP issue and must be fixed. If TNP code is not working always that`s OK but if it works worse then without it, then its really bad.

Syres916, 2024-07-06:
Yes it's a TNP issue and let's be polite and say there's still a lot of work to do before it's going to be acceptable for release.

Sorry I didn't realise the question mark at the start of the geometry was a documented TNP resolver issue, I thought it was just a string that hadn't been 'cleaned' up properly.

Syres916, 2024-07-07:
Branch for CAM Topo Tests https://github.com/FreeCAD/FreeCAD/compare/main...Syres916:FreeCAD:CAM_Topo_Tests , just one drilling completed, they say the first is the most difficult!!

Note: this was created with https://github.com/FreeCAD/FreeCAD/pull/15177 and https://github.com/FreeCAD/FreeCAD/pull/15251 compiled in the test build but as noted in the test comments the face/edge/vertex lookups still fail and move two holes outside of the plate. If the Recompute on line 196 is commented out the holes are kept in the correct location.

I'll work my way through other operations probably later next week, I have to get my staircase progressing before it turns into a health and safety issue.

```
OS: Linux Mint 20.3 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38049 (Git)
Build type: Release
Branch: main
Hash: 03db773181f7f12f3918f7cc6dff835c1dfaf9a5
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.8.1
Locale: English/United Kingdom (en_GB)
Installed mods: 
  * FreeCAD-themes 2024.7.24
  * CfdOF 1.27.3
```

Syres916, 2024-07-13:
ProfileTopo test added to branch above

maxwxyz, 2024-07-15:
@Syres916 do you work on this? I'm assigning this to you, if this is wrong, feel free to remove yourself.

bgbsww, 2024-08-11:
At the risk of pointing in the wrong direction, it looks to me like the breakage happens at the Model-Body  Clone, so draftobjects.clone.Clone not correctly getting updated TNP names.

You can cause this just by opening the document, and recomputing the Pad, no sketch changing required.

The missing element marks are getting set by a python call in PropertyPartShape::setValue inside the execute of a Feature implemented in python, from the recompute.  What we aren't matching up looks like this:

bgbsww, 2024-08-12:
It's none of the above:  this appears to affect only files saved with bad versions of 0.22, earlier on in the release cycle.  Unfortunately, due to the way that clone works, and Drilling refers to the clone, this cannot be corrected with a file recompute; that will result in the Drilling.Base property referring to missing elements.  In this case, it becomes ?Face5, ?Face6, ?Face7, ?Face8  for the four drilled holes.

If you replace this property in python with Face5, Face6, Face7, Face8   (plus the correct surrounding tuple entries) then the file is repaired and works normally from that point.

So I think the bugs that caused this have been solved, however we don't have a user friendly way to repair files that were written with this error in them.  First case of that.  I am exploring if that can be an option, but no guarantees.

@maxwxyz This needs confirmation that it is fixed on files not saved in early 0.22, and if that's the case, it could close.  I do think we want @Syres916 work on new tests though - that's valuable.

bgbsww, 2024-08-12:
Oh, and there may be an unrelated bug here in how the stock reacts if you make the plate larger.  But I'm not sure what the expected behavior should be.

jkvfc, 2024-08-12:
@bgbsww it seems you are absolutely right. I have reconstructed the plate from scratch with the newest version
OS: Ubuntu 22.04.4 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38434 (Git)
Build type: Unknown
Branch: main
Hash: d62444e230a0202056956a8650b93f104b80c1ec
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.8.1
Locale: English/United States (en_U
[PlateParametricManufacturingNew.zip](https://github.com/user-attachments/files/16588857/PlateParametricManufacturingNew.zip)
If the plate's length is changed from 200 to 150 everything looks fine. Concerning stock, you have to double click stock and then choose refresh. Thanks a lot for your awesome work on TNP!
Do you know why the following message appears in the report view if the attached file is loaded?
[image]

bgbsww, 2024-08-12:
It's there because something is wrong.  I'm seeing it too, not sure why yet, but I'll get it.

bgbsww, 2024-08-13:
Hasher table was fixed in #15869.  I'm still looking at if there's a way to add code to make the repair ( but not automatically ), but I think this bug is done.

maxwxyz, 2024-08-13:
I'll close this then. Let me know if I've misunderstood something.

maxwxyz, 2024-10-14:
Reopened as reported here: https://forum.freecad.org/viewtopic.php?p=787041&sid=d38aa5e96832aa9614c43616ba28d1b5#p787041

[image]

jkvfc, 2024-10-15:
I repeated the tests concerning this issue with the newest version of FreeCAD (see below). For this purpose I constructed the plate with version 0.21.2 and 1.1.0dev.
[PlateParametricManufacturing0-21-2.zip](https://github.com/user-attachments/files/17377437/PlateParametricManufacturing0-21-2.zip)
[PlateParametricManufacturing1-1-0dev.zip](https://github.com/user-attachments/files/17377445/PlateParametricManufacturing1-1-0dev.zip)
I was able to construct and change the plate in both versions flawlessly and moreover the 0.21.2 version could be smoothly imported in version 1.1.0dev. Even a 'Recomputation required' message was issued. 
I therefore believe that @maxwxyz was right to close this issue.

OS: Ubuntu 24.04.1 LTS (KDE/plasma/xcb)
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.38977 (Git)
Build type: Unknown
Branch: main
Hash: facd8ca6e0303cf96297a7a7398185770c1a1e0b
Python 3.12.3, Qt 5.15.13, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion

Syres916, 2024-10-16:
@luzpaz I'll build a 1.0 branch build this weekend and make sure that all the appropriate backports have been completed, the logic that it was correct to close this in August weeks before RC2 was released makes absolutely no sense to me, just because it was working in August doesn't mean another regression could not have taken place since and what's to say it's not a Windows/Conda specific issue.

Therefore I'll build in effect a 1.0RC3 locally and retest as some backports have been held back for good reason.

luzpaz, 2024-10-16:
Thanks for your explanation. I'll re-open this in that case.

Syres916, 2024-10-20:
My conclusion carrying out really simple tests, it's fine if the component does not have any changes to the number of holes particularly if they are reduced. On that change the sea of red in the Report View is off the charts:

```
10:36:28  <PropertyLinks> PropertyLinks.cpp(447): Drilling_1_0_Test#Drilling.Base missing element reference Drilling_1_0_Test#Clone2D ;g16;SKT;:H67b,E.Edge12
10:36:28  pyException: Traceback (most recent call last):
  File "/home/username/freecad-rc3-build/Mod/CAM/Path/Op/Base.py", line 527, in onChanged
    if prop == "Base" and self.sanitizeBase(obj):
                          ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/username/freecad-rc3-build/Mod/CAM/Path/Op/Base.py", line 709, in sanitizeBase
    o.Shape.getElement(sub)
<class 'ValueError'>: {'sclassname': 'N4Base10ValueErrorE', 'sErrMsg': 'Invalid shape name ?Edge12', 'sfile': '/home/username/freecad-rc3-source/src/Mod/Part/App/TopoShapeExpansion.cpp', 'iline': 2019, 'sfunction': 'Part::TopoShape Part::TopoShape::getSubTopoShape(const char*, bool) const', 'swhat': 'Invalid shape name ?Edge12', 'btranslatable': False, 'breported': False}
10:36:28  pyException: Traceback (most recent call last):
  File "/home/username/freecad-rc3-build/Mod/CAM/Path/Op/Base.py", line 531, in onChanged
    self.updateDepths(obj, True)
  File "/home/username/freecad-rc3-build/Mod/CAM/Path/Op/Base.py", line 663, in updateDepths
    fbb = base.Shape.getElement(sub).BoundBox
          ^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'ValueError'>: {'sclassname': 'N4Base10ValueErrorE', 'sErrMsg': 'Invalid shape name ?Edge12', 'sfile': '/home/username/freecad-rc3-source/src/Mod/Part/App/TopoShapeExpansion.cpp', 'iline': 2019, 'sfunction': 'Part::TopoShape Part::TopoShape::getSubTopoShape(const char*, bool) const', 'swhat': 'Invalid shape name ?Edge12', 'btranslatable': False, 'breported': False}
10:36:28  pyException: Traceback (most recent call last):
  File "/home/username/freecad-rc3-build/Mod/CAM/Path/Op/Base.py", line 531, in onChanged
    self.updateDepths(obj, True)
  File "/home/username/freecad-rc3-build/Mod/CAM/Path/Op/Base.py", line 663, in updateDepths
    fbb = base.Shape.getElement(sub).BoundBox
          ^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'ValueError'>: {'sclassname': 'N4Base10ValueErrorE', 'sErrMsg': 'Invalid shape name ?Edge12', 'sfile': '/home/username/freecad-rc3-source/src/Mod/Part/App/TopoShapeExpansion.cpp', 'iline': 2019, 'sfunction': 'Part::TopoShape Part::TopoShape::getSubTopoShape(const char*, bool) const', 'swhat': 'Invalid shape name ?Edge12', 'btranslatable': False, 'breported': False}
10:36:28  pyException: Traceback (most recent call last):
  File "/home/username/freecad-rc3-build/Mod/CAM/PathScripts/PathUtils.py", line 70, in new_function
    res = function(*args, **kwargs)
          ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/username/freecad-rc3-build/Mod/CAM/Path/Op/Base.py", line 748, in execute
    self.sanitizeBase(obj)
  File "/home/username/freecad-rc3-build/Mod/CAM/Path/Op/Base.py", line 709, in sanitizeBase
    o.Shape.getElement(sub)
<class 'ValueError'>: {'sclassname': 'N4Base10ValueErrorE', 'sErrMsg': 'Invalid shape name ?Edge12', 'sfile': '/home/username/freecad-rc3-source/src/Mod/Part/App/TopoShapeExpansion.cpp', 'iline': 2019, 'sfunction': 'Part::TopoShape Part::TopoShape::getSubTopoShape(const char*, bool) const', 'swhat': 'Invalid shape name ?Edge12', 'btranslatable': False, 'breported': False}
10:36:28  Drilling: {'sclassname': 'N4Base10ValueErrorE', 'sErrMsg': 'Invalid shape name ?Edge12', 'sfile': '/home/username/freecad-rc3-source/src/Mod/Part/App/TopoShapeExpansion.cpp', 'iline': 2019, 'sfunction': 'Part::TopoShape Part::TopoShape::getSubTopoShape(const char*, bool) const', 'swhat': 'Invalid shape name ?Edge12', 'btranslatable': False, 'breported': False}
10:36:28  pyException: Traceback (most recent call last):
  File "/home/username/freecad-rc3-build/Mod/CAM/Path/Op/Base.py", line 531, in onChanged
    self.updateDepths(obj, True)
  File "/home/username/freecad-rc3-build/Mod/CAM/Path/Op/Base.py", line 663, in updateDepths
    fbb = base.Shape.getElement(sub).BoundBox
          ^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'ValueError'>: {'sclassname': 'N4Base10ValueErrorE', 'sErrMsg': 'Invalid shape name ?Edge12', 'sfile': '/home/username/freecad-rc3-source/src/Mod/Part/App/TopoShapeExpansion.cpp', 'iline': 2019, 'sfunction': 'Part::TopoShape Part::TopoShape::getSubTopoShape(const char*, bool) const', 'swhat': 'Invalid shape name ?Edge12', 'btranslatable': False, 'breported': False}
10:36:28  pyException: Traceback (most recent call last):
  File "/home/username/freecad-rc3-build/Mod/CAM/Path/Op/Base.py", line 531, in onChanged
    self.updateDepths(obj, True)
  File "/home/username/freecad-rc3-build/Mod/CAM/Path/Op/Base.py", line 663, in updateDepths
    fbb = base.Shape.getElement(sub).BoundBox
          ^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'ValueError'>: {'sclassname': 'N4Base10ValueErrorE', 'sErrMsg': 'Invalid shape name ?Edge12', 'sfile': '/home/username/freecad-rc3-source/src/Mod/Part/App/TopoShapeExpansion.cpp', 'iline': 2019, 'sfunction': 'Part::TopoShape Part::TopoShape::getSubTopoShape(const char*, bool) const', 'swhat': 'Invalid shape name ?Edge12', 'btranslatable': False, 'breported': False}
10:36:28  pyException: Traceback (most recent call last):
  File "/home/username/freecad-rc3-build/Mod/CAM/PathScripts/PathUtils.py", line 70, in new_function
    res = function(*args, **kwargs)
[... cut ...]
Build type: Release
Branch: releases/FreeCAD-1-0
Hash: 24bbd0769e1b6890169380fff8cc453517e1e94a
Python 3.12.3, Qt 5.15.13, Coin 4.0.3, Vtk 9.1.0, OCC 7.8.2.dev
Locale: English/United Kingdom (en_GB)
Stylesheet/Theme/QtStyle: Behave-dark.qss/Behave-dark/Qt default
Installed mods: 
  * FreeCAD-themes 2024.9.14
  * Silk 0.1.5
  * sheetmetal 0.4.24
  * Behave-Dark-Colors 0.1.1
  * CfdOF 1.27.5
  * 3dfindit-freecad-integration 1.2.0
  * fasteners 0.5.29
  * freecad.gears 1.3.0
  * Curves 0.6.48
  * A2plus 0.4.68
```

jkvfc, 2024-10-21:
@Syres916 you are quite right. There is a problem if you a) delete a hole or b) if  holes comes to lie partially or completely outside of the plate.
a) If I delete a hole I get the error message below and all holes disappear (see  below).
b) If holes are partially or completely outside the plate, I get a bunch of error messages similar to yours.

But I still think that this issue should be closed, as the original cause of this issue was a TNP problem and this has been solved. I propose to open a new issue for a) and b). I don't know what you, @sliptonic and others think about these problems. I mean for real life both changes should be made impossible.

```
13:53:58  Base.ERROR: Parent job Job doesn't have a base object
```

[image]

sslupsky, 2024-11-01:
I have run into the "?FaceNN" issue several times with RC1, RC2 and the weekly 1.1.0dev builds.  Not sure the TNP problem is completely fixed.  Next time I run into it I'll grab the errors and come back and post to this issue.

Also having many recompute issues when using CAM.  I will create a new issue for that.

maxwxyz, 2024-11-01:
Known issue which will be addressed post 1.0

luzpaz, 2024-11-01:
@maxwxyz is this a duplicate then ?

maxwxyz, 2024-11-01:
Not sure if it is the same for CAM here. https://github.com/FreeCAD/FreeCAD/pull/16803

drwho495, 2025-07-26:
could you try this with the string hasher disabled?

to do this, select the document then in the property view disable the property called "string hasher"

drwho495, 2025-08-21:
issue is no longer present in latest daily (i needed to recompute the part design body first, then the clone of the body in the job before making a change)

luzpaz, 2025-08-21:
ping @jkvfc

closed 2025-10-16 by sliptonic