FreeCAD/FreeCAD #15224: Graphviz is expected to be at /usr/bin by default instead of anywhere in PATH

adrianinsaval, 2024-07-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

This is not good for the appimage (and potentially other packaging formats) as it can't find the bundled binary by default and the user can't properly configure the path for the bundled binary since it changes on every launch. 
https://github.com/FreeCAD/FreeCAD/blob/561e521817034ea7660de987d8ab7183ad583595/src/Gui/GraphvizView.cpp#L286-L290

Instead this should look for the binaries in PATH by default

### Full version info

```shell
OS: Arch Linux (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38001 (Git) AppImage
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: English/United Kingdom (en_GB) ]
Installed mods:
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2024-07-05:
not a release blocker but would be nice to have this fixed before 1.0

leoheck, 2024-07-06:
This is a regression, I never seen such an issue.
[image]

adrianinsaval, 2024-07-06:
do you have graphviz installed on your system?

leoheck, 2024-07-06:
we don't know the message is not clear.

leoheck, 2024-07-06:
Yes, I do have it.
[image]

luzpaz, 2024-07-14:
Can you share the output of `whereis graphviz` pls ?

leoheck, 2024-07-15:
I Will check this later when on my computer... but why does this matter. You should use tools that are on PATH. Doesn't matter where they are.

luzpaz, 2024-07-15:
Just trying to cover all bases

onekk, 2024-07-21:
> I Will check this later when on my computer... but why does this matter. You should use tools that are on PATH. Doesn't matter where they are.

Yes and no, as on some cases you could have different versions, it however a development choice, probably adding the ability to set "binary search path" or "non essential" tools path in preferences should be a good move.

If auto detection fails (or as example detect two version) a simple message could tell to set a "preference" to specify a desired location.

If autodetect is succeding this variables should be correctly set at first startup.

Similar solution is adopted by some softwares around at least in Linux world.

Regards

Carlo D.

luzpaz, 2024-07-22:
bump @leoheck

leoheck, 2024-07-22:
?

It should use the tools from the PATH. If a version is required, then Freecad should check the tool version and indicate that the current version is not good... and the user can still make a wrap script to tweak the version to use the tool he has.

luzpaz, 2024-07-22:
> I Will check this later when on my computer... 

Please post output for `whereis graphviz` nonetheless. Thank you.

leoheck, 2024-07-23:
Sorry dude, it doesn't matter. It is on $PATH.

leoheck, 2024-09-11:
This is still an issue, Freecad is failing and I cannot use this tool to check what is going on. Nothing is working anymore. Yesterday it was all good.

luzpaz, 2024-09-11:
@wwmayer do you have a suggestion perhaps ? :pray:

closed 2024-10-21 by yorikvanhavre

adrianinsaval, 2024-10-21:
@leoheck can you test if this is fixed for you when the next weekly is available?

commit d1572286 (2024-10-21): Gui: try calling graphviz binaries from PATH by default [fix #15224]
commit 7094af67 (2024-10-21): Gui: try calling graphviz binaries from PATH by default [fix #15224]