FreeCAD/FreeCAD #15225: Freecad Crashes when Moving objects on the new Assembly

leoheck, 2024-07-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

How to reproduce:
1. Launch Freecad
2. Launch the Example > `AssemblyEXample.FCStd`
3. Activate the Assembly
4. Click and drag the Bucket001 (if a gentle drag does not make it crash, drag it like crazy)

```
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape001
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape012
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape013
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape014
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape015
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape016
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape017
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape018
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape019
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape020
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape021
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape022
<App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x7bcb2ea45320]
#1  0x7bca3557f027 in MbD::VectorNewtonRaphson::calcdxNorm() from /tmp/.mount_FreeCA6mKubw/usr/lib/./libOndselSolver.so.1+0x17
#2  0x7bca3556261a in MbD::SystemNewtonRaphson::calcdxNorm() from /tmp/.mount_FreeCA6mKubw/usr/lib/./libOndselSolver.so.1+0x7a
#3  0x7bca354e399d in MbD::NewtonRaphson::calcDXNormImproveRootCalcYNorm() from /tmp/.mount_FreeCA6mKubw/usr/lib/./libOndselSolver.so.1+0xd
#4  0x7bca354e3b2f in MbD::NewtonRaphson::iterate() from /tmp/.mount_FreeCA6mKubw/usr/lib/./libOndselSolver.so.1+0xaf
#5  0x7bca3557eedf in MbD::VectorNewtonRaphson::run() from /tmp/.mount_FreeCA6mKubw/usr/lib/./libOndselSolver.so.1+0x2f
#6  0x7bca35565959 in MbD::SystemSolver::runPosICDrag(std::shared_ptr<std::vector<std::shared_ptr<MbD::Part>, std::allocator<std::shared_ptr<MbD::Part> > > >) from /tmp/.mount_FreeCA6mKubw/usr/lib/./libOndselSolver.so.1+0x129
#7  0x7bca35565b24 in MbD::SystemSolver::runDragStep(std::shared_ptr<std::vector<std::shared_ptr<MbD::Part>, std::allocator<std::shared_ptr<MbD::Part> > > >) from /tmp/.mount_FreeCA6mKubw/usr/lib/./libOndselSolver.so.1+0x44
#8  0x7bca3556011d in MbD::System::runDragStep(std::shared_ptr<std::vector<std::shared_ptr<MbD::Part>, std::allocator<std::shared_ptr<MbD::Part> > > >) from /tmp/.mount_FreeCA6mKubw/usr/lib/./libOndselSolver.so.1+0x9d
#9  0x7bca3536c5ee in MbD::ASMTAssembly::runDragStep(std::shared_ptr<std::vector<std::shared_ptr<MbD::ASMTPart>, std::allocator<std::shared_ptr<MbD::ASMTPart> > > >) from /tmp/.mount_FreeCA6mKubw/usr/lib/./libOndselSolver.so.1+0x3ee
#10  0x7bcae705ed03 in Assembly::AssemblyObject::doDragStep() from /tmp/.mount_FreeCA6mKubw/usr/lib/AssemblyApp.so+0x3e3
#11  0x7bcadd27701d in AssemblyGui::ViewProviderAssembly::mouseMove(SbVec2s const&, Gui::View3DInventorViewer*) from /tmp/.mount_FreeCA6mKubw/usr/lib/AssemblyGui.so+0x90d
#12  0x7bcb32162460 in Gui::ViewProvider::eventCallback(void*, SoEventCallback*) from /tmp/.mount_FreeCA6mKubw/usr/bin/../lib/libFreeCADGui.so+0x400
#13  0x7bcb2e4c703e in SoEventCallback::handleEvent(SoHandleEventAction*) from /tmp/.mount_FreeCA6mKubw/usr/bin/../lib/./libCoin.so.80+0xbe
#14  0x7bcb2e2c5e48 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCA6mKubw/usr/bin/../lib/./libCoin.so.80+0x68
#15  0x7bcb2e47fcba in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_FreeCA6mKubw/usr/bin/../lib/./libCoin.so.80+0x15a
#16  0x7bcb2e4cfcd8 in SoGroup::doAction(SoAction*) from /tmp/.mount_FreeCA6mKubw/usr/bin/../lib/./libCoin.so.80+0x58
#17  0x7bcb2e4eaf7a in SoSeparator::doAction(SoAction*) from /tmp/.mount_FreeCA6mKubw/usr/bin/../lib/./libCoin.so.80+0x2a
#18  0x7bcb2e2c5e48 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCA6mKubw/usr/bin/../lib/./libCoin.so.80+0x68
#19  0x7bcb2e47fcba in SoChildList::traverse(SoAction*, int, int) from /tmp/.mount_FreeCA6mKubw/usr/bin/../lib/./libCoin.so.80+0x15a
#20  0x7bcb2e4cfcd8 in SoGroup::doAction(SoAction*) from /tmp/.mount_FreeCA6mKubw/usr/bin/../lib/./libCoin.so.80+0x58
#21  0x7bcb2e4eaf7a in SoSeparator::doAction(SoAction*) from /tmp/.mount_FreeCA6mKubw/usr/bin/../lib/./libCoin.so.80+0x2a
#22  0x7bcb2e2c5e48 in SoAction::traverse(SoNode*) from /tmp/.mount_FreeCA6mKubw/usr/bin/../lib/./libCoin.so.80+0x68
#23  0x7bcb2e2cfa3c in SoHandleEventAction::beginTraversal(SoNode*) from /tmp/.mount_FreeCA6mKubw/usr/bin/../lib/./libCoin.so.80+0xac
#24  0x7bcb2e2c6b2a in SoAction::apply(SoNode*) from /tmp/.mount_FreeCA6mKubw/usr/bin/../lib/./libCoin.so.80+0x2ea
#25  0x7bcb2e49bdc2 in SoEventManager::actuallyProcessEvent(SoEvent const*) from /tmp/.mount_FreeCA6mKubw/usr/bin/../lib/./libCoin.so.80+0x72
#26  0x7bcb320e74ee in SIM::Coin3D::Quarter::QuarterWidget::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCA6mKubw/usr/bin/../lib/libFreeCADGui.so+0x1e
#27  0x7bcb320f28c8 in SIM::Coin3D::Quarter::SoQTQuarterAdaptor::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCA6mKubw/usr/bin/../lib/libFreeCADGui.so+0x58
#28  0x7bcb320feb50 in Gui::NavigationStyle::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCA6mKubw/usr/bin/../lib/libFreeCADGui.so+0x40
#29  0x7bcb3210537f in Gui::TouchpadNavigationStyle::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCA6mKubw/usr/bin/../lib/libFreeCADGui.so+0x2df
#30  0x7bcb320fede5 in Gui::NavigationStyle::processEvent(SoEvent const*) from /tmp/.mount_FreeCA6mKubw/usr/bin/../lib/libFreeCADGui.so+0x65
#31  0x7bcb3212744b in Gui::View3DInventorViewer::processSoEvent(SoEvent const*) from /tmp/.mount_FreeCA6mKubw/usr/bin/../lib/libFreeCADGui.so+0x6b
#32  0x7bcb320ddd15 in SIM::Coin3D::Quarter::EventFilter::eventFilter(QObject*, QEvent*) from /tmp/.mount_FreeCA6mKubw/usr/bin/../lib/libFreeCADGui.so+0x1e5
#33  0x7bcb2f0b68bb in QCoreApplicationPrivate::sendThroughObjectEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCA6mKubw/usr/bin/../lib/libQt5Core.so.5+0x7b
#34  0x7bcb2ff7039e in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCA6mKubw/usr/bin/../lib/libQt5Widgets.so.5+0x6e
#35  0x7bcb2ff778f9 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA6mKubw/usr/bin/../lib/libQt5Widgets.so.5+0xa39
#36  0x7bcb31d3cc49 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA6mKubw/usr/bin/../lib/libFreeCADGui.so+0x99
#37  0x7bcb2f0b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCA6mKubw/usr/bin/../lib/libQt5Core.so.5+0x108
#38  0x7bcb2ff763eb in QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) from /tmp/.mount_FreeCA6mKubw/usr/bin/../lib/libQt5Widgets.so.5+0x1ab
#39  /tmp/.mount_FreeCA6mKubw/usr/bin/../lib/libQt5Widgets.so.5(+0x1c6e6f) [0x7bcb2ffc6e6f]
#40  /tmp/.mount_FreeCA6mKubw/usr/bin/../lib/libQt5Widgets.so.5(+0x1ca0c9) [0x7bcb2ffca0c9]
#41  0x7bcb2ff703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCA6mKubw/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#42  0x7bcb31d3cc49 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCA6mKubw/usr/bin/../lib/libFreeCADGui.so+0x99
[... 23 lines cut ...]
Build type: Release
Branch: main
Hash: 1fc6da959b936a2999774a4fe5b4c4b12bf6ac8d
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * Assembly4 0.50.12
  * InventorLoader 1.4.0
  * ose-piping
  * OpenTheme 2024.7.3
  * SheetMetal 0.4.20
  * fasteners 0.5.24
  * asm4_disable_version_check_popup
  * OpticsWorkbench 1.0.18
  * lattice2 1.0.0
  * kicadStepUpMod
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

leoheck, 2024-07-06:
It is not crashing anymore on  `0.22.0dev.38043`

But the Example is still getting crazy.
It starts this way. 
[image]

After some movements, it gets this way with the selected cylinders in the wrong place.
[image]

There is also this message
```
03:37:51  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape001
03:37:51  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape012
03:37:51  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape013
03:37:51  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape014
03:37:51  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape015
03:37:51  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape016
03:37:51  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape017
03:37:51  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape018
03:37:51  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape019
03:37:51  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape020
03:37:51  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape021
03:37:51  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape022
03:37:51  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape
03:38:32  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape001
03:38:32  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape012
03:38:32  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape013
03:38:32  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape014
03:38:32  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape015
03:38:32  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape016
03:38:32  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape017
03:38:32  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape018
03:38:32  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape019
03:38:32  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape020
03:38:32  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape021
03:38:32  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape022
03:38:32  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape
03:40:47  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape001
03:40:47  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape012
03:40:47  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape013
03:40:47  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape014
03:40:47  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape015
03:40:47  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape016
03:40:47  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape017
03:40:47  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape018
03:40:47  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape019
03:40:47  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape020
03:40:47  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape021
03:40:47  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape022
03:40:47  <App> PropertyTopoShape.cpp(433): Pending recompute for generating element map: AssemblyExample#Shape
```

maxwxyz, 2024-07-06:
This seems to be a duplicate of https://github.com/FreeCAD/FreeCAD/issues/15141

maxwxyz, 2024-07-06:
With the new file I get also issues:
[image]

maxwxyz, 2024-07-06:
@leoheck could you rephrase the original issue to include only the issue in the latest weekly?

PaddleStroke, 2024-07-08:
The solver should not accept flipping of joints during dragging. I think then it would improve the situation and prevent huge failure like this one.
There is already an issue about this : https://github.com/FreeCAD/FreeCAD/issues/14185

Perhaps this one can be closed and discussion moved there.

leoheck, 2024-07-08:
@PaddleStroke makes sense, thanks.

closed 2024-07-08 by leoheck