FreeCAD/FreeCAD #15227: Supports converting model edges to sketch lines

HaydenHu, 2024-07-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

[image]
Like this feature in the link branch version

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git)
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Chinese/China (zh_CN)
Installed mods: 
  * Nodes
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-06:
Duplicate

closed 2024-07-06 by maxwxyz

HaydenHu, 2024-07-06:
> * issues

Duplicate with that issues ?