FreeCAD/FreeCAD #15228: The BIM_column tool creates an object named "beam".

kaiwas, 2024-07-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When a column is created, a "beam" object is created.
I understand that the way beams and columns are constructed is very similar, but the name should probably be generated differently (as column rather than beam)

https://github.com/FreeCAD/FreeCAD/assets/32273833/4bd2a182-1077-455e-b5c9-d99d46023b61

### Full version info

```shell
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git) AppImage
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Russian/Russia (ru_RU)
Installed mods: 
  * OpenTheme 2024.7.3
  * PitchedRoof
```

### Subproject(s) affected?

BIM/Arch

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

kaiwas, 2024-07-06:
> is there a difference between the underlying object?

in version 0.21 everything was correct.
[image]

yorikvanhavre, 2024-07-08:
I cannot reproduce the problem, for me the created object has the correct Column IFC type...
Can you look at what happens on the python console, and try pasting mine here, to see if the result is different?

```python
import Arch
import WorkingPlane
s = Arch.makeStructure(length=300.0,width=400.0,height=3000.0)
s.Placement.Base = FreeCAD.Vector(-10.0, 20.0, 0.0)
wp = WorkingPlane.get_working_plane()
s.Placement.Rotation = s.Placement.Rotation.multiply(wp.get_placement().Rotation)
```

Roy-043, 2024-07-08:
I can confirm this. If you pause the video at 0:04 you will notice the settings that cause the error.

Steps:
1. Start BIM_Beam.
2. Specify Category "Precast concrete" and Preset "Beam"
3. Create a beam by picking 2 points. Notice BTW that with these settings the given preview does not match the produced beam (but that is a different issue).
4. Start BIM_Column.
5. Pick a point for the column.
6. Result: wrong object name.

kaiwas, 2024-07-08:
I noticed this problem in version 0.22, but it seems to be reproducible in version 0.21.
And it really comes down to the material chosen.
If you use a metal profile, then everything is fine.
But if you choose concrete, then something is not working correctly.
There is no “column” preset for a concrete column. The problem is probably related to this.
Demonstration video. (Thanks @Roy-043 )

https://github.com/FreeCAD/FreeCAD/assets/32273833/7e41eeae-7c17-49b2-abf8-5b338b0879ad

kaiwas, 2024-07-09:
I would also like to add that due to the fact that the beam/column tool is connected to each other, they sometimes work in an incomprehensible way. Topic for discussion on the forum.

https://forum.freecad.org/viewtopic.php?t=88989

yorikvanhavre, 2024-07-11:
Normally both column and beam modes should save their own settings separately... There seems to be a problem there indeed, I'll have a better look

FEA-eng, 2025-10-02:
Still relevant

```
OS: Windows 10 build 19045
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow)
Build date: 2025/09/30 23:55:55
Build type: Release
Branch: main
Hash: b08a0f87d68dd38cfcf39046209a7075adf11576
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: Polish/Poland (pl_PL)
Navigation Style/Orbit Style/Rotation Mode: CAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Light/
Logical DPI/Physical DPI/Pixel Ratio: 96/90.0633/1
```