FreeCAD/FreeCAD #15239: 0.22dev crashed on Part Design/sketcher close

piranha32, 2024-07-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

FreeCAD 0.22.0dev.38043 crashed when I was closing Sketcher with a fully constrained sketch. Unfortunately I can't pinpoint which operation made bodybuilding the sketch lead to the crash. The sketch built during the crashed session is lost, so I can't test if it will reproduce the crash.
Here's the stack trace:

```
Traceback (most recent call last):
  File "<string>", line 1, in <module>
<class 'ValueError'>: Not able to move point with the id and type: (5, 0)
App.getDocument('Unnamed').getObject('Sketch001').movePoint(5,0,App.Vector(101.227798,29.646217,0),0)
Drag curve: Not able to move point with the id and type: (5, 0)
Program received signal SIGSEGV, Segmentation fault.
#0  /lib64/libc.so.6(+0x3e9a0) [0x7f6b1525c9a0]
#1  0x7f6b1747eead in Data::ElementMap::addName(Data::MappedName&, Data::IndexedName const&, QVector<App::StringIDRef> const&, bool, Data::IndexedName*) from /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libFreeCADApp.so+0xed
#2  0x7f6b1747fc97 in Data::ElementMap::setElementName(Data::IndexedName const&, Data::MappedName const&, long, QVector<App::StringIDRef> const*, bool) from /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libFreeCADApp.so+0x297
#3  0x7f6addcf91b6 in Sketcher::SketchObject::buildShape() from /tmp/.mount_FreeCACEo7KE/usr/lib/Sketcher.so+0x236
#4  0x7f6addcfa1ea in Sketcher::SketchObject::execute() from /tmp/.mount_FreeCACEo7KE/usr/lib/Sketcher.so+0xda
#5  0x7f6b1725d552 in App::DocumentObject::recompute() from /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libFreeCADApp.so+0xc2
#6  0x7f6aed2a23cc in Part::Feature::recompute() from /tmp/.mount_FreeCACEo7KE/usr/lib/./Part.so+0xc
#7  0x7f6b1723773c in App::Document::_recomputeFeature(App::DocumentObject*) from /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libFreeCADApp.so+0x32c
#8  0x7f6b17243ae5 in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libFreeCADApp.so+0x625
#9  0x7f6b172c21ae in App::DocumentPy::recompute(_object*) from /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libFreeCADApp.so+0x2ee
#10  0x7f6b172c22c4 in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libFreeCADApp.so+0x24
#11  /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libpython3.11.so.1.0(+0x206068) [0x7f6b17a06068]
#12  /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x7f6b179e5b5c]
#13  /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x7f6b179f1532]
#14  /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libpython3.11.so.1.0(+0x2abd1d) [0x7f6b17aabd1d]
#15  /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7f6b17aab794]
#16  /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libpython3.11.so.1.0(+0x2cabf3) [0x7f6b17acabf3]
#17  /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libpython3.11.so.1.0(+0x2c669a) [0x7f6b17ac669a]
#18  /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x7f6b17aba9c9]
#19  0x7f6b16d3d48c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libFreeCADBase.so+0x6c
#20  0x7f6b18387964 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libFreeCADGui.so+0x94
#21  0x7f6b18387af3 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libFreeCADGui.so+0xe3
#22  0x7f6b18387c14 in Gui::Command::updateActive() from /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libFreeCADGui.so+0x44
#23  0x7f6a04555197 in SketcherGui::ViewProviderSketch::unsetEdit(int) from /tmp/.mount_FreeCACEo7KE/usr/lib/SketcherGui.so+0x317
#24  0x7f6b18759c03 in Gui::ViewProvider::finishEditing() from /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libFreeCADGui.so+0x13
#25  0x7f6b182fccad in Gui::Document::_resetEdit() from /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libFreeCADGui.so+0x9d
#26  0x7f6b1827c5aa in Gui::Application::setEditDocument(Gui::Document*) from /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libFreeCADGui.so+0x3a
#27  0x7f6b18311750 in Gui::DocumentPy::resetEdit(_object*) from /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libFreeCADGui.so+0x30
#28  0x7f6b18311784 in Gui::DocumentPy::staticCallback_resetEdit(_object*, _object*) from /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libFreeCADGui.so+0x24
#29  /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libpython3.11.so.1.0(+0x206068) [0x7f6b17a06068]
#30  /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x7f6b179e5b5c]
#31  /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x7f6b179f1532]
#32  /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libpython3.11.so.1.0(+0x2abd1d) [0x7f6b17aabd1d]
#33  /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7f6b17aab794]
#34  /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libpython3.11.so.1.0(+0x2cabf3) [0x7f6b17acabf3]
#35  /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libpython3.11.so.1.0(+0x2c669a) [0x7f6b17ac669a]
#36  /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x7f6b17aba9c9]
#37  0x7f6b16d3d48c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libFreeCADBase.so+0x6c
#38  0x7f6b18387964 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libFreeCADGui.so+0x94
#39  0x7f6b18387af3 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libFreeCADGui.so+0xe3
#40  0x7f6a04591615 in SketcherGui::TaskDlgEditSketch::reject() from /tmp/.mount_FreeCACEo7KE/usr/lib/SketcherGui.so+0x145
#41  0x7f6b186c5dd7 in Gui::TaskView::TaskView::reject() from /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libFreeCADGui.so+0x67
#42  /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x7f6b156eb6f2]
#43  /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libQt5Widgets.so.5(+0x2fab70) [0x7f6b166fab70]
#44  /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libQt5Core.so.5(+0x2eb964) [0x7f6b156eb964]
#45  0x7f6b1665b522 in QAbstractButton::clicked(bool) from /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libQt5Widgets.so.5+0x32
#46  /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libQt5Widgets.so.5(+0x25b73a) [0x7f6b1665b73a]
#47  /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libQt5Widgets.so.5(+0x25d1c8) [0x7f6b1665d1c8]
#48  0x7f6b1665d3d7 in QAbstractButton::mouseReleaseEvent(QMouseEvent*) from /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libQt5Widgets.so.5+0xe7
#49  0x7f6b165aec08 in QWidget::event(QEvent*) from /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libQt5Widgets.so.5+0x178
#50  0x7f6b165703ae in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libQt5Widgets.so.5+0x7e
#51  0x7f6b16578262 in QApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libQt5Widgets.so.5+0x13a2
#52  0x7f6b1833cc69 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libFreeCADGui.so+0x99
#53  0x7f6b156b6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCACEo7KE/usr/bin/../lib/libQt5Core.so.5+0x108
[... 19 lines cut ...]
#73  /tmp/.mount_FreeCACEo7KE/usr/bin/freecad(+0x7981) [0x5596fd4f0981]
```

### Full version info

```shell
OS: Fedora Linux 39 (Thirty Nine) (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git) AppImage
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-06:
Could you provide an example file or steps how to reproduce?

piranha32, 2024-07-06:
it will be difficult, because I lost all the work which lead to the crash

xtemp09, 2024-07-06:
This issue seems to be a duplicate of #15102.

maxwxyz, 2024-07-07:
Duplicate of #15102

closed 2024-07-07 by maxwxyz