FreeCAD/FreeCAD #15240: In certain situations, applying "Horizontal" or "Vertical" constrain to centers of construction circles in polygons leads to degenerated shape

piranha32, 2024-07-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Applying "Horizontal" or "Vertical" constrain to centers of circles defining polygons in Sketcher, when the difference in positions is greater than the radius of one of the circles (the center of one of the circles is outside of the horizontal/vertical lines bounding the other circle), leads to degeneration of one of the polygons. 
Attached is a test file and screenshots showing "Before" and "After" applying "Horizontal" constrain to centers of the polygons.

Steps to reproduce:
1. Open 'polygon fail test.FCStd'
2. Open the sketch for editing
3. Select centers of the polygons
4. Apply Horizontal constrain

[image]
[image]

### Full version info

```shell
OS: Fedora Linux 39 (Thirty Nine) (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git) AppImage
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
```

### Subproject(s) affected?

None

### Anything else?

[polygon fail test.zip](https://github.com/user-attachments/files/16116834/polygon.fail.test.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-06:
Is this also an issue in the current stable 0.21.2 version?

piranha32, 2024-07-06:
The same behavior 

```
OS: Fedora Linux 39 (Thirty Nine) (KDE/plasmax11)
Word size of FreeCAD: 64-bit
Version: 0.21.2.33771 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.21.2)
Hash: b9bfa5c5507506e4515816414cd27f4851d00489
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
```

Syres916, 2024-07-07:
Surely this is a duplicate of https://github.com/FreeCAD/FreeCAD/issues/8711

maxwxyz, 2024-07-08:
Duplicate of https://github.com/FreeCAD/FreeCAD/issues/8711