FreeCAD/FreeCAD #15243: Segfault when closing sketch

nazar-pc, 2024-07-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Take the following file:
[Core.zip](https://github.com/user-attachments/files/16116919/Core.zip)

Double click on MotherboardBase to open sketch editing, following is printed in logs:
```
Traceback (most recent call last):
  File "<string>", line 7, in <module>
  File "/tmp/.mount_FreeCAqRT6gi/usr/Mod/Show/mTempoVis.py", line 366, in get_all_dependent
    cnt_chain = Containers.ContainerChain(doc_obj)
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/tmp/.mount_FreeCAqRT6gi/usr/Mod/Show/Containers.py", line 246, in ContainerChain
    raise ContainerTreeError("Container tree is not a tree")
<class 'Show.Containers.ContainerTreeError'>: Container tree is not a tree
setEdit: visibility automation failed with an error: 
```

Exiting sketch now crashed the whole app:

<details>

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x709476e45320]
#1  0x70947927eead in Data::ElementMap::addName(Data::MappedName&, Data::IndexedName const&, QVector<App::StringIDRef> const&, bool, Data::IndexedName*) from /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libFreeCADApp.so+0xed
#2  0x70947927fc97 in Data::ElementMap::setElementName(Data::IndexedName const&, Data::MappedName const&, long, QVector<App::StringIDRef> const*, bool) from /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libFreeCADApp.so+0x297
#3  0x7094344f91b6 in Sketcher::SketchObject::buildShape() from /tmp/.mount_FreeCAg79Upa/usr/lib/Sketcher.so+0x236
#4  0x7094344fa1ea in Sketcher::SketchObject::execute() from /tmp/.mount_FreeCAg79Upa/usr/lib/Sketcher.so+0xda
#5  0x70947905d552 in App::DocumentObject::recompute() from /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libFreeCADApp.so+0xc2
#6  0x709441aa23cc in Part::Feature::recompute() from /tmp/.mount_FreeCAg79Upa/usr/lib/./Part.so+0xc
#7  0x70947903773c in App::Document::_recomputeFeature(App::DocumentObject*) from /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libFreeCADApp.so+0x32c
#8  0x709479043ae5 in App::Document::recompute(std::vector<App::DocumentObject*, std::allocator<App::DocumentObject*> > const&, bool, bool*, int) from /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libFreeCADApp.so+0x625
#9  0x7094790c21ae in App::DocumentPy::recompute(_object*) from /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libFreeCADApp.so+0x2ee
#10  0x7094790c22c4 in App::DocumentPy::staticCallback_recompute(_object*, _object*) from /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libFreeCADApp.so+0x24
#11  /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libpython3.11.so.1.0(+0x206068) [0x709479806068]
#12  /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x7094797e5b5c]
#13  /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x7094797f1532]
#14  /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libpython3.11.so.1.0(+0x2abd1d) [0x7094798abd1d]
#15  /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7094798ab794]
#16  /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libpython3.11.so.1.0(+0x2cabf3) [0x7094798cabf3]
#17  /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libpython3.11.so.1.0(+0x2c669a) [0x7094798c669a]
#18  /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x7094798ba9c9]
#19  0x709478b3d48c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libFreeCADBase.so+0x6c
#20  0x70947a187964 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libFreeCADGui.so+0x94
#21  0x70947a187af3 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libFreeCADGui.so+0xe3
#22  0x70947a187c14 in Gui::Command::updateActive() from /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libFreeCADGui.so+0x44
#23  0x7091ee955197 in SketcherGui::ViewProviderSketch::unsetEdit(int) from /tmp/.mount_FreeCAg79Upa/usr/lib/SketcherGui.so+0x317
#24  0x70947a559c03 in Gui::ViewProvider::finishEditing() from /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libFreeCADGui.so+0x13
#25  0x70947a0fccad in Gui::Document::_resetEdit() from /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libFreeCADGui.so+0x9d
#26  0x70947a07c5aa in Gui::Application::setEditDocument(Gui::Document*) from /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libFreeCADGui.so+0x3a
#27  0x70947a111750 in Gui::DocumentPy::resetEdit(_object*) from /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libFreeCADGui.so+0x30
#28  0x70947a111784 in Gui::DocumentPy::staticCallback_resetEdit(_object*, _object*) from /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libFreeCADGui.so+0x24
#29  /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libpython3.11.so.1.0(+0x206068) [0x709479806068]
#30  /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libpython3.11.so.1.0(_PyObject_MakeTpCall+0x26c) [0x7094797e5b5c]
#31  /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libpython3.11.so.1.0(_PyEval_EvalFrameDefault+0x6f2) [0x7094797f1532]
#32  /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libpython3.11.so.1.0(+0x2abd1d) [0x7094798abd1d]
#33  /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libpython3.11.so.1.0(PyEval_EvalCode+0xa4) [0x7094798ab794]
#34  /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libpython3.11.so.1.0(+0x2cabf3) [0x7094798cabf3]
#35  /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libpython3.11.so.1.0(+0x2c669a) [0x7094798c669a]
#36  /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libpython3.11.so.1.0(PyRun_StringFlags+0x69) [0x7094798ba9c9]
#37  0x709478b3d48c in Base::InterpreterSingleton::runString[abi:cxx11](char const*) from /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libFreeCADBase.so+0x6c
#38  0x70947a187964 in Gui::Command::_runCommand(char const*, int, Gui::Command::DoCmd_Type, char const*) from /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libFreeCADGui.so+0x94
#39  0x70947a187af3 in Gui::Command::_doCommand(char const*, int, Gui::Command::DoCmd_Type, char const*, ...) from /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libFreeCADGui.so+0xe3
#40  0x7091ee991615 in SketcherGui::TaskDlgEditSketch::reject() from /tmp/.mount_FreeCAg79Upa/usr/lib/SketcherGui.so+0x145
#41  0x70947a4c5dd7 in Gui::TaskView::TaskView::reject() from /tmp/.mount_FreeCAg79Upa/usr/bin/../lib/libFreeCADGui.so+0x67
[... 75 lines cut ...]
```

The first error is the same, but crash is replaced with this:

<details>

```
Sketcher::SketchObject: Link(s) to object(s) 'Sketch' go out of the allowed scope 'Sketch007'. Instead, the linked object(s) reside within 'Body'.
<PropertyLinks> PropertyLinks.cpp(433): Core#Sketch008.AttachmentSupport auto change element reference Core#Sketch003 ;g27228v1;SKT;:H,V.Vertex9 -> ;g57v1;SKT;:H,V.Vertex9
<PropertyLinks> PropertyLinks.cpp(433): Core#Sketch004.AttachmentSupport auto change element reference Core#Sketch003 ;g27226v1;SKT;:H,V.Vertex14 -> ;g55v1;SKT;:H,V.Vertex14
<PropertyLinks> PropertyLinks.cpp(433): Core#Sketch004.ExternalGeometry auto change element reference Core#Sketch003 ;g27224v1;SKT;:H,V.Vertex12 -> ;g53v1;SKT;:H,V.Vertex12
<PropertyLinks> PropertyLinks.cpp(433): Core#Sketch004.ExternalGeometry auto change element reference Core#Sketch003 ;g27217;SKT;:H,E.Edge1 -> ;g46;SKT;:H,E.Edge1
Sketcher::SketchObject: Link(s) to object(s) 'Sketch' go out of the allowed scope 'Sketch005'. Instead, the linked object(s) reside within 'Body'.
App::DocumentObjectGroup: Link(s) to object(s) 'Sketch' go out of the allowed scope 'Group002'. Instead, the linked object(s) reside within 'Body'.
<PropertyLinks> PropertyLinks.cpp(433): Core#Sketch008.ExternalGeometry auto change element reference Core#Extrude ;g27228v1;SKT;:H,V;:Hb5e,V;:G;XTR;:Hb5e:7,E.Edge23 -> ;g57v1;SKT;:H,V;:Hb5e,V;:G;XTR;:Hb5e:7,E.Edge23
<PropertyLinks> PropertyLinks.cpp(433): Core#Sketch008.ExternalGeometry auto change element reference Core#Extrude ;g27239v2;SKT;:H,V;:Hb5e,V;:G;XTR;:Hb5e:7,E.Edge14 -> ;g68v2;SKT;:H,V;:Hb5e,V;:G;XTR;:Hb5e:7,E.Edge14
```

</details>

So this is a regression in some sense.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

depthoffocus, 2024-07-06:
I have been experiencing similar issues I can't reproduce.

But  can reproduce this on:

OS: macOS 14.5
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git)
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * ThreadProfile 1.90.0
  * freecad.gears 1.2.0
  * OpenTheme 2024.6.22
  * fasteners 0.5.24
  * Curves 0.6.40

Text file attached has what I believe is the relevant bit from the Mac crash report for thread 0.

[issues.15243.macos.crashlog.txt](https://github.com/user-attachments/files/16116969/issues.15243.macos.crashlog.txt)

xtemp09, 2024-07-06:
This issue seems to be a duplicate of #15102.

maxwxyz, 2024-07-07:
Duplicate of #15102

closed 2024-07-07 by maxwxyz