FreeCAD/FreeCAD #15244: Visibility icon default state (active) is not shown in Preferences (Appimage)

MiqCG, 2024-07-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

With a fresh Appimage (deleted .config .local/share and .cache folders), the Visibility icon is shown by default in the Tree as expected for merged PR #14393 but if you look into the Edit > Preferences > Display > UI , the "Show visibility icon" check box is inactive. See video attached.

The expected condition is that it is active by default.

I have previously searched in Github with [is:issue is:open visibility icon]

### Full version info

```shell
OS: Devuan GNU/Linux 5 (daedalus) (XFCE/xfce)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git) AppImage
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Catalan/Spain (ca_ES)
```

### Subproject(s) affected?

None

### Anything else?

https://github.com/FreeCAD/FreeCAD/assets/18265355/8e1a473c-c375-49a1-b9be-23c40c37ad74

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-07:
@NomAnor FYI

NomAnor, 2024-07-08:
It looks like this concerns all options on the Display -> UI page (e.g. HideScrollBar, HideHeaderView). 

@kadet1090 changed the preferences in https://github.com/FreeCAD/FreeCAD/commit/05aaf0ed48bf81007bf85684eb6a07425bc56323 to a proper ui file but it seems he missed the https://github.com/FreeCAD/FreeCAD/blob/99d63a9693316bad118740dc8ee86098cca5e16b/src/Gui/TreeParams.py file that previously (and currently?) defines the list of options and their defaults.

I don't think other preferences use a similar file, so this file should propably be removed and its usage in the code replaced with whatever the other preferences use.

kadet1090, 2024-07-08:
I'll sync up this manually, getting rid of that file is not easy :/

luzpaz, 2024-07-24:
Any progress here ?

maxwxyz, 2024-07-24:
It still resets all wrong on that page. Reset shows all columns in the tree view, even the internal names, hides the visibility icon,...

Syres916, 2024-07-25:
@maxwxyz please double-check my investigation, is this what you believe a new user should see in Preferences>Display>UI  :

[image]

maxwxyz, 2024-07-25:
@Syres916 Hide description also checked by default.

Syres916, 2024-07-25:
@maxwxyz thanks, sorted and PR #15622 created.

PR #15622 (2024-07-29): [Gui] Syncronise preferences page Display > UI with default settings
Fixes #15244

Default new user's settings:

[image]

FYI @kadet1090
closed 2024-07-29 by chennes