FreeCAD/FreeCAD #15246: Disable unit replacement

nazar-pc, 2024-07-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

The closest issue to what I have is probably https://github.com/FreeCAD/FreeCAD/issues/13040

In a project that uses metric units I have a few components that use imperial units. I take those units from specs, type in and they are immediately converted to millimeters. Very annoying because I will not be able to find what does a particular measurement correspond to in the spec.

The only workaround I found so far was to use `0.304 in * 1` as a formula. When `* 1` is used in a formula it finally gives up and preserves the units as I enter them, but this is a huge usability issue because:
* it is way harder to type
* sketcher still shows converted units in mm unless I open formula editor

So what I would really like to see is if I type `0.304 in` to keep it `0.304 in` or `0.304 ''` (whatever is the canonical way) without conversion to mm even though internally it may end up using mm indeed.

Going a bit further, it might also make sense to not replace user input at all (just apply formatting), like https://github.com/FreeCAD/FreeCAD/issues/13040 requests. Simply show a compute result (like already right now) that user can click to apply it (replacing original input) if they want compute result instead of original input.

All this helps self-documentation greatly, for example in code when I need hours or other units in seconds I typically write `24 * 3600` or `24 * 60 * 60` rather than `86400` since it is easier to understand, especially when the value is less common.

### Full version info

```shell
Currently dev branch, but I don't think this is version-specific
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

nazar-pc, 2024-07-08:
Updated description with a quick screencast with additional explanation

mahatma-panda, 2024-12-04:
I am also experiencing this issue. The unit replacement is causing confusion in the sketcher constraints:
[image]

I realized that multiplying by 1000 (unitless) was the issue, and that adding the units ("thou") caused the expression to evaluate as expected. 
[image]