FreeCAD/FreeCAD #15247: Configuration table doesn't respect/save cell range

nazar-pc, 2024-07-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Specifying cell range for configuration table doesn't seem to be respected or saved properly, here is a demo:

https://github.com/FreeCAD/FreeCAD/assets/928965/ead10aa0-3563-4c20-a77c-fb674be9c413

### Full version info

```shell
OS: Ubuntu 24.04 LTS (Custom/custom)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37928 (Git) AppImage
Build type: Release
Branch: main
Hash: e7699b531eba4bee262f7214699f07b96f66ba7b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 
  * OpenTheme 2024.7.3
  * fasteners 0.5.24
  * parts_library
  * Curves 0.6.40
```

### Subproject(s) affected?

Spreadsheet

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

tarman3, 2024-12-07:
Not reproduced

<details>
  <summary>FreeCAD 1.0.0.39109 (Git) makepkg, Arch Linux (KDE/plasma/wayland)</summary>

```
OS: Arch Linux (KDE/plasma/wayland)
Architecture: x86_64
Version: 1.0.0.39109 (Git)
Build type: Release
Branch: makepkg
Hash: 2fcc5317fe3aee96ca73475986a577719fc78e20
Python 3.12.7, Qt 6.8.0, Coin 4.0.3, Vtk 9.3.1, OCC 7.8.1
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Installed mods: lattice2 1.0.0; CurvedShapes 1.0.13; MeshRemodel 1.10.35; freecad.gears 1.3.0; fasteners 0.5.33; sheetmetal 0.6.0; Silk 0.1.5; Curves 0.6.51
```
</details>

nazar-pc, 2024-12-08:
Certainly still reproducible exactly like in the original post, here is from latest weekly build:
```
OS: Ubuntu 24.04.1 LTS (Custom/custom/xcb)
Architecture: x86_64
Version: 1.1.0dev.39503 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: da45328f6f9b35cbbdc7167d7faadb6a07634bdb
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C) [ OS: Ukrainian/Ukraine (uk_UA) ]
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Qt default
Installed mods: 
  * OpenTheme 2024.7.3
  * fasteners 0.5.25
  * parts_library
  * Curves 0.6.42
```

https://github.com/user-attachments/assets/0528a321-9dbe-4a70-823d-7f139dba8715

FEA-eng, 2026-10-21:
Should still be relevant