FreeCAD/FreeCAD #15249: Configuration table is duplicated on property name change

nazar-pc, 2024-07-06:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

When configuration table property name changes, it creates a new property without removing the old one for some reason:

https://github.com/FreeCAD/FreeCAD/assets/928965/8fb08e4b-8014-4082-98f0-4b7fb329075b

### Full version info

```shell
OS: Ubuntu 24.04 LTS (Custom/custom)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37928 (Git) AppImage
Build type: Release
Branch: main
Hash: e7699b531eba4bee262f7214699f07b96f66ba7b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 
  * OpenTheme 2024.7.3
  * fasteners 0.5.24
  * parts_library
  * Curves 0.6.40
```

### Subproject(s) affected?

Spreadsheet

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-07:
I guess maybe not to break the use of features which reference the old name? 
Changing alias in e.g. Excel also does not remove the old one.

kaiwas, 2024-07-07:
At the moment there is no way to adjust existing properties. To avoid creating a duplicate, you will have to delete the original configuration table and then create it again.

nazar-pc, 2024-07-07:
I can edit aliases of indivudual cells and changes propagate nicely to other spreadsheets within the same document. This is why renaming configuration table with silent duplication was unexpected to me.