FreeCAD/FreeCAD #15250: Segfault on Redo

nazar-pc, 2024-07-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

Wasn't able to find clear reproduction, but managed to trigger sefault multiple times in different cases by doing Redo:

<details>

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x45320) [0x71c6da445320]
#1  0x71c6d9cc5cd7 in SoAction::hasTerminated() const from /tmp/.mount_FreeCASwikg2/usr/bin/../lib/./libCoin.so.80+0x7
#2  0x71c6d9ed9402 in SoNode::callbackS(SoAction*, SoNode*) from /tmp/.mount_FreeCASwikg2/usr/bin/../lib/./libCoin.so.80+0x12
#3  0x71c6dd6e89e2 in Gui::Document::handleChildren3D(Gui::ViewProvider*, bool) from /tmp/.mount_FreeCASwikg2/usr/bin/../lib/libFreeCADGui.so+0x42
#4  0x71c6dd6fc027 in Gui::Document::redo(int) from /tmp/.mount_FreeCASwikg2/usr/bin/../lib/libFreeCADGui.so+0x97
#5  0x71c6ddb123f5 in Gui::View3DInventor::onMsg(char const*, char const**) from /tmp/.mount_FreeCASwikg2/usr/bin/../lib/libFreeCADGui.so+0x535
#6  0x71c6dd679cbb in Gui::Application::sendMsgToActiveView(char const*, char const**) from /tmp/.mount_FreeCASwikg2/usr/bin/../lib/libFreeCADGui.so+0x3b
#7  0x71c6dd78f098 in Gui::Command::_invoke(int, bool) from /tmp/.mount_FreeCASwikg2/usr/bin/../lib/libFreeCADGui.so+0x2a8
#8  0x71c6dd78f454 in Gui::Command::invoke(int, Gui::Command::TriggerSource) from /tmp/.mount_FreeCASwikg2/usr/bin/../lib/libFreeCADGui.so+0x134
#9  /tmp/.mount_FreeCASwikg2/usr/bin/../lib/libQt5Core.so.5(+0x2eb6f2) [0x71c6daaeb6f2]
#10  0x71c6db96a462 in QAction::triggered(bool) from /tmp/.mount_FreeCASwikg2/usr/bin/../lib/libQt5Widgets.so.5+0x32
#11  0x71c6db96ce2b in QAction::activate(QAction::ActionEvent) from /tmp/.mount_FreeCASwikg2/usr/bin/../lib/libQt5Widgets.so.5+0xab
#12  0x71c6dd7ecfc0 in Gui::ShortcutManager::onTimer() from /tmp/.mount_FreeCASwikg2/usr/bin/../lib/libFreeCADGui.so+0x100
#13  0x71c6dd7edfcd in Gui::ShortcutManager::checkShortcut(QObject*, QKeySequence const&) from /tmp/.mount_FreeCASwikg2/usr/bin/../lib/libFreeCADGui.so+0x52d
#14  0x71c6dd7f0237 in Gui::ShortcutManager::eventFilter(QObject*, QEvent*) from /tmp/.mount_FreeCASwikg2/usr/bin/../lib/libFreeCADGui.so+0x97
#15  0x71c6daab67ca in QCoreApplicationPrivate::sendThroughApplicationEventFilters(QObject*, QEvent*) from /tmp/.mount_FreeCASwikg2/usr/bin/../lib/libQt5Core.so.5+0x6a
#16  0x71c6db9703e8 in QApplicationPrivate::notify_helper(QObject*, QEvent*) from /tmp/.mount_FreeCASwikg2/usr/bin/../lib/libQt5Widgets.so.5+0xb8
#17  0x71c6dd73a479 in Gui::GUIApplication::notify(QObject*, QEvent*) from /tmp/.mount_FreeCASwikg2/usr/bin/../lib/libFreeCADGui.so+0x99
#18  0x71c6daab6b28 in QCoreApplication::notifyInternal2(QObject*, QEvent*) from /tmp/.mount_FreeCASwikg2/usr/bin/../lib/libQt5Core.so.5+0x108
#19  0x71c6db162d63 in QShortcutMap::dispatchEvent(QKeyEvent*) from /tmp/.mount_FreeCASwikg2/usr/bin/../lib/libQt5Gui.so.5+0x423
#20  0x71c6db16368b in QShortcutMap::tryShortcut(QKeyEvent*) from /tmp/.mount_FreeCASwikg2/usr/bin/../lib/libQt5Gui.so.5+0x5b
#21  0x71c6db10e60a in QWindowSystemInterface::handleShortcutEvent(QWindow*, unsigned long, int, QFlags<Qt::KeyboardModifier>, unsigned int, unsigned int, unsigned int, QString const&, bool, unsigned short) from /tmp/.mount_FreeCASwikg2/usr/bin/../lib/libQt5Gui.so.5+0x16a
#22  0x71c6db13129f in QGuiApplicationPrivate::processKeyEvent(QWindowSystemInterfacePrivate::KeyEvent*) from /tmp/.mount_FreeCASwikg2/usr/bin/../lib/libQt5Gui.so.5+0x7f
#23  0x71c6db10b0bc in QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCASwikg2/usr/bin/../lib/libQt5Gui.so.5+0xac
#24  /tmp/.mount_FreeCASwikg2/usr/plugins/platforms/../../lib/libQt5XcbQpa.so.5(+0x64bea) [0x71c6d5294bea]
#25  /tmp/.mount_FreeCASwikg2/usr/bin/../lib/./libglib-2.0.so.0(+0x5bad7) [0x71c6d9110ad7]
#26  /tmp/.mount_FreeCASwikg2/usr/bin/../lib/./libglib-2.0.so.0(+0x5eac7) [0x71c6d9113ac7]
#27  /tmp/.mount_FreeCASwikg2/usr/bin/../lib/./libglib-2.0.so.0(g_main_context_iteration+0x2e) [0x71c6d91140ce]
#28  0x71c6dab0aa36 in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCASwikg2/usr/bin/../lib/libQt5Core.so.5+0x66
#29  0x71c6daab570b in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) from /tmp/.mount_FreeCASwikg2/usr/bin/../lib/libQt5Core.so.5+0x11b
#30  0x71c6daabd212 in QCoreApplication::exec() from /tmp/.mount_FreeCASwikg2/usr/bin/../lib/libQt5Core.so.5+0x82
#31  0x71c6dd69cbb2 in Gui::Application::runApplication() from /tmp/.mount_FreeCASwikg2/usr/bin/../lib/libFreeCADGui.so+0x972
#32  /tmp/.mount_FreeCASwikg2/usr/bin/freecad(+0x7620) [0x56c586f76620]
#33  /lib/x86_64-linux-gnu/libc.so.6(+0x2a1ca) [0x71c6da42a1ca]
#34  /lib/x86_64-linux-gnu/libc.so.6(__libc_start_main+0x8b) [0x71c6da42a28b]
#35  /tmp/.mount_FreeCASwikg2/usr/bin/freecad(+0x7981) [0x56c586f76981]
Exit code 1
```

</details>

Hopefully this is useful.

### Full version info

```shell
OS: Ubuntu 24.04 LTS (Custom/custom)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37928 (Git) AppImage
Build type: Release
Branch: main
Hash: e7699b531eba4bee262f7214699f07b96f66ba7b
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: Ukrainian/Ukraine (uk_UA)
Installed mods: 
  * OpenTheme 2024.7.3
  * fasteners 0.5.24
  * parts_library
  * Curves 0.6.40
```

### Subproject(s) affected?

Core

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-07:
Any chance to provide example file or steps what you have done?

xtemp09, 2024-07-07:
This is useless without reproduction steps. What workbench is it?

nazar-pc, 2024-07-07:
This was mostly in Part workbench. What I was doing it clicking Undo a few times quickly, then Redo and it crashed. Document doesn't seem to matter, I tried to reproduce from memory a set of actions before crash on the same document after restart and wasn't able to crash it intentionally. But it did crash like 5 times for me yesterday overall.

luzpaz, 2024-07-17:
@nazar-pc thanks for reporting nonetheless.

nazar-pc, 2024-07-17:
If there is a useful debug level I can set I can share logs if it crashes again

xtemp09, 2024-07-17:
@nazar-pc, a demo file and the reproduction steps are better. To catch it, you can record your screen. "Five times a day" seems to be very frequent, meaning that, easy to catch.

nazar-pc, 2024-07-21:
Just crashed again. I have been adding some points in Sketch, then turning some lines into construction geometry. Undo/Redo and it crashed. Recovered document, repeated steps and it didn't crash this time. Tried in different order.

Is there some way to have a detailed log of what is happening in FreeCAD (maybe trace level) that would help FreeCAD developers to diagnose it?

luzpaz, 2024-12-13:
@nazar-pc any progress ?

nazar-pc, 2024-12-13:
I haven't been using FreeCAD much since last report unfortunately, so don't have any new information to add at this time

luzpaz, 2024-12-13:
Thanks for the response. Due to no clear way to reproduce, unfortunately we'll need to close this ticket. If you encounter this issue again please try again to recreate it, and report... so we can address it. TIA.

closed 2024-12-13 by luzpaz