FreeCAD/FreeCAD #15253: Crash when calling "Content"

wwmayer, 2024-07-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

1. Create a Part box
2. Call `App.ActiveDocument.Box.Content` in Python console
3. Crashes in FreeCAD/src/Base/Writer.cpp:247: std::string Base::Writer::addFile(const char *, const Base::Persistence *) because of a failing assert().

The crash is caused by https://github.com/FreeCAD/FreeCAD/pull/14898 because PropertyPartShape::Save() no longer ignores the case when `writer.isForceXML()` returns false.

### Full version info

```shell
OS: Ubuntu 22.04.4 LTS (XFCE/xubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git)
Build type: debug
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: German/Germany (de_DE)
```

### Subproject(s) affected?

Part

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2024-07-07:
@bgbsww FYI

bgbsww, 2024-07-07:
Yes, this breaks many things :(  The code at https://github.com/bgbsww/FreeCAD/blame/4d90e1655a9756c1075993f77b0d760da7404253/src/Mod/Part/App/PropertyTopoShape.cpp#L301 is wrong.   Just need to ascertain what's right.

PR #15258 (2024-07-09): Adjust PropertyTopoShape::Save code for correctness.
Fix #15253.
closed 2024-07-09 by chennes