FreeCAD/FreeCAD #15255: Broken external constraint

Murmele, 2024-07-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Problem description

# Steps to reproduce:
1) Open attached project
2) Open "StickHoleBottom" (From SubtractiveLoft001) in Sketchmode
3) Recompute sketch

# Expected
1) Seeing a list of external constraints even they are broken
2) Seeing an error message which points me to the correct sketch

# Observed

[image]

The problem is that Sketch018 is not available, because it was renamed to StickHoleBottom (I found with `App.getDocument('StativFlat').getObject('Sketch018').Label`)

It would be nice to show the label name in the error message instead of the internal name.

[StativFlat.zip](https://github.com/user-attachments/files/16118725/StativFlat.zip)

### Full version info

```shell
OS: Fedora Linux 40 (Workstation Edition) (GNOME/gnome)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git) AppImage
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
```

### Subproject(s) affected?

None

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Murmele, 2024-07-07:
Solution for me:
1) Close sketch mode
2) Delete all external Geometries and set again

maxwxyz, 2024-07-07:
@shaise is this the same as https://github.com/FreeCAD/FreeCAD/issues/15205 ?

shaise, 2024-07-07:
My issue is already solved in this version.
Regardless, I tried to reproduce the issue under OS: Windows 11 build 22631, Version: 0.22.0dev.38043 (Git)
but could not. Perhaps I did not understand the sequence correctly. I recompute the sketch by right clicking StickHoleBottom then recompute

Syres916, 2024-07-07:
@Murmele if you want to send me a pull request for any other places on top of the two mentioned in https://github.com/FreeCAD/FreeCAD/pull/15184

maxwxyz, 2024-07-07:
@bgbsww maybe you can weight in here

Murmele, 2024-07-07:
> My issue is already solved in this version. Regardless, I tried to reproduce the issue under OS: Windows 11 build 22631, Version: 0.22.0dev.38043 (Git) but could not. Perhaps I did not understand the sequence correctly. I recompute the sketch by right clicking StickHoleBottom then recompute

Can you move the circle in the sketch mode and then recomputing to reproduce?

Murmele, 2024-07-07:
> @Murmele if you want to send me a pull request for any other places on top of the two mentioned in #15184

I will check it out and create a pullreq

bgbsww, 2024-07-07:
So, the two proposed clarifications of showing errors with external constraint are good ideas;  showing both the internal name and the label ( which should be visible from python, but aren't to the user )  **somewhere** makes sense.

I'm not sure it makes sense in the c++ log messages, both because there are a lot of them and they generally don't have this information in them, but also because the more processing you add to a log message, the more likely it is to break.

The real solution is not to have the problem in the first place, of course, but that won't always be the case.

I do think some kind of "Show me everything about broken external references in the UI" would be a neat thing to have.  I'd certainly love to have had it during the current debugging.

The internal names with a ? prefix  - "?Edge1" are because we failed to find that in the referenced object.  The question becomes if there is an Edge1 in the referenced object, which I would expect if there is any geometry in it at all.  Shaise is likely fixing the broken reference with his recompute before entering the sketch.

If a file is saved without the correct elementMap entry for the external reference, we don't currently fix this automatically on reopening - the recompute is required to do this.

We might want to, although the LS3 approach is to put up a dialog and ask if recompute should happen.

Syres916, 2024-07-07:
> So, the two proposed clarifications of showing errors with external constraint are good ideas; showing both the internal name and the label ( which should be visible from python, but aren't to the [...]

I only would submit them when I have repeatedly received them and (as a fairly competent user) find no way around it or advise the user when the time to fix is a complete unknown/not being actively worked on.

> I do think some kind of "Show me everything about broken external references in the UI" would be a neat thing to have. I'd certainly love to have had it during the current debugging.
> 

So you would be after a Sketcher Validation Tool on steroids?!!

> If a file is saved without the correct elementMap entry for the external reference, we don't currently fix this automatically on reopening - the recompute is required to do this.
> 

So would you rather I put https://github.com/FreeCAD/FreeCAD/pull/15229 into Draft or close it?

bgbsww, 2024-07-07:
Well, with the caveat that I'm just one opinion here:

I'd keep that patch - I have no problem with descriptive warning messages.  I do see a slippery slope of expecting warnings that shouldn't happen to be guides for users.  But I have no idea where that line lies.  I'm merely pointing out that given messages that should not trigger, the goal should be to make them not trigger, not to make them beautiful.  Whether any specific message is in that category, I don't know.

For the case of broken constraints, that's gonna happen regardless of how good the code is, and I'd like us to have a useful procedure for users to follow, thus the remark that leads towards a Sketcher Validation Tool on steroids ( love that phrase ).

Too many words to say "#15229 seems fine, but I'm not sure the proposed solution of #15255 makes sense, though the goal does."

bgbsww, 2024-08-11:
@maxwxyz  this is actually about lack of visibility of the labels as well as the names in error message, so not a Topic: Toponaming.

maxwxyz, 2025-07-11:
Is this still relevant? The mentioned issues are already all closed and PRs merged.

Syres916, 2025-07-11:
> Is this still relevant? The mentioned issues are already all closed and PRs merged.

I believe this can be closed.

maxwxyz, 2025-07-11:
@Murmele let us know if this is still an issue

closed 2025-07-11 by maxwxyz